ghdl/ghdl #1686: GHDL | unit "numeric_std_unsigned" not found in library "ieee"

mohamadahmad, 2021-03-14:
Hi All,

I am trying to analyze my VHDL file with GHDL but I got this error:
**error: unit "numeric_std_unsigned" not found in library "ieee"**

I tried to uninstall and re-install GHDL, but without success.

Any help or suggestion?
Thank you,
Best regards
Mohamad

eine, 2021-03-15:
This is probably because you installed GHDL through the package manager of some Debian or Ubuntu. Can you please confirm which version of GHDL are you using and how did you install it?

mohamadahmad, 2021-03-15:
> This is probably because you installed GHDL through the package manager of some Debian or Ubuntu. Can you please confirm which version of GHDL are you using and how did you install it?
> 
> Ref: [#1649 (comment)](https://github.com/ghdl/ghdl/issues/1649#issuecomment-783704866)

Concerning the installation, I have first installed it using the apt package manager. Then I removed it and re-install the new version by compiling the software using the Makefile.

For the version:

_root@XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX# ghdl -v_
_GHDL 2.0.0-dev (1.0.0.r75.g6a9939f3) [Dunoon edition]_
_Compiled with GNAT Version: 9.3.0_
_mcode code generator_
_Written by Tristan Gingold._

_Copyright (C) 2003 - 2021 Tristan Gingold._
_GHDL is free software, covered by the GNU General Public License.  There is NO_
_warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE._

eine, 2021-03-15:
Which specific commands did you use for building and installing using the Makefile?

mohamadahmad, 2021-03-15:
> Which specific commands did you use for building and installing using the Makefile?

I used
make --> for building 
make install --> for installing

eine, 2021-03-15:
You should have used `./configure` first. The options you passed to it are relevant. Did you use `enable_gplcompat`?

mohamadahmad, 2021-03-15:
> You should have used `./configure` first. The options you passed to it are relevant. Did you use `--enable-gplcompat`?

Ok I will re-install it. 
I checked the configure file, the enable_gplcompat is `enable_gplcompat=unknown`

tgingold, 2021-03-15:
Also note that numeric_std_unsigned is only available with children 2008. Do you use `--std=08` ? 

-- 
Envoyé de mon appareil Android avec Courriel K-9 Mail. Veuillez excuser ma brièveté.

mohamadahmad, 2021-03-15:
> Also note that numeric_std_unsigned is only available with children 2008. Do you use `--std=08` ?
> […](#)
> -- Envoyé de mon appareil Android avec Courriel K-9 Mail. Veuillez excuser ma brièveté.

Can you, please, help me how/where should I use the `--std=08`? Thank you

eine, 2021-03-15:
Keep the enable gplcompat as `unknown` or set it to `false`, NOT true.

> Can you, please, help me how/where should I use the --std=08? Thank you

You need to set it after `ghdl -a`, `ghdl -e` or `ghdl --synth`. See https://ghdl.github.io/ghdl/quick_start/simulation/index.html

closed 2021-08-01 by tgingold