ghdl/ghdl #1687: Named associaton in function calls (slice direction runtime error / follow up)

MG-96, 2021-03-18:
**Description**
As mentioned in #1684 here is the follow up.
Following will be the same MWE which now produces a runtime error when executed.

**Expected behaviour**
After the bugfix from 5fa4c82fb259725fa544a5a432219f343e5dc043, named associations of ranges should work as expected. 

**How to reproduce?**
When running the MWE a error states that `slice direction doesn't match index direction` when calling the function.

```vhd :file: ent.vhd
use work.pack_RC_Add_n_F.all;

entity RC_Add_n_F is
    generic(n : natural := 4);
    port(A, B : in bit_vector(n-1 downto 0); Cin: in bit; Sum: out bit_vector(n-1 downto 0); Cout: out bit);
end RC_Add_n_F;

architecture Arch_RC_Add_n_F of RC_Add_n_F is
    signal result: bit_vector(n downto 0);
begin
    -- result <= RC_Add_n(A(3 downto 0), B, Cin);                               -- Works
    -- result <= RC_Add_n(A => A(3 downto 0), B => B, Cin => Cin);              -- Works
    result <= RC_Add_n(A(3 downto 0) => A(3 downto 0), B => B, Cin => Cin);     -- Fails
    Sum <= result(n-1 downto 0);
    Cout <= result(n);
end Arch_RC_Add_n_F;
```
```vhd :file: rcaddfn.vhd
package pack_RC_Add_n_F is
    function RC_Add_n( A, B :bit_vector; Cin : bit) return bit_vector;
end pack_RC_Add_n_F;
    
package body pack_RC_Add_n_F is
    function RC_Add_n( A, B :bit_vector; Cin : bit) return bit_vector is
        variable C:bit := Cin;
        variable SUM:bit_vector(A'length downto 0);
    begin
        loop_add_m: for I in 0 to A'length-1 loop
            SUM(I) := (A(I) xor B(I)) xor C;
            C := (A(I) and B(I)) or (C and (A(I) xor B(I) ));
        end loop loop_add_m;
        SUM(A'length) := C;
        return SUM;
    end RC_Add_n;
end pack_RC_Add_n_F;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a rcaddfn.vhd ent.vhd
ghdl -e RC_Add_n_F
ghdl -r RC_Add_n_F
```
When calling `ghdl -r RC_Add_n_F` this results in following error:
```
ghdl:error: slice direction doesn't match index direction at rcaddfn_ent.vhd:13
in process .rc_add_n_f(arch_rc_add_n_f).P0
  from: process work.rc_add_n_f(arch_rc_add_n_f).P0 at rcaddfn_ent.vhd:13
ghdl:error: simulation failed
```

**Context**
Please, provide the following information:

- OS:  Fedora 33 (Workstation Edition)
- Origin:
  - [x] Built from sources: 009f0b955b5084594b6912d261405fd1936750f9

tgingold, 2021-03-18:
According to LRM 2008 5.3.2.2 e) 2)
The direction of the range of the formal is defined by the base type of the index type, so the direction is to.

tgingold, 2021-03-19:
So I fear your code is not valid.  `A(3 downto 0)` designates a null slice, so it cannot be associated with `A(3 downto 0)`.
This looks like a weird (or uninformed) rule in the LRM.

MG-96, 2021-03-19:
Ok, thanks for the effort.

Since my professor used it like this in one of his examples, ModelSim will probably accept this form.
For my given MWE this constraint isn't a problem but my initial usecase was to associate a smaller slice to the function
(like `A => A(7 downto 4)`).
I'll probably have to figure out another way to do this since the above won't work.

closed 2021-03-19 by MG-96

diogratia, 2021-03-19:
Curious, I looked up the history.

IEEE Std 1076-2008 9.3.4 Function calls

>Evaluation of a function call includes evaluation of the actual parameter expressions specified in the call and evaluation  of  the  default  expressions  associated  with  formal  parameters  of  the  function  that  do  not  have  actual parameters associated with them. In both cases, the resulting value shall belong to the subtype of the associated  formal  parameter.  (If  the  formal  parameter  is  of  an  unconstrained  or  partially  constrained  composite type, then any undefined index ranges of subelements of the formal parameter are determined as described  in  5.3.2.2.)  The  function  body  is  executed  using  the  actual  parameter  values  and  default  expression values as the values of the corresponding formal parameters.

(As a standard implementer I'd naturally be suspicious about a semantic rule specified in a parenthetical sentence. You'd think the -2008 standard would have dropped the parentheses.)

5.3.2.2  Index constraints and discrete ranges

>The index range for each index of an array object or array subelement of a composite object is determined as follows:  
>>...  
>>e)  For an interface object of an array type, or a subelement of an interface object for which the subelement type is an array type, each index range is obtained as follows: Let the subtype index range be the corresponding index range of the subtype indication of the declaration of the object.
>>>  
>>>...  
>>>2)  If the subtype index range is undefined, and the interface object or subelement is associated by more than one association element or is associated by a single association element in which the formal designator is a slice name, then the direction of the index range of the object is that of the corresponding index subtype of the base type of the interface object, and the high and low bounds  of  the  index  range  of  the  object  are  respectively  determined  from  the  maximum  and  minimum values of the indices given in the association element or elements corresponding to the interface object or subelement.  
>>>...

For those noticing the lack of the `--std=08` command line argument in MG-98's description of the problem the answer can require a more analysis when using earlier revisions of the standard, here -1993:

 7.3.3 Function calls

>For each formal parameter of a function, a function call must specify exactly one corresponding actual parameter. This actual parameter is specified either explicitly, by an association element (other than the actual part open) in the association list, or in the absence of such an association element, by a default expression (see 4.3.2 ).  
>  
>Evaluation of a function call includes evaluation of the actual parameter expressions specified in the call and evaluation of the default expressions associated with formal parameters of the function that do not have actual parameters associated with them. In both cases, the resulting value must belong to the subtype of the associated formal parameter. (If the formal parameter is of an unconstrained array type, then the formal parameter takes on the subtype of the actual parameter.) The function body is executed using the actual parameter values and default expression values as the values of the corresponding formal parameters. 

Here the parenthetical says the formal takes on the subtype of the actual. Both of these methods can work (thanks to how subprogram call headers are elaborated) while potentially being incompatible as shown in this issue.

The parenthetical sentence was changed in -2008 (Language Change Specification LCS-2006-113, source documents FT-14, IR2054, [IR2063](http://www.eda-twiki.org/isac/IRs-VHDL-2002/IR2063.txt)). -1987 through -2002 show the same language as in -1993 above. 

So what's going on? We'd expect a notification in the standard of a change in semantics between revisions. 

-2008 Annex E (informative) Changes from IEEE Std 10776-2002:

>**Clause 9**  
>> ...
>>—  9.3.4: Additional rule relating to uninstantiated subprograms; clarification of parameter index range rules.  
>>...

Besides the use of 'uninstantiated' which doesn't appear in the Oxford Dictionary that the IEEE-SA uses as a English reference standard this appears to more than a clarification, it's incompatible or is it? The story can be seen in Issue Report 2063:

```none
Description of Problem  
----------------------  

Clauses 2.1.1.1 and 2.1.1.2 state that, for a formal whose interface  
declaration has a subtype indication that denotes an unconstrained  
array type, "the subtype of the formal in any call to the subprogram  
is taken from the actual associated with that formal in the call to  
the subprogram".  Yet in 3.2.1.1, in the places where it mentions  
formals of unconstrained array types, only the derivations of the  
"index ranges" are specified, because no subtype is in fact being  
created or implied.  
 
Proposed Resolution  
-------------------  
  
I think the intent is to have the index range(s) obtained from the  
actual, and the use of the word "subtype" in 2.1.1.1 and 2.1.1.2 is  
incorrect.  Perhaps they should be reworded to avoid using the term  
"subtype".  

VASG-ISAC Analysis & Rationale  
------------------------------  
  
The inconsistency should be resolved. However, it is anticipated that  
clauses 2.1.1.1 and 2.1.1.2 will be extensively rewritten as part of  
[... cut ...] era VHDL website in if I recall correctly 2015. He would likely have gotten someone to search backups. LCS_2006_Combined.pdf has inaccurate footnotes referencing LCS numbers while the page footers for the combined list are accurate. LCS-2006-113.pdf was used in this analysis after identifying the correct LCS and identifies FT-14, IR2056 and IR2063, the latter both -2002 Issue Reports.

(You might gather revision history can be important for those implementing multiple revision supporting tools.)

tgingold, 2021-03-19:
Note that `A => A(7 downto 4)` is Ok and working.