ghdl/ghdl #1689: Incorrect assignment when using unconstrained array of slv

avelure, 2021-03-19:
**Description**
I'm not fully sure yet what is the root cause of this issue.
I have a type that is an unconstrained array of unconstrained slv.
I have a constrained constant that uses this type and a variable of the same type and constraint, but one array element wider, where the width is defined with the constant's length attribute.
I then do a partial assignment between these variables and this seems to go horribly wrong.
I assign a fixed value to the first element of the variable and then the value of the constant to the remaining elements, but somehow the assignment overwrites from element 0 instead of starting from 1 as specified.
If I instead define the width of the variable with a literal value, same width as before, it works.
Checking the value of the length attribute in the process shows it is the correct value, so something is strange.
If I constrain the slv in the type it also works.

**Expected behaviour**
I expect it to be assigned to the right elements.

**How to reproduce?**
Following is an MWE

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;

architecture beh of test is
   type     t_slv_array  is array (natural range <>) of std_logic_vector;
   constant test_val : t_slv_array(0 to 1)(15 downto 0) := (others => x"BBBB");
begin
  
  process
	variable v_val : t_slv_array(0 to test_val'length)(15 downto 0); -- Change test_val'length to 2 and it works
  begin
	report "test_val'length " & to_string(test_val'length) & " v_val'length " & to_string(v_val'length);
    v_val(0) := x"AAAA";
    v_val(1 to 2) := test_val;
    assert v_val(0) = x"AAAA" report "Incorrect value, v_val is: " & to_hstring(v_val(0)) & " " & to_hstring(v_val(1)) & " " & to_hstring(v_val(2)) severity failure;
	wait;
  end process;

end architecture beh;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --ieee=standard --std=08 test.vhd
ghdl --elab-run --ieee=standard --std=08 test
```

**Context**
Please, provide the following information:

- OS: Win10
- Origin:
  - [x] Released binaries: `nightly-mingw64-llvm GHDL 1.0.0rc3 (1.0.0rc1.r2.g4dc37524) [Dunoon edition]`

Output from running the test file
```
test.vhd:16:8:@0ms:(report note): test_val'length 2 v_val'length 3
test.vhd:19:5:@0ms:(assertion failure): Incorrect value, v_val is: BBBB BBBB XXXX
.\test:error: assertion failed
in process .test(beh).P0
.\test:error: simulation failed
```

avelure, 2021-03-19:
I just noticed in another piece of my code that it is the same issue if the range selection is in the other direction.

```vhd :file: test2.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
end entity test2;
architecture beh of test2 is
   type     t_slv_array  is array (natural range <>) of std_logic_vector;
begin
  
  process
    variable v_val_1 : t_slv_array(0 to 1)(15 downto 0);
    variable v_val_2 : t_slv_array(0 to v_val_1'length)(15 downto 0) := (x"AAAA", x"BBBB", x"CCCC");
  begin
    report "v_val_1'length " & to_string(v_val_1'length) & " v_val_2'length " & to_string(v_val_2'length);
    v_val_1 := v_val_2(1 to v_val_1'length);
    assert v_val_1(0) = x"BBBB" report "Incorrect value, v_val is: " & to_hstring(v_val_1(0)) & " " & to_hstring(v_val_1(1)) severity failure;
    wait;
  end process;

end architecture beh;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --ieee=standard --std=08 test2.vhd
ghdl --elab-run --ieee=standard --std=08 test2
```

output
```
test2.vhd:16:8:@0ms:(report note): v_val_1'length 2 v_val_2'length 3
test2.vhd:18:5:@0ms:(assertion failure): Incorrect value, v_val is: AAAA BBBB
.\test2:error: assertion failed
in process .test(beh).P0
.\test2:error: simulation failed
```

I can of course work around this by assigning in a loop, but it is a bit difficult to find all the places in my code where the case applies when there is no error.

eine, 2021-03-19:
What about `v_val(1 to 2) := test_val(0 to 1);` and `v_val_1(0 to 1) := v_val_2(1 to v_val_1'length);`?

avelure, 2021-03-19:
> What about `v_val(1 to 2) := test_val(0 to 1);` and `v_val_1(0 to 1) := v_val_2(1 to v_val_1'length);`?

Same behavior.

closed 2021-08-01 by tgingold