ghdl/ghdl #1690: PSL cover - failure

Blebowski, 2021-03-19:
**Description**
When a PSL cover point is not covered during simulation, at the end, error (failure) is reported.

**Expected behaviour**
Uncovered PSL cover points shall not report error (failure), this is also how commercial simulators behave.
The rationale behind is:
 - One test-bench with many cover points scattered through TB and RTL.
 - Many tests. Each test is separate run of simulator (different procedure), but the same test-bench. E.g. as in Vunit.
 - Each test indends to verify (and cover) different scenario, situation.
 - After whole regression, functional coverage from all tests is merged together to find out if there were some cover poi

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tmeissner, 2021-03-19:
The simulation finishes without error, ghdl's return value is `0`. Only the text report on the command line contains cover errors like this:

```shell
../src/psl_cover.vhd:56:3:@12ns:(psl cover failure error): sequence not covered
```
Is it questionable if an error report is useful here, maybe we should report a warning instead.

But if want to get precise informations about PSL coverage you have to use the psl coverage report generated with the `--psl-report=<FILENAME>` option anyway.

Blebowski, 2021-03-19:
Hello, thanks for quick reaction.
I am not sure whether even warning is useful, maybe not as default, but enabled with some option (like --psl-warn-uncovered)

If you are running tests in "fail-fast" manner, meaning any report with "error" severity will make the test fail,
(I do that in my project, see: [](https://gitlab.fel.cvut.cz/canbus/ctucanfd_ip_core)), then the behavior makes all the
tests fail.

Btw. I use --psl-report in that test-bench, and I collect outputs from all test runs, merge them together, and create
final report like this:
[](http://canbus.pages.fel.cvut.cz/ctucanfd_ip_core/functional_coverage/functional_coverage_report.html)
This is as close as I can get to "professional verification" with GHDL + Vunit combination.

tmeissner, 2021-03-19:
Nice project.

> Btw. I use --psl-report in that test-bench, and I collect outputs from all test runs, merge them together, and create
> final report like this:
> [PSL report](http://canbus.pages.fel.cvut.cz/ctucanfd_ip_core/functional_coverage/functional_coverage_report.html)

Ah, cool. We do some similar at work with Questa, collecting all kinds of coverage from all sources (functional, code, assertions, OSVVM alerts & OSVVM bins). But our tools only aggregate data one unit test each.

tmeissner, 2021-03-19:
I didn't know that PSL cover directives are also taken in account when using the `--assert-level=` option. I could reproduced your problem:

```shell
cd work/psl_cover; ghdl -r --std=08 -fpsl tb_psl_cover --assert-level=warning --wave=psl_cover.ghw --psl-report=psl_cover_psl_coverage.json
../src/psl_cover.vhd:32:3:@2ns:(psl cover note): Transfer requested
../src/psl_cover.vhd:37:3:@3ns:(psl cover note): Transfer in progress
../src/psl_cover.vhd:37:3:@4ns:(psl cover note): Transfer in progress
../src/psl_cover.vhd:76:3:@8ns:(psl cover note): Transfer of length 3
../src/psl_cover.vhd:42:3:@9ns:(psl cover note): Transfer done
simulation finished @12ns with status 0
../src/psl_cover.vhd:56:3:@12ns:(psl cover failure error): sequence not covered
./tb_psl_cover:error: psl cover failure failed
in process .tb_psl_cover(sim).dut@psl_cover(psl).P1
make: *** [Makefile:17: psl_cover] Error 1
```

I agree that this isn't that useful. Initially I thought you only had the warnings in the report, but with that option teh simulation also fails when cover points weren't hit during simulation. So I also vote for a runtime option which explicitly enables the severity settings for PSL cover directives.

Blebowski, 2021-03-19:
Project: Thanks, I was inspired by Cadences Incisive and their coverage merging when doing the PSL merge script.
If you want, test/testfw/parse_psl_coverage.py does the job.

Issue:
If you could point me to a place in the code, where this is done (the reporting), I will could to try to add this feature into GHDL.
I tried to implement something in GHDL only once (dumping of PSL states, however, I failed there, since I have only rudimentary knowlege of ADA).

eine, 2021-03-19:
@Blebowski, there is some support in VUnit for merging coverage reports generated by GHDL: https://github.com/VUnit/vunit/pull/627. It might be interesting to have your PSL coverage feature added too.

tgingold, 2021-03-20:
What about a `--cover-severity=` option ?

Blebowski, 2021-03-20:
@eine : Yes, it might be good idea.  The script is however "quick and dirty" solution, so it would probably need to be reworked.

@tgingold @tmeissner IMHO any option name will do. Question is, how to specify the behavior, and what behavior has plausible use-case. Does it really make sense to report error/warning/info by un-covered cover points, if the same information can be anyway found out by parsing result of "--psl-report" ?

Further, if we put "--cover-severity=info", does that mean, that uncovered endpoints will report with info severity, correct? The same goes for "--cover-severity=error/failure". Is there a use-case where user would like to report uncovered points with these severities?

And what about if the cover point is hit. I typically supply custom message (which I expect to be fired with "info/note" severity at the time when it is hit). I assume that "--cover-severity" will not affect this case. It makes no sense to report anything else but "info" at time when cover point is hit. For that, PSL has "assert". In such case, the name "--cover-severity" is bit mis-leading...

If we truly want to specify the severity for the scenario, I would go with:
`--psl-uncovered-severity=none|info|warning|error`
with `none` being default option. If not, then I vote for simple:
`--psl-warn-uncovered`

PS: @tgingold if you can point me to place in the code where the report from uncovered PSLs is fired, I can try to implement it.

tgingold, 2021-03-20:
In grt-lib.adb, function ghdl_psl_cover_failed

Blebowski, 2021-03-20:
Thanks, I will go on, and let you know when there is a pull request.

Blebowski, 2021-03-20:
Added a PR in: [Pull request](https://github.com/ghdl/ghdl/pull/1694)

LarsAsplund, 2021-03-20:
Just a note. From a VUnit point of view the important thing is not the severity level being reported but whether or not the return code indicates an error or not. A printed warning with 0 return code will be interpreted as a pass.

PR #1694 (2021-03-22): Psl warn uncovered
**Description** Please explain the changes you made here.
Resolves failing psl cover points which were not covered during simulation. Added Wpsl-uncovered option.

- Issue fixed:
- [Close #1690](https://github.com/ghdl/ghdl/issues/1690)
- 
**When contributing to the GHDL codebase...**

- [x] DO make sure you are requesting to **pull a topic/feature/bugfix branch** (right side). Don't request your master!
- [x] DO make sure you are making a pull request against the **master branch** (left side). Also you should start *your branch* off *our master*.
- [x] DO make sure that GHD [... cut]
PR #1696 (2021-03-23): Psl warn uncovered
**Description** Please explain the changes you made here.
Added documentation for --psl-report-uncovered option.

 Closes #1690

**When contributing to the GHDL codebase...**

- [ x] DO make sure you are requesting to **pull a topic/feature/bugfix branch** (right side). Don't request your master!
- [ x] DO make sure you are making a pull request against the **master branch** (left side). Also you should start *your branch* off *our master*.
- [ x] DO make sure that GHDL can be successfully built. See [Building GHDL](https://github.com/ghdl/ghdl#building-ghdl).
- [ x] CONSIDE [... cut]
closed 2021-03-23 by tgingold