ghdl/ghdl #1692: Mcode bug assert_failure ortho_code-types

avelure, 2021-03-20:
**Description**
I tried running a tesbench with mcode backend, which already was running fine with the LLVM backend, but it fails with a bug during elaboration.
A quick look at `ortho_code-types.adb:113` indicates it might be related to VHDL-2008 unconstrained arrays.

**Expected behaviour**
Simulates fine.

**How to reproduce?**
As this happens during elab on a big project, I'm unable to create an MWE unless someone can point me in the direction of which kind of error it might be.

**Context**

- OS: Win10
- Origin:
  - [x] Released binaries: `Nightly MCODE GHDL release: 2.0.0-dev (1.0.0.r84.g9a80de30) [Dunoon edition]`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r84.g9a80de30) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: i686-w64-mingw32
C:\proj\top_tb_tc5_03\39518512\
Command line:
c:\prog\GHDL\nightly-mingw64-mcode\bin\ghdl.exe --elab-run --ieee=standard --std=08 -frelaxed-rules -fsynopsys --no-vital-checks --warn-others --warn-unused --warn-body --warn-parenthesis --warn-specs --warn-runtime-error --warn-pure --warn-static --warn-nested-comment --workdir=C:\proj\sim\hdlunit\library\work\v08 --work=work -PC:\proj\sim\hdlunit\library/ top_tb_tc5_03 beh
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ortho_code-types.adb:113
Call stack traceback locations:
0xed523c 0xedfb41 0xedfe69 0xee1c9e 0xf056b0 0xf44f36 0xf42982 0xf46a87 0xf4b352 0xf4b3a9 0xf38ee6 0xf39020 0xf4b1ec 0xf4b3dd 0xf20dde 0xf32a78 0xf21eef 0xf0f015 0xf1e360 0xf1876f 0xf5a6f3 0xeaa78b 0xdfb372 0xf5d448 0xc02c90 0xc01394 0x7627fa27 0x771e7c7c 0x771e7c4c
******************************************************************
```

tgingold, 2021-03-20:
I got a similar bug with #1689.  Maybe the latter would fix this one.

avelure, 2021-03-29:
> I got a similar bug with #1689. Maybe the latter would fix this one.

I tested latest nightly `GHDL 2.0.0-dev (v1.0.0-116-g9ab4fd18)` but get the same error, so it seems we weren't that lucky.

tgingold, 2021-08-01:
I am closing this issue as there is no testcase.  If you have one, please reopen this issue with it.

closed 2021-08-01 by tgingold