ghdl/ghdl #1693: Enable libbacktrace support in CI compiled LLVM-backed MacOS builds

benjaminmordaunt, 2021-03-20:
I've just installed the 9a80de30 LLVM MacOS nightly build as I wanted backtrace support for the otherwise useless assertion warnings that are received from the numeric_std package.

Unfortunately, even when analysing with -g and using --backtrace-severity=warning on the command-line, I'm still not seeing any backtrace.

Does the CI build use libbacktrace? Is there some way this can be enabled?

tgingold, 2021-03-20:
I don't think that libbacktrace works on MacOS.
If you are a MacOS developer, you can try to port a MacOS backtrace system library.

eine, 2021-03-20:
In fact, libbactrace is disabled in all CI builds (not only MacOS). See https://github.com/ghdl/ghdl/pull/1578 and https://github.com/ghdl/docker/issues/35.