ghdl/ghdl #1695: PSL - Dump assertions to GHW

Blebowski, 2021-03-22:
**Is your feature request related to a problem? Please describe.**
To be more user friendly, PSL assertions could be added to waveform dump (GHW only since FST and VCD).

**Describe the solution you'd like**
Each PSL assertion would be dumped as record of:
    - state  - Inactive, Running, Failed|Passed - for "assert" statement, Covered - For "cover" statement
    - number of currently running instances. Since SEREs can span multiple clock cycles, multiple in

**Describe alternatives you've considered**
Add a clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tgingold, 2021-03-23:
That should be doable, but the number of currently running instances (as it is not well defined!).

Blebowski, 2021-03-23:
Do I understand it right, that state of each NFA is held in vector with size equal to number of NFA states? Logic 1 in
n-th position means that there is "assertion being executed" in this state.
Therefore if we sum up number of "ones" in state vector (ignoring start and end state), should not we get number
of assertion instances running ? At least that is the trivial way how I would maybe try to do it.

Alternatively, one can keep counter and increment always upon leaving start state, and decrement upon reaching
final state, or upon aborting.

tgingold, 2021-03-23:
Yes, a PSL assertion is converted to an NFA.  But a 1 in a state can become several ones in case of a disjunction.

When you have implications like `always A -> B`, 1 will often leave the initial state but then fade either when the LHS become false (so the precondition is not true and the assertion is not activated) or when the RHS become true (the assertion holds).

Fundamentally, the number of running instances is not well defined (for good reasons) and even not defined by PSL.

But having a trace of whether an assertion is running or not is interesting.

Blebowski, 2021-03-24:
Hi,
you are right, number of running instances can be tricky. Dumping "state" and current value of "count" will be sufficient for each assertion.

With regards to implementation, I will try to do it, but I will need little help.
My main questions are:
  1. NFAs are built at analysis time, then translated to Ortho. Ortho is then translated via backend object code.
      Is AST (IIRs) available also during run-time? Or is it thrown out once it is generated? Are NFAs available
      at run-time (I mean "Nfat") ? I am thinking about extending this table with "state" record", alternatively with
      count record. Count would point to the "count" variable in generated code. State variable would also point
      to variable inside generated code (introduced in Traslate_Psl_Directive). 
 2. Alternatively, the information could be shared via RTI (If I understand the purpose of RT correctly). How does
     that work? How can I expose variable inside generated code via RTI?

In general, do you have some documentation about architecture of GHDL? Some presentation would be
good showing block diagrams, data flow diagrams, main steps of compilation/translation, description of main
data structures. I would really like to implement this feature (since it find it usefull in Modelsim and Cadence tools),
however the learning curve is very steep, and I only have rudimentary compiler knowledge...

umarcor, 2021-03-24:
@Blebowski, I hope these references can be useful for you:

- https://ghdl.github.io/ghdl/internals/Overview.html
  - See the following sections too (AST, Run Time Information, etc.)
  - This is a variant of the same diagram:
  
[image]

- https://github.com/ghdl/ghdl/issues/1189
  - The most similar to the content you want is probably the one from FOSDEM2015

tgingold, 2021-03-24:
The AST is not available during simulation but RTI describe the most important part of the design.
There is an RTI for each PSL construct and it points to the count variable, which is followed by the vector.

Maybe you can start with adding a few information to the RTI: number of states and for assertions of the form 'always a -> b', the transition state.

Or you can start by adding the assertions to the wave dump so that the name appears and whether it has passed or not.

Tell me where you prefer to start and I will give you more details.

Blebowski, 2021-03-25:
Hello thanks for links to documentation, it is more detailed than last time I was looking
at it couple years back.

@tgingold I think I will start at extending the translation and RTI with "assertion state" variable.

Blebowski, 2021-03-25:
Hello, I did dig through the RTI implemenation and read documentation, and I am still having trouble grasping the way how
it works. Particularly how is the RTI Associated with its context, and how does "Depth" affect it. Is there maybe some drawing  showing the list of generated RTIs as Ortho code in an address space, and its related "context" ? 
Also, I see that PSLs are Ghdl_Rtin_Object, which has no parent attribute. If running with "--trace-processes", then
Context associated with process generated by PSL assertion will fail in function "Get_Path_Name", which clearly is built for Ghdl_Rtin_Block, and therefore does not allow building full path up to the root of the design (since there is no parent attribute).

tgingold, 2021-03-26:
The RTIs describe an instance: the signals, the constants, the subtypes, etc declared by an architecture.  Because an architecture can be instantiated many times, the RTIs are relative and thus you should know the associated context to use them.

The file grt-psl.adb implements `--psl-report` which dumps the status of PSL directives.  It walks through the RTIs.

Depth is used for relations between RTIs: an object RTI also point to a subtype RTI.  But as they are both relative, the depth is used to know the parent degree of the subtype.

It is possible that "--trace-processes" doesn't handle correctly the PSL directive.  Either the parent should be added in the PSL RTIs, or they should be specially handled by "--trace-processes".

To dump assertions, you should either add and maintain a status for the PSL assert (so that would be mainly changes in src/vhdl/translate/trans-chap9), or recompute the status at run-time.

If you need to extend the RTIs for the PSL directives, you will need to update both `translate/` and `grt/` as RTIs are created at compile time but used at run-time.

Do not hesitate to ask questions, even if my answers are short...

Blebowski, 2021-03-26:
Hi, thanks. I will try to sum up what is clear for me:
  1. RTIs are created at compile time by trans_rtis. Their memory laytout (e.g. as created by "Generate_Psl_Directive")

Blebowski, 2021-03-26:
Hi, thanks. I will try to sum up what is clear to me:
 - RTIs code is created at compile time (trans_rtis). Memory layout of particular RTI must match how is it then read at run-time
   Therefore fields generated e.g. in "Generate_Psl_Directive" match memory layout of "Ghdl_Rtin_Object".
 - When translating PSL expression into "Ortho_Info", pointer to Ortho_Info is appended to IIr, so that during RTI translate pass,
   one can locate particular Ortho_Info_Acc (by Get_Info (Decl)).
 - E.g. for architecture, there is only one RTI, and data related to particular instance of this architecture are given by context.

What I dont understand, is how is it layed out in memory, I am trying to decode it from nm, objdump output of compiled architecture, but I am not very smart of it. Particularly, how is the context associated with the RTI? Also, how do I get from "Loc" attribute to the position of "Psl_Count_Var", since it seems that Loc should give us offset of particular instance of PSL assertion count variable.
Sorry if the questions are too trivial, but I am having trouble grasp the concept, and picture it. Some sort of UML diagram, or just diagrams with pointers and tables/structures would be really helpfull. I realize that knowing only little ADA also adds degree of difficulty, however, I am fluent in VHDL and familiar with C/C++ family languages, so I understand concepts of memory layout and pointers...

umarcor, 2021-03-26:
@Blebowski, just for you to be aware: you are doing very good. You already arrived much further than most of us could in the last several years. The topic is complex and you are tackling it nicely. As Tristan said, do not hesitate to ask, and iterate over it as much as you/we need. He is brief, but won't stop clarifying if you need to.

> What I dont understand, is how is it layed out in memory, I am trying to decode it from nm, objdump output of compiled architecture, but I am not very smart of it. 

I know that Tristan uses gdb, and I've used it before for inspecting the layout of fat pointers. That might be useful for you? https://ghdl.github.io/ghdl/development/Debugging.html#gnu-debugger-gdb

> Sorry if the questions are too trivial, but I am having trouble grasp the concept, and picture it.

This is the problem we all have with GHDL. Some/many potential contributors are mislead by the language (Ada). But the real issue is the complexity of the internal architecture, which is kind of independent of the implementation language. GHDL is a compiler/code-generator and a runtime library for a non-trivial execution model. Stay strong!

> Some sort of UML diagram, or just diagrams with pointers and tables/structures would be really helpfull.

I do agree. However, it changes rather frequently. Hence, creating (UML) diagrams can be rather worthless, if they become more misleading than helpful in the short term. I believe that's why Tristan prefers to answer specific questions about the codebase to anyone going deep into it. 

I've read that it's possible to generate code from UML descriptions or vice versa. Do you think that might be useful? Or is it probably overkill and challenging (with Ada)? Note that there have been discussions about (re)writing (parts of) GHDL in Rust, and there's a branch already: https://github.com/ghdl/ghdl/tree/rust. Therefore, in the mid term, the Ada constraints might not exist.

Blebowski, 2021-03-26:
Hello @umarcor , thanks for advice, I will give it a try. I tried to do this extension once already (more than year ago), but I had zero compiler know-how back then, now its just slightly better.
I dont think that describing the architecture with some diagrams is bad idea, even if it means that the diagrams could get out-dated due to rapid changes. After-all, the goal of each implemenation is to get stable enough after some time, that "massive reworks" are no longer needed. Otherwise, it is only @tgingold who can contribute everyone else is lost.
With regards to documentation generation from code, that can work, however, if one writes docs by hand, I often find it more explanatory. If the docs are generated (which is of course needed too), then they pretty much copy the code without much added value/simplification.
I will try to understand by examination of the code.

tgingold, 2021-03-26:
This is correct, but note that RTIs form a tree.  An architecture has a list of children and some of them (like processes) also have children.
You can dump the rtis using the `--dump-rti`.

If you use the gcc or llvm backend, you can compile your design with `-g` and use gdb to inspect values:

```
$ ghdl_gcc -a -g pkg.vhdl accum.vhdl tb_accum.vhdl 
$ ghdl_gcc -e -g tb_accum
```
```
(gdb) print work__tb_accum__ARCH__behav__RTI
$8 = {common = {kind = __ghdl_rtik_architecture, depth = 1 '\001', 
    mode = 0 '\000', max_depth = 0 '\000'}, name = 0x1000cd9c8, loc = 0x0, 
  linecol = 2318, parent = 0x1000f5100, nbr_child = 5, children = 0x1000f52a0}
```

The names of the symbols usually follow this pattern: library__entity__ARCH__architecture__identifier__RTI.

In `grt-psl.adb`, the count of PSL directives are printed using the RTI and the context.  The RTI gives the offset from the context to the count variable.

So how to get the context ?  Well, you cannot get it.  The top-level context (the address of the top-level instance) is known as well as its corresponding RTI.  From there, it is possible to follow the tree.  If an RTI designates an object, the location is usually an offset on the context, and the address of the object is the address in the context plus the offset.  If the RTI designates a sub-hierarchy, the offset in the RTI gives the indirect new address for the context.

BUT: you don't really need to know that.  It is already possible to gather all the PSL directives and get the address of the count variable. If you add another variable, its offset will be fixed wrt the count variable so there is nothing to add to the RTIs.  If you need new informations (like size of the state vector or the index of the active state), you will have to create another RTI type for the PSL directives.  Currently they use the ghdl_rtin_object type, which is used for all the objects.

umarcor, 2021-03-26:
> I tried to do this extension once already (more than year ago), but I had zero compiler know-how back then, now its just slightly better.

In fact, this "knowledge" you are asking about is useful for some other things that me and others tried. So, even if it feels very specific about PSL, features such as pyGHDL would also benefit.

> I dont think that describing the architecture with some diagrams is bad idea, even if it means that the diagrams could get out-dated due to rapid changes. After-all, the goal of each implemenation [...]

I can agree with that. Then, the point is, where to start? Tristan knows absolutely everything, so it is understandably daunting to ask him to "create diagrams" in general. We need to be more specific.

He did write the Overview and AST sections of the docs a few months ago. The diagram was picked from one of the slides, and I reworked it as shown above, with his feedback. I also created another couple of diagrams about libghdl and mixed-language simulation). Another user (@benreynwar) contributed section RTI with Tristan's feedback (#1306). In parallel, cosimulation stuff was moved to ghdl/ghdl-cosim and 20+ specific examples were added. All of this was done in the last year.

This is to say that we advance slow, and documentation is not the highest priority, but it's something we are willing to improve because we know that lower level details are preventing many potential contributions. Please, let us know which specific content you need. Of course, I know that's a very cheeky request. Typically, as soon as you know what specific content you need, you can also provide (partial) answers yourself. But that's the way it needs to be; we need to walk together. You investigate and ask, Tristan replies, I look and try to put it nice in the docs. Naturally, the users/identities behind "you" and "I" do vary.

> With regards to documentation generation from code, that can work, however, if one writes docs by hand, I often find it more explanatory. If the docs are generated (which is of course needed too), then they pretty much copy the code without much added value/simplification.

Indeed... https://ghdl.github.io/ghdl/gnatdoc/index.html is generated with AdaCore's `gnatdoc`. As you say, it provides some info, but it does not help much with understanding the overall architecture.

tgingold, 2021-03-27:
In the long term, I'd like to get rid off RTIs.  They duplicate the AST and they aren't very easy to maintain.
I think the best model would be to always use a JIT and keep the AST during runtime.
That would also mean no more object files so we have to think about how cosim would work.

That's also why (or a bad excuse) for not having a good documentation of RTIs...

Blebowski, 2021-03-27:
@tgingold Thanks, I will give it a try with gdb and see whether I can understand the memory layout.

With regards to documenation, I think it is much better than about two years ago.

Personally, I have no clue whether it is better to have JIT, or stick with RTIs, it is true that RTI duplicates AST though.
OTOH, RTI implementation is working, and it is possible to extend it (some people say: If it works, dont touch it :) )

If we are discussing architecture of GHDL (btw. once again thanks for the time dedicated!), I was always wondering
why are there multiple backends in GHDL? I never really undestood it. From FPGA/ASIC developers perspective,
the user does not see the backend (nor understands compiler architecture), therefore, it does not provide him/her
any feature. It is different for SW people who can dig into e.g. LLVM internals. It seems to me, that due to multiple
backends, the chances of bugs multiple as you get combinations of language/simulation features multiplied with
each backend implementation.

I personally would much more like to have proper coverage engine, full VHDL 2008 (especially hierarchical names),
or integrated Verilog support (to do gate level sims with SDF annotations, modern FPGA tools like Vivado does not allow
extraction of VHDL timing netlist), than multiple backends.

Please dont take it that I try to doubt the way this project goes, I just work with Cadence tools every day (as an ASIC
designer), and GHDL lacks lot of features which are IMHO more important than various backends...
If I have a time, I will try to contribute some (as this PSL in waves, which I also use in Cadence) :)

umarcor, 2021-03-27:
> If we are discussing architecture of GHDL (btw. once again thanks for the time dedicated!), I was always wondering
why are there multiple backends in GHDL? I never really undestood it.

This is pretty simple to understand, with the following hints (see https://ghdl.github.io/ghdl/development/building/index.html):

- The main backend is `mcode`, which is manually designed/implemented. That is an in-memory backend, which provides several performance advantages for analysis and elaboration. Therefore, it is the best solution for analysis, language servers, synthesis, etc. It is also the easiest to build because it has the less dependencies. The disadvantage is that, because it's written manually, each new target architecture needs to be supported explicitly. That's why it works for x86 only, and on Windows 32 bits only. No ARM, no PPC, no RISCV.

- LLVM allows targeting "any" architecture, because that was the main design goal when it was created. It does support JIT and it has multiple optimisation stages. Therefore, it's the most robust solution for supporting multiple architectures. Runtime (execution) of the simulations is also faster than `mcode`, but analysis is slower. It's the best solution for heavy/intensive simulations.

- GCC provides the same features as LLVM, but it does not support JIT, the build procedure is more complex, and it's heavier than others. However, GCC provides a unique feature which is not supported with LLVM yet: code coverage analysis. 

Moreover, debug `-g` is only supported with LLVM or GCC. At the same time, for co-simulation, mcode has certain limitations. Currently, LLVM or GCC allow linking simulation models with foreign language, and/or generating shared libraries. With mcode you can only use existing features in shared libraries. That is, you need to build the foreign code beforehand (see https://ghdl.github.io/ghdl-cosim/vhpidirect/index.html).

Overall, the idea in the future might be to deprecate GCC and support mcode and LLVM only. In order to do so, the missing features which are only supported by GCC need to be supported in LLVM too. For instance, debugging with any version of LLVM was enabled last august (#866). Code coverage is missing yet.

Deprecating GCC is also related to the following:

> https://github.com/ghdl/ghdl/issues/800#issuecomment-486076526
> The compile & link model used by LLVM and gcc backends prevents from doing post elaboration optimizations. And this model is not used by commercial simulators. Improving the simulation speed means getting rid of that model. That's why it could be deprecated in the future.

So, deprecating the compile & link model would affect when are the designs elaborated. Currently, generics are runtime (CLI) options. That is, a user can generate a single executable model of a VHDL design, and then run/evaluate it with different generics. Switching to JIT only (mcode and/or LLVM) would involve ditching that feature. See #1388. Also #781 and #980.

> From FPGA/ASIC developers perspective, the user does not see the backend (nor understands compiler architecture), therefore, it does not provide him/her any feature.

> Please dont take it that I try to doubt the way this project goes, I just work with Cadence tools every day (as an ASIC
designer), and GHDL lacks lot of features which are IMHO more important than various backends...

As you see, each backend does provide at least one unique feature which several users of GHDL rely on. Unexperienced users might not be aware of that, but intermediate and advanced users do select the best backend depending on their task. For instance, even though simulation with LLVM is faster, Jim from OSVVM uses mcode because of https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-time-resolution. At the same time, code coverage is a limited but very demanded/used feature. Hence, we cannot drop any of the backends right now.

I do agree that maintaining three backends is an effort. But it was done because supporting all three was more feasible than implementing all the features in a single one. If you are interested in the idea of "supporting code coverage through LLVM and switching the LLVM backend to JIT", I'm sure that Tristan will be happy to have some help for deprecating (lowering the priority) GCC.

umarcor, 2021-03-27:
> I think the best model would be to always use a JIT and keep the AST during runtime.
> That would also mean no more object files so we have to think about how cosim would work.

IMHO, this is not a great deal, from the user's perspective. Sure, I will need to update most of the ghdl-cosim examples so that any `*.c` and/or `*.h` files are compiled to a shared library BEFORE elaborating/running the GHDL/VHDL model. That might sound annoying at first, but I think it is sensible if the context is "switching GHDL to JIT only". It would still be partially compliant with the "standard" (which is being discussed).

My only concern is how would wrapping `ghdl_main` work. Currently, there are two options:

- `-e -Wl,main.c`: https://github.com/ghdl/ghdl-cosim/blob/master/vhpidirect/quickstart/wrapping/time/run.sh#L11
- `--bind` and `--list-link`: https://github.com/ghdl/ghdl-cosim/blob/master/vhpidirect/quickstart/wrapping/time/run.sh#L18-L21

None of those would work, because `--bind` and `--list-link` would not exist in a JIT only solution, isn't it?
Hence, the only solution would be to:

1. Create a shared lib with the foreign content to be used by VHDL.
2. Build the simulation model (which imports VHPIDIRECT functions from the shared lib) as a shared library with `ghdl -e -shared`.
3. Build a main foreign app, which loads the "hdl shared lib", which would itself load the "foreign shared lib". The foreign app can have a `main` and call `ghdl_main` anywhere (declared as extern).

I think that is a clean structure, even though building two libs and one app might feel unnecessary at first. Annoyingly, that fails on some platforms.

Blebowski, 2021-03-27:
Hello @umarcor,

thanks for reply, see my comments below.

> The main backend is mcode, which is manually designed/implemented. That is an in-memory backend, which provides several performance advantages for analysis and elaboration. Therefore, it is the best [...]

So part of GHDL can be used as language server for VS-Code. Aren't there other syntax highlightig plugins which will do the
job? Also, Sigasi free version is damn good for even very complex projects...
With regards to JIT, that is compiler internals, not a feature user can see. Portability is user visible feature, and I also agree that GCC has code coverage, OTOH it is very rudimentary (not comparable with proper toggle/code/branch/statement coverage)...

>  Jim from OSVVM uses mcode because of https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-time-resolution. At the same time, code coverage is a limited but very demanded/used feature. Hence, we cannot drop any of the backends right now.

Is there somewhere a list of "which option is supported with what backend" matrix ? 

What I was trying to say, is:
I dont understand why are there three backends, experimental synthesis, python bindings, etc...  if features such as
code/toggle/statement coverage, full VHDL 2008, full PSL, integration of Verilog (then possibly SV), interactive simulation (via GUI, important for quick debug, stuff like: pause, go, inspect delta cycle), good wave dumping configurability (variables, PSLs, chance to enable dumping in middle of sim) and performance [Optimize wave dump](https://github.com/ghdl/ghdl/pull/782) are not yet done?

IMHO, these are much more important for any hobby/entry level/pro FPGA/ASIC designer, than language server or experimental synthesis (though the connection to formal engine definitely seems interesting). Dont you think so ?

I would like to start contributing to the project regularly (since I have been using GHDL for some time, but I am hitting limits with lack of above described features), but I need to know the priorities in the project :)

There is one good quote for this:
"Perfection is achieved, not when there is nothing more to add, but when there is nothing left to take away."

tgingold, 2021-03-27:
The reason to have 3 backends is mainly historical.

The first available backend was gcc.  Then, I thought it was possible to write a JIT backend that would be much faster to generate code than gcc. Later, when LLVM appeared I tried to use it as a backend; it took time but it was finally available.

Yes, maintaining three backends costs a little bit; but it also makes the code more portable and avoid a too strong dependency with another project.

tgingold, 2021-03-27:
I don't pretend to do a perfect project!

There are no well defined priorities.  Or the priorities are defined by the contribution.

I worked on the language server because I wanted to learn about LSP and I thought it was also helpful to improve recovery errors.

I also think it was very important to be able to do synthesis so that VHDL could still be used in open source hardware.

All the features you have mentioned are important features.  Why aren't they already available ?  It's a matter of time and of priorities.

Although Ada is a very nice language, it is not widely known.  I will try to introduce a little bit of Rust (if it makes sense) for some low-level infrastructure and maybe also for the runtime, with the hope it will help contributions.

For some other features (hierarchical names, verilog integration) I fear they require elaboration during compilation.  Doable of course, but a significant change.  Again a matter of time!

tmeissner, 2021-03-27:
> IMHO, these are much more important for any hobby/entry level/pro FPGA/ASIC designer, than language server or experimental synthesis (though the connection to formal engine definitely seems interesting). Dont you think so

That's your opinion, but not of all GHDL users. The synthesis feature is very important for visibility of VHDL in the whole Yosys/Symbiflow/Librecores etc. universe. GHDL has got a lot of attention because of the synthesis feature. And in the end, almost all VHDL/Verilog descriptions end in real HW like FPGAs. It's kind of a game changer, that free & open source tool flows exist for FPGA and even ASIC implementation.

I, for example, use GHDL in my spare time for simulation, formal verification and synthesis. But the fact that GHDL supports formal (through synthesis) is the feature why I also use it at work now.

umarcor, 2021-03-28:
> So part of GHDL can be used as language server for VS-Code.

It's not exactly "part of GHDL". The "core" of GHDL is built as a CLI tool and as a shared library (`libghdl.so`) at the same time. The codebase is the same. Some tools, such as the language server or the [ghdl-yosys-plugin](https://github.com/ghdl/ghdl-yosys-plugin) do interact with the shared library, instead of the CLI. Again, building the shared library was not a priority/target per se, but a mechanism for using GHDL from other tools.

The language server is written in Python ([pyGHDL.lsp](https://github.com/ghdl/ghdl/tree/master/pyGHDL)), and it's not for VSCode only. It's for any client which understands LSP ([langserver.org](https://langserver.org/)). So far, people have used it with VSCode, emacs and vim. See [ghdl/ghdl-language-server](https://github.com/ghdl/ghdl-language-server).

> Aren't there other syntax highlightig plugins which will do the job?

There are not. In fact, adding an LSP to GHDL was delayed, because there were other projects trying to provide the feature. There are [kraigher/rust_hdl](https://github.com/kraigher/rust_hdl) (Rust) and [Paebbels/pyVHDLParser](https://github.com/Paebbels/pyVHDLParser) (Python). See https://github.com/rodrigomelo9/FOSS-for-FPGAs/issues/1#issuecomment-696827330.

- 1-2 years ago, Kraigher suggested combining rust_hdl with GHDL, because rust_hdl provides parsing, but not semantic analysis. That was recently commented/discussed again in [gitter.im/ghdl1/Lobby](https://gitter.im/ghdl1/Lobby).
- At the begining of this year, pyGHDL was reworked for adding `pyGHDL.dom`. [vhdl/pyVHDLModel](https://github.com/vhdl/pyVHDLModel) is a common interface/model that GHDL, pyVHDLParse and (potentially) rust_hdl can use.

rust_hdl does currently provide a language server. pyVHDLParser does not yet. None of them is a replacement for GHDL (yet).

> Also, Sigasi free version is damn good for even very complex projects...

Sigasi does not develop open source tooling. The actual interesting features are not available in the free version, not even in the evaluation/trial versions. The solution with most traction in the open source community is TerosHDL (see [terostechnology.github.io/terosHDLdoc](https://terostechnology.github.io/terosHDLdoc/), [TerosTechnology/vscode-terosHDL](https://github.com/TerosTechnology/vscode-terosHDL) and [gitter.im/TerosHDL/community](https://gitter.im/TerosHDL/community)).

> With regards to JIT, that is compiler internals, not a feature user can see. Portability is user visible feature,

The implications of JIT are primarily when is elaboration completed. That is, when can generics be defined/overriden. That is visible enough for deserving an specific request not to be deprecated: #1388. Moreover, as discussed above, the compile and link model has direct effect on co-simulation features.

> and I also agree that GCC has code coverage, OTOH it is very rudimentary (not comparable with proper toggle/code/branch/statement coverage)...

Code coverage with GCC was never implemented per se. It was/is a fortunate side effect of using GCC. That's why it's rudimentary. Moreover, it is not accurate. There are false positives/negatives, because it is not specifically developed for understanding VHDL. Nonetheless, AFAIAA it's the only open source solution for achieving it. There are several users that do use it, because it is useful regardless of the inaccuracies. Hence, there is no motivation for removing the feature, even though no one is going it improve it in the short-mid term.

> Is there somewhere a list of "which option is supported with what backend" matrix ?

See https://ghdl.github.io/ghdl/development/building/index.html. Apart from that, there's the `--time-resolution` (which we might add to that table, indeed). Differences for co-simulation are explained in [ghdl.github.io/ghdl-cosim](https://ghdl.github.io/ghdl-cosim/).

> What I was trying to say, is:
> I dont understand why are there three backends, experimental synthesis, python bindings, etc... if features such as
> code/toggle/statement coverage, full VHDL 2008, full [...]

I beg your pardon, but the base premise is incorrect. There is no pool of "generic" hours/resources which we can pick for "any feature", and "any feature" does not require the same amount of time/effort to implement. It is not true that "if we didn't support three backends, we would already have VHDL 2008 support", or "if we didn't support experimental synthesis, we would already have code/toggle/statement coverage". Technical and human implications are much deeper. In fact, we are working on all the features you mentioned, in parallel. When someone has time and knowledge for contributing, we provide the best info we can about the areas they want. However, we don't tell anyone what to do because of the features that a potential market would prefer. GHDL is not a commercial project. We want to provide the best product we can, but providing a product is not a target but a result of the exciting programming puzzles that we like solving. Each one is his/her own market and this is a common marketplace for sharing.

- code/toggle/statement coverage: as said, coverage with GCC is a side effect. No one did ever have motivation for studying and implementing coverage with GHDL. However, there are other tools which provide coverage features and which can work with GHDL. I'm thinking about OSVVM and probably cocotb.
[... cut ...]
Indeed, since I don't find myself capable of implementing some of the features I wish, a relevant part of my activity in this project is cleaning up. When I see anything duplicated or anything seems more complex than it should, I try to reorganize/rework. See https://github.com/ghdl/ghdl/graphs/contributors. I "added" 38kLOC and I "removed" 39kLOC. In this repo, I ain't a "code generator/creator". That role corresponds to Tristan (core), Patrick (Python and vendor scripts), Torsten (PSL), VPI/VHPI (Ben, Marlon, etc.)...

umarcor, 2021-03-28:
See also:

- https://github.com/ghdl/ghdl/wiki
  - https://github.com/ghdl/ghdl/wiki/Ideas:-GSOC
  - https://github.com/ghdl/ghdl/wiki/Ideas:-Documentation
  
Feel free to add some of the features discussed here to the wiki.

tgingold, 2021-03-28:
Priorities depend on the user!

Blebowski, 2021-03-28:
Hello all,
thanks for the replies, I will try to reply to everything I can.

> The reason to have 3 backends is mainly historical.
I see, I understand it is always lot of work to treat large existing code-bases.

> Yes, maintaining three backends costs a little bit; but it also makes the code more portable and avoid a too strong dependency with another project.
That is great point, in general, I like modularity of GHDL. OTOH, the added maintanence cost is what scares me.

> I don't pretend to do a perfect project!
> There are no well defined priorities.  Or the priorities are defined by the contribution.
I understand that. The time effort and amount of work done is enormous.
The fact that priorities are defined by contribution, is also important to know. There are OS projects,
e.g. Linux kernel which have hierarchical system, and there some code might not get accepted
if it does not match the vision of maintainer of particular sub-system (also, code quality requirements
are huge in Linux kernel). That is also one reason I was asking this. I would not like to spend time
on features (e.g. as PSL dump), if it then would not get accepted to master, because priorites are
different. But since there are no priorites (or I can set them by myself), this gives chance for lot of improvements I would like :)

> I also think it was very important to be able to do synthesis so that VHDL could still be used in open source hardware.
Here I dont understand why. Could you please explain? There are free versions of FPGA synthesis tools (not OS though),
which can be used without need to buy expensive tool. 
For ASIC, timing and placement driven synthesis (topographical flows) is usually needed (if you target something slightly technologically challenging), and STA engine therefore needs to be integrated. AFAIK yosys does not integrate timing driven synthesis? So anyway, for ASIC tape-outs I am not sure if it is possible to reliably TO without commercial tool (I am not talking about PoC). Please correct me if I am wrong.

Do I understand it right, that synthesis in GHDL provides some generic/simplified netlist (sort-of G-TECH), mainly for formal verification via Symbiosys ? That seems reasonable since I dont know about other options how to do "formal for free".
Btw. who requested the synthesis? I really would like to know :)

> All the features you have mentioned are important features.  Why aren't they already available ?  It's a matter of time and of priorities.
I see, it is still OS project, so amount of time is limited. I will try to contribute :) Btw. did you try to get some funding e.g. via
Linux foundation?  Or is it not the intent to do it? I dont know if it is possible, I am just asking...

> Although Ada is a very nice language, it is not widely known.  I will try to introduce a little bit of Rust (if it makes sense) for some low-level infrastructure and maybe also for the runtime, with the hope it will help contributions.
WBU C++ ??

> Sigasi does not develop open source tooling. The actual interesting features are not available in the free version, not even in the evaluation/trial versions.
OK, true, trace to definition across files is not available in free version of Sigasi. I will test "state machine viewer" of Teros-HDL, then :)

> Code coverage with GCC was never implemented per se. It was/is a fortunate side effect of using GCC. That's why it's rudimentary.
I know, I use it and I would definitely welcome more detailed implementation. I first need to boot into the code-base, I dont
think I am capable of writing coverage engine with my current GHDL knowledge.

> I beg your pardon, but the base premise is incorrect. There is no pool of "generic" hours/resources which we can pick for "any feature", and "any feature" does not require the same amount of [...]
I see. OTOH, if there were less backends and "side" features, maybe some features would be already available (which would make GHDL every bit as good, or event better than e.g. Modelsim PE). As was already said by @tgingold, it is matter of time and priorities. I will try to start contributing some of those features :)
I dare to disagree that providing a best product (whether OS, or commerical) is not about target, but about result of exciting
programming challenges. While exciting programming challenges are definitively part of innovation, it is only enough to create PoC (remember the 80/20 rule). PoCs are unusable by anyone but hobbyists. Creating good product means defining goals, priorities, and focusing on those. Ergo, my question: What are the priorities? :)

> full VHDL 2008: most of VHDL 2008 features are supported. Generic types and unconstrained types are not fully implemented yet. This is because those features made the design space explode. As useful [...]
I see. I was mainly talking about hierarchical names. I understand that needs change in elaboration, as one needs to dynamically locate the "mirrored" signals above the current compilation scope after the components are instantiated.

> full PSL: PSL was included in GHDL as a side effect of it being included in the VHDL 2008 LRM. Strictly, it is referenced. So, strictly, Tristan implemented the keywords for the parser not to choke [...]
Good to hear. I know about many people who do for VHDL, and SVAs too in SV.

[... cut ...]
of using VHDL2008 "all" keyword...

> The priorities of the project are your own. As you see, I provided references and a brief description of the state of each of the features you are interested in. From there on, set your own agenda!
Thanks for the references and time spent. I prefer to have debate (even stronger, with disagreement and possibly fight) in advance, than trying to implement something that would be completely useless in the end.

Lets get back to work (well, tommorrow, today is Sunday) :)

umarcor, 2021-03-28:
> The fact that priorities are defined by contribution, is also important to know. There are OS projects,
> e.g. Linux kernel which have hierarchical system, and there some code might not get accepted [...]

Yeah. The point with large projects such as the Linux kernel is that they get more requests than they can handle. Hence, they need to put huge filters/barries for lowering the volume that the benevolent dictator can review. This is the opposite: any contribution is to be accepted, unless it's fundamentally broken. Of course, it is desirable to discuss and plan beforehand, so you can be sure that the implementation is correct.

> Here I dont understand why. Could you please explain? There are free versions of FPGA synthesis tools (not OS though), which can be used without need to buy expensive tool.

"*Free as in specch, not as in beer*". GHDL is GPL. That's an strong philosophical/etical/political statement. 

Making a non open source tool free with a limited feature set is a marketing strategy for creating captive users. Those free versions you mention cannot be used with development boards which are not supported (sold) by the vendors. Even if it works, sometimes you are limited by the EULA to using one specific board. Moreover, most vendor tools cannot be used on non-x86 hosts. GHDL + Yosys + nextpnr work on ARM, ARM64, PPC64, etc.

Open and free/libre software allow using tools, boards and computers which big vendors do not consider profitable. The purpose of a company is making profit, not building good products.

> For ASIC, timing and placement driven synthesis (topographical flows) is usually needed (if you target something slightly technologically challenging), and STA engine therefore needs to be integrated. AFAIK yosys does not integrate timing driven synthesis?

GHDL + Yosys are mostly focused on the FPGA side on things. Not exactly on FPGAs, but on functional/behavioural verification of the designs. That's the frontend, the higher abstraction to middle abstraction. As a complement, ASIC specific tooling is gathered under the umbrella of OpenROAD/OpenLane. For STA, there is [The-OpenROAD-Project/OpenSTA](https://github.com/The-OpenROAD-Project/OpenSTA). See also [efabless/openlane: OpenLANE Architecture](https://github.com/efabless/openlane#openlane-architecture) and [efabless/openlane: OpenLANE Design Stages](https://github.com/efabless/openlane#openlane-design-stages).

[image]

Hence, Yosys synthesis does not need timing and placement driven synthesis, because placement is done afterwards. That's done by nextpnr, Verilog-to-Routing, Triton, etc. depending on the target.

>  So anyway, for ASIC tape-outs I am not sure if it is possible to reliably TO without commercial tool (I am not talking about PoC). Please correct me if I am wrong.

It is possible. That's out of discussion: https://www.efabless.com/projects/29. The quality of the results is still to be seen, though. Using open source tooling for ASICs might produce not as fast results or not as area-efficient layouts as using commercial tools. However, that's not the point. The point is whether results with open source tools are good enough. A year ago, it was not possible to evaluate because no open source PDK existed. Potentially, hundreds or thousands of projects will be targeting technology nodes which don't require perfection.

> Do I understand it right, that synthesis in GHDL provides some generic/simplified netlist (sort-of G-TECH), mainly for formal verification via Symbiosys ? That seems reasonable since I dont know about other options how to do "formal for free".

`ghdl --synth` provides some non optimised netlist. From a language perspective, it picks the synthesisable subset, and does some syntax transformations for simplifying the complexity of the used language features. The current output is VHDL. Generating EDIF has been discussed too: #1174.

When using ghdl-yosys-plugin, the unoptimised netlist is "converted" to Yosys RTLIL by using the C++ API. So, each element/component of GHDL's "internal netlist" is registered as a module in Yosys' internal representation of a circuit.

Formal verification is, once again, a fortunate side effect. ghdl-yosys-plugin was written for allowing synthesis. However, since people in YosysHQ are interested in formal verification, and Yosys' output can be used with SymbiYosys, with GHDL we were given a formal verification framework "for free". My understanding is that, similarly to how circuit/netlist components are registered through Yosys' API, assertions/verification constructs/statements are registered too.

> Btw. who requested the synthesis? I really would like to know :)

Dozens of people. Honestly, it would be shorter to answer "who didn't want synthesis"? The reply is probably "no one". Unfortunately, the old mailing list died, so it's not easy to find past discussions:

- https://www.mail-archive.com/search?l=ghdl-discuss@gna.org&q=subject:%22Re%5C%3A+%5C%5BGhdl%5C-discuss%5C%5D+Synthesis+for+FPGAs%22&o=newest
- https://ghdl-discuss.gna.narkive.com/1t4iT9o8/synthesis-for-fpgas

> Btw. did you try to get some funding e.g. via
> Linux foundation? Or is it not the intent to do it? I dont know if it is possible, I am just asking...

Did never try seriously. However, see [March 11, 2021 6:49 PM](https://gitter.im/vhdl/General?at=604a5840457d6b4a94bb579f).

[... cut ...] 0/d3-wave), [wavedrom/zoom)](https://github.com/wavedrom/zoom) and [tikz-timing](https://www.ctan.org/pkg/tikz-timing).

> Thanks for the references and time spent. I prefer to have debate (even stronger, with disagreement and possibly fight) in advance, than trying to implement something that would be completely useless in the end.

Since we already "introduced" ourselves, feel free to open specific issues or to ask in the chat for more in-deep discussions about any of the feature we mentioned.