ghdl/ghdl #1698: Crash using sequence in formal verification

MJoergen, 2021-03-25:
**Description**
GHDL crashes during synthesis (formal verification).

**Expected behaviour**
There should be no crash. I'm not sure, if the syntax is legal. If not, an error message should be shown.

**How to reproduce?**

```vhd :file: ent.psl
-- ent.psl
vunit i_ent(ent(rtl))
{
   default clock is rising_edge(clk);

   assert always {[+]; input} |=> {output};
}
```

```vhd :file: ent.vhd
-- ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
   port (
      clk    : in  std_logic;
      input  : in  std_logic;
      output : out std_logic
   );
end entity ent;

architecture rtl of ent is

   signal r : std_logic;

begin

   process (clk)
   begin
      if rising_edge(clk) then
         r <= input;
      end if;
   end process;

   output <= r;

end architecture rtl;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd ent.psl -e ent
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit SHA`
```
GHDL 2.0.0-dev (1.0.0.r102.g9f796794) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
```

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
ghdl:warning: unexpected extension for vhdl file "ent.psl" [-Wlibrary]

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r102.g9f796794) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/mike/
Command line:
ghdl --synth --std=08 ent.vhd ent.psl -e ent
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : psl-nodes.adb:155 index check failed
Call stack traceback locations:
0x55b4e6eed5ea 0x55b4e6f022c5 0x55b4e6f0280a 0x55b4e6f027b7 0x55b4e6f027a1 0x55b4e6f02847 0x55b4e6f0278b 0x55b4e7036b98 0x55b4e70377e5 0x55b4e707df2c 0x55b4e7033a86 0x55b4e70348fa 0x55b4e7034ab4 0x55b4e70e162a 0x55b4e710821b 0x55b4e7108aca 0x55b4e702b4b7 0x55b4e71b7504 0x55b4e6dddaad 0x7fc5595f90b1 0x55b4e6ddc1fc 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

closed 2021-03-27 by tgingold

tmeissner, 2021-03-27:
I think it is legal. Should be similar to something as
```vhdl
assert always {true[+]; input} |=> {output};
```

eine, 2021-03-27:
@tmeissner did you mean "not legal" or "illegal"?

tmeissner, 2021-03-27:
> @tmeissner did you mean "not legal" or "illegal"?

Legal - as I wrote 😉

eine, 2021-03-27:
Ok. So, in other words, you mean that "`true` is optional in this case"?

- `{true[+]; input} |=> {output}`
- `{[+]; input} |=> {output}`

tmeissner, 2021-03-27:
> Ok. So, in other words, you mean that "`true` is optional in this case"?
> 
> 
> 
> - `{true[+]; input} |=> {output}`
> 
> - `{[+]; input} |=> {output}`

Yes, I think so.

commit 7a53e87a (2021-03-27): psl-subsets.adb: allow optionnal sequence to plus repeat. Fix #1698