ghdl/ghdl #1699: GHDL synthesis creates 2 arrays for memory

antonblanchard, 2021-03-25:
If I write out verilog before yosys optimisation (ie `ghdl --std=08 test2.vhdl -e test2; write_verilog`), I see two arrays for some memories, eg:

```
  reg [2:0] \$mem$\13  [15:0];
  reg [2:0] \13  [15:0];
```

`$mem$13` is never read or written, and disappears after optimisation. I thought it was because the array was read asynchronously, but it reproduces with synchronous reads. Perhaps I am misunderstanding what is going on here, but it makes me wonder if we aren't using the memory created by GHDL.

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
    port (
        clk  : in std_ulogic;
        addr : in std_ulogic_vector(4 downto 0);
        data : out std_ulogic_vector(2 downto 0)
    );
end entity test2;

architecture rtl of test2 is
    type result_t is array(integer range 0 to 15) of std_ulogic_vector(2 downto 0);

    constant result_select : result_t := (
        0      => "001",
        1      => "001",
        2      => "001",
        3      => "001",
        4      => "001",
        5      => "001",
        6      => "001",
        7      => "001",
        others => "000"
        );
begin

    --lookup_0: process(all)
    --begin
        --data <= result_select(to_integer(unsigned(addr)));
    --end process;

    lookup_0: process(clk)
    begin
        if rising_edge(clk) then
            data <= result_select(to_integer(unsigned(addr)));
        end if;
    end process;
end;
```

tgingold, 2021-03-26:
Is it new or it is old ?
I have pushed a commit a few days ago to handle read after write for memories.

umarcor, 2021-03-26:
@antonblanchard containers were updated 7-9h ago. I restarted them now: https://github.com/hdl/containers/actions. Please, pull and retry.

antonblanchard, 2021-03-26:
I see this in some VHDL to Verilog converted code from a month ago, so it isn't new. I managed to dump something earlier out of yosys, and it seems like we are creating the memory correctly, but the \$mem\$13 array is just unused.

```
module test2(clk, addr, data);
  wire [3:0] _0_;
  wire [47:0] _1_;
  wire [2:0] _2_;
  input [4:0] addr;
  input clk;
  output [2:0] data;
  reg [2:0] \$mem$\13  [15:0];
  \$mem  #(
    .ABITS(32'd4),
    .INIT(48'h249249000000),
    .MEMID("\\13"),
    .OFFSET(32'd0),
    .RD_CLK_ENABLE(1'h1),
    .RD_CLK_POLARITY(1'h1),
    .RD_PORTS(32'd1),
    .RD_TRANSPARENT(1'h0),
    .SIZE(32'd16),
    .WIDTH(32'd3),
    .WR_CLK_ENABLE(1'h1),
    .WR_CLK_POLARITY(1'h1),
    .WR_PORTS(32'd0)
  ) \13  (
    .RD_ADDR(_0_),
    .RD_CLK(clk),
    .RD_DATA(_2_),
    .RD_EN(1'h1),
    .WR_ADDR(),
    .WR_CLK(),
    .WR_DATA(),
    .WR_EN()
  );
  assign _0_ = 4'hf - addr[3:0];
  assign data = _2_;
endmodule
```

antonblanchard, 2021-03-26:
I have very little idea what I am doing here, but this makes the extra arrays go away, while still producing verilog that I can successfully simulate:

```
@@ -425,7 +425,7 @@ static void import_memory(RTLIL::Module *module, std::vector<RTLIL::Wire *> &net
        memory->name = "$mem$" + mem_str;
 
        //  Add it to module.
-       module->memories[memory->name] = memory;
+       //module->memories[memory->name] = memory;
```

tgingold, 2021-03-27:
I think you are just plain right.
I removed the extra memory.
Thanks!

closed 2021-03-27 by tgingold