ghdl/ghdl #1702: Top-level of ghdl synth output seems not to be processed

umarcor, 2021-03-29:
**Description**
When processing a design with `ghdl --synth`, the top level entity (and libraries) seem not to be modified. 

**Expected behaviour**
Since other `context` statements are "translated", I would expect the one at the top to be translated too. At the same time, generics are shown (and used in the top-level ports), despite the design being elaborated already (thus generics being fixed). I would expect generics to be removed.

**How to reproduce?**

[synth_context_generics.zip](https://github.com/ghdl/ghdl/files/6220355/synth_context_generics.zip)

`synth/axis_buffer_synth.vhd` is generated by executing `synth/ghdlsynth.sh`.

```vhdl
library ieee;
context ieee.ieee_std_context;
entity axis_buffer is
  generic (
    data_width: integer := 32;
    fifo_depth: integer := 3
  );
  port (
    s_axis_clk: in std_logic;
    s_axis_rstn: in std_logic;
    s_axis_rdy: out std_logic;
    s_axis_data: in std_logic_vector (data_width - 1 downto 0);
    s_axis_valid: in std_logic;
    s_axis_strb: in std_logic_vector (data_width / 8 - 1 downto 0);
    s_axis_last: in std_logic;
    m_axis_clk: in std_logic;
    m_axis_rstn: in std_logic;
    m_axis_valid: out std_logic;
    m_axis_data: out std_logic_vector (data_width - 1 downto 0);
    m_axis_rdy: in std_logic;
    m_axis_strb: out std_logic_vector (data_width / 8 - 1 downto 0);
    m_axis_last: out std_logic
  );
end axis_buffer;
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity fifo_37_3 is
  port (
    clkw : in std_logic;
    clkr : in std_logic;
...
```

**Context**
Please, provide the following information:

- OS: MSYS2
- Origin:
  - [x] Package manager: `GHDL 2.0.0-dev (1.0.0.r101.g420a1d3e)`

tgingold, 2021-03-29:
It is done on purpose so that the synthesized design can be simulated with the same testbench.

But we could add a format to write the entity as it is.

umarcor, 2021-03-29:
Making it optional would be really nice. In fact, this is related to #1703. Both issues make Vivado choke on the output of `ghdl --synth`.

tgingold, 2021-03-29:
Why does Vivado chock on full entity ?

umarcor, 2021-03-29:
Not on the entity, but on the `context` declaration in the library section of the entity. I believe it would be ok to preserve the entity if the context is rewritten with use clauses only.

closed 2021-03-30 by tgingold

umarcor, 2021-03-30:
FTR, when using `--out=vhdl` (default), the top-level entity is shown at the top of the output. However, when using `--out=raw-vhdl`, the top-level entity might be printed in the middle of the output.

tgingold, 2021-03-31:
I suppose this is the last entity displayed.
Not sure if it is a problem.