ghdl/ghdl #1704: Returning Aggregate Crashes

hughesjs, 2021-03-29:
**Description**
Returning Aggregate Crashes

**Expected behaviour**
Aggregate returned from function

**How to reproduce?**

Call the function below (note that the first two parameters below are function inputs in the actual code.

```vhdl
subtype word is std_logic_vector(word_len - 1 downto 0);
function F return word is 
begin
  return ("1010","101010101010", others => '0');
end function;   
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:  Ubuntu 20.04 LFS
- Origin:
  - [x] Built from sources: `1.0.0.r107.ga05d3cb7`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues 
GHDL release: 2.0.0-dev (1.0.0.r107.ga05d3cb7) [Dunoon edition]
Compiled with GNAT Version: 9.3.0 
Target: x86_64-linux-gnu
Command line: ghdl -r -v --std=08 --workdir=./build --warn-no-hide -frelaxed assembler                                                  Exception TYPES.INTERNAL_ERROR raised 
Exception information: 
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations: 
0x560093afa1e2 0x560093d159c0 0x560093d164ce 0x560093d1850d 0x560093d18646 0x560093d180cd 0x560093cdca98 0x560093cdcb76 0x560093cdd296 0x560093cdd531 0x560093cdd4ab 0x560093cdf1db 0x560093cdf294 0x560093ce20c8 0x560093ce234b 0x560093ce2158 0x560093c9c0d6 0x560093caf662 0x560093caf82f 0x560093ca233e 0x560093ca2007 0x560093ca238c 0x560093caf69e 0x560093caf867 0x560093d1d2ad 0x560093ccdf1b 0x560093cb744d 0x560093d2547e 0x560093c8d57f 0x560093c8a6d2 0x560093c987d6 0x560093d29ded 0x560093c68f2b 0x560093b9f527 0x560093d2d70a 0x560093951aad 0x7f3aaec6e0b1 0x5600939501fc 0xfffffffffffffffe                      ******************************************************************                                                        
```

tgingold, 2021-04-01:
I cannot reproduce the crash.  Please provide an MVP.

diogratia, 2021-04-01:
See [Use Others with Aggregate in VHDL](https://stackoverflow.com/questions/66846696/use-others-with-aggregate-in-vhdl) on Stack Overflow (where no reproducer is provided there as well). 

This appears to be a -2008 aggregate where aggregate element expressions `get_opc(ins.op)` and  `std_logic_vector(ins.val)` are of the type of the aggregate.

IEEE Std -2008 9.3.3.3 Array aggregates:

>For an aggregate of a one-dimensional array type, each choice shall specify values of the index type, and the expression of each element association shall be of either the element type or the type of the aggregate. If the type  of  the  expression  of  an  element  association  is  the  type  of  the  aggregate,  then  either  the  element  association shall be positional or the choice shall be a discrete range.  
> ...  
>Apart  from  a  final  element  association  with  the  single  choice  **others**,  the  rest  (if  any)  of  the  element  associations of an array aggregate shall be either all positional or all named. A named association of an array aggregate is allowed to have a choice that is not locally static, or likewise a choice that is a null range, only if the aggregate includes a single element association and this element association has a single choice. An **others** choice is locally static if the applicable index constraint is locally static.

>The index range of an array aggregate that has an **others** choice shall be determinable from the context. That is, an array aggregate with an **others** choice shall appear only in one of the following contexts:  
>> ...  
>>d) As  the  result  expression  of  a  function,  where  the  corresponding  function  result  type  is  a  fully  constrained array subtype
>> ...  

The subtype `word` used as a type mark for the function result type appears to be fully constrained. This corresponds to -1993 73.2.2 Array aggregates wherein choices must be of the aggregate element type, a limitation evaded in the -2008 array aggregate feature.

You could note the type mark of the result type for function `get_opc` is not shown and may be unconstrained. There is no indication the type of record `ins` element `val` type of `ins` is unconstrained but is likely to be, providing an immediate value of a singular type `instruction` where the immediate value could vary by instruction class.

There'd be a work around involved in declaring a subtype for the the zero fill designated by the **others** choice as an aggregate and using concatenation instead of aggregation for the first two elements of the aggregate shown in the image. It would require knowing the length of an object of subtype `word`.

```vhdl
    
    function to_word (ins: instruction) return word is
        subtype zeros is std_logic_vector 
            (word_length - ins.op'length - ins.val'length - 1 downto 0);
    begin
      -- return get_opc(ins.op) & std_logic_vector(ins.val) & (others => '0');
      return get_opc(ins.op) & std_logic_vector(ins.val) & zeros'(others => '0');
    end function;
```

This method could also be used in VHDL design specifications compliant to revisions of the standard earlier than -2008.

closed 2021-08-01 by tgingold