ghdl/ghdl #1708: PSL: Goto/equal repeat operator on non-boolean expressions crashes

Blebowski, 2021-04-03:
**Description**
GHDL crashes when GOTO/Equal repeat operators are applied on non-boolean expression
(e.g. braced SERE created by concatenation).

**Expected behaviour**
Not to crash and report error since PSL LRM (6.1.1.1.6) states that Equal/Goto repetition shall
be acceptable only for boolean values (not SEREs).

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: tb_top.vhd
Library ieee;
use ieee.std_logic_1164.all;
use STD.textio.all;

entity tb_top is
end entity;

architecture tb of tb_top is

    signal a,b : std_logic := '0';
    signal clk_sys : std_logic;

begin

    gen_clock_proc : process
    begin    	
    	clk_sys <= '1';
        wait for 5 ns;
        clk_sys <= '0';
        wait for 5 ns;
    end process;

    test_proc : process
    begin
          a <= '1';
          wait for 50 ns;
          std.env.finish;
    end process;

    -- psl default clock is rising_edge(clk_sys);

    -- psl my_seq : assert never {a = '1'; b = '1'}[=3];

end architecture tb;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -c --std=08 -fpsl tb_top.vhd -r tb_top
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 800656a30c2385b591766281cbc35de1582ce8f5

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r129.gedbca9df.dirty) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/DOKUMENTY/Projekty/ghdl/psl_ghdl_testing/
Command line:
ghdl -c --std=08 -fpsl top.vhd -r tb_top
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f028f2bb3be 0x556cdf7e8354 0x556cdf7e3ee0 0x556cdf931b37 0x556cdf931b0c 0x556cdf931f96 0x556cdf931e85 0x556cdf931fd1 0x556cdf932df7 0x556cdf939ebf 0x556cdf939fcc 0x556cdf93a123 0x556cdf972fbb 0x556cdf97a24c 0x556cdf92fc72 0x556cdf930ae6 0x556cdf930ca0 0x556cdf9df725 0x556cdf9f1e86 0x556cdfab1e8d 0x556cdf9f26ba 0x556cdf9276a3 0x556cdfab5a44 0x556cdf6daa9d 0x7f028f0be0b1 0x556cdf6d91ec 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

closed 2021-04-03 by tgingold

Blebowski, 2021-04-04:
Hello, I tested this issue with 5d22419, unfortunately, the behaviour is still buggy.
It is posssible, I wrongly described the expected behavior, so I will try to describe it more closely.
I tried to examine PSL spec, and I also ran following examples over other simulators on EDA playground (shortly just EDAP).
I will try to describe several cases which seem to be problematic.

Following examples apply for both GOTO/EQUAL([->n]/[=n]) repetition.

Lets assume we have:
 signal a,b,c,d,clk  : std_logic;
-- psl default clock is rising_edge(clk);

Example 1:
-- psl my_seq : assert never a[=3];
- GHDL: crashes
- EDAP: runs ok
- Expected: Since 'a' is 'HDL_Expr' in PSL spec, I would expect it to be accepted by GOTO/EQUAL repetition.

Example 2:
-- psl my_seq : assert never (a = '1')[->3];
- Semantically equal to Example 1
- EDAP: runs ok
- GHDL: Throws compile error, although, a = '1' is HDL_Expr
- Expected: I would expect it to be accepted by GOTO/EQUAL repetition, since a = '1' is HDL_Expr which is in production
  of  Boolean according to A.3.7 of PSL spec.

Example 3:
-- psl my_seq : assert never {a = '1'}[->3];
- Semantically equal to Example 1 and 2. But expression is now defined as braced SERE sequence of 1 cycle repeated 3 times.
- EDAP: runs ok
- GHDL: Throws compile error ('[->' not allowed on a SERE).
- Expected: I would expect compiler to figure out that this sequence is in fact the same as previous case (altough defined as braced SERE. maybe emit a warning with sth like: GOTO/EQUAL not allowed on SERE, but since SERE is equivalent to
simple boolean, reducing to simple boolean). PSL spec does not strictly allow this, as there is no way how to produce 'Boolean'
from braced SERE (Sequence).
Before the fixes introduced in `6617613f93e70df2e2b964f380652ea43e2bb2a5` this worked.

Example 4:
-- psl my_seq : assert never {a = '1'; b = '1'}[->3];
- Now different, object to be repeated is now not equivalent to boolean expression, it is sequence of two cycles.
- EDAP: Throws error that the [->] operator cannot be used on non-bool expression.
- GHDL: Throws compile error ('[->' not allowed on a SERE).
- Expected: Throw compile error saying that the operator is not allowed on non-bool expressions.
Before the fixes introduced in `6617613f93e70df2e2b964f380652ea43e2bb2a5` this crashed GHDL.

The trick is in interpretation of 'Boolean' production of PSL spec. Having {a} or {a = '1'} is not strictly
allowed for these GOTO/EQUAL repeats since by surrounding the simple HDL expression by {}
suddenly makes it sequence. But semantically, it is still just a simple expression. I think only sequences
which can not be reduced to boolean expressions (like ";" or ":" concats) should be rejected by GOTO/EQUAL.  

Example 5:
-- psl my_seq : assert never (a = '1')[*3];
- Related to star repeat, rather than EQUAL/GOTO repeat.
- EDAP: Runs OK.
- GHDL: Throws compile error (';' is expected instead of '[*'), since it interprets "( )" parenthesis as those of property,
  and not those of simple HDL_expression (Parse_Property). To make it work, one must surround "a='1' " in {} braces.
From PSL spec, this should not work without surrounding by curly braces because:
assert FL_property;
FL_property ::=
   SERE
Sequence ::=
   {SERE}
SERE ::=
  | SERE[*x]
  | Boolean
Boolean ::=
   HDL_Expression
which then gives the "()" braces.

Is it possible to implicitly assume "{}" around SERE in "always SERE" expression to create sequence?

tmeissner, 2021-04-04:
In the book I own (A practical Introduction to PSL) they clearly distinguish between SERE (`s`) and boolean expressions (`b`). And the consecutive repetition operators are defined for `b` and `s`, but the non-consecutive only for `b`.

But they don't write, if a SERE could also be interpreted as boolean expression, the only say it the other way:

> We assume two designated boolean expressions true and false belong to B
> Every boolean expression b ∈ B is a SERE

For my understanding, Examples 1 & 2 should work, but I'm not sure for example 3. Example 4 is clearly invalid. Example 5 should works as it is similar to example 2.

commit 6617613f (2021-04-03): psl: prefix of goto/non-consecutive repetition is a bool.  Fix #1708