ghdl/ghdl #1715: CONSTRAINT_ERROR crash on combo of stable attribute and signal type conversion

abyszuk, 2021-04-07:
**Description**
MWE posted here causes GHDL to crash if file is analyzed in `--std=08` mode.
After some investigation, I can say that it's caused by combination of these two lines of code:
`wait until clk'stable`
and
`    a_i  => std_logic_vector'(a),`
If I either:
1. comment out the `wait until clk'stable` line
2. remove the (unnecessary) type conversion in port assignment `a_i => a,`

**Expected behaviour**
Not to crash and either work or report error.

**How to reproduce?**
I'm attaching MWE. This file causes GHDL crash when using VHDL2008 mode.
Tested with mcode and GCC backends.

```vhd :file: mwe_stable.vhd
library ieee;
use ieee.std_logic_1164.all;

entity comp1 is
  port (
    a_i  : in std_logic_vector(3 downto 0);
    x_o : out std_logic
  );
end entity;

architecture arch of comp1 is
begin
  x_o <= a_i(0) and a_i(1) after 5 ns;
end arch;

library ieee;
use ieee.std_logic_1164.all;

entity mwe is
end entity;

architecture arch of mwe is
  signal clk : std_logic := '0';
  signal a   : std_logic_vector(3 downto 0);
  signal x   : std_logic;
begin

  process
  begin
    a <= "1100";
    wait for 10 us;
    a <= "1011";

    wait until clk'stable;
  end process;

  x_comp1 : entity work.comp1
  port map (
    a_i  => std_logic_vector'(a), -- this crashes if 'wait until clk'stable' isn't commented out
  --  a_i  => a, -- this works
    x_o  => x
  );

end arch;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 mwe_stable.vhd
```

**Context**
Please, provide the following information:

- OS: Official GHDL Docker image running on CentOS 8

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-144-g68a7f85c) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/work/cce/cce/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/8.3.0/ghdl1 --ghdl--std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o mwe_stable.s mwe_stable.vhd                                                                                                                      
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap6.adb:1165 access check failed
Call stack traceback locations:
0x752e9e 0x7a3c7c 0x7750ba 0x773735 0x77cec5 0x76567c 0x743cab 0x735daa 0x7338e7 0x6494e8 0xb66c5d 0x642211 0x64e359 0x7a3f59 0x6406fd 0x7f2837102099 0x6445e8 0xfffffffffffffffe                                                                                                       
******************************************************************

Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/8.3.0/ghdl1 terminated by unhandled exception
raised CONSTRAINT_ERROR : trans-chap6.adb:1165 access check failed
Call stack traceback locations:
0x752e9e 0x7a3c7c 0x7750ba 0x773735 0x77cec5 0x76567c 0x743cab 0x735daa 0x7338e7 0x6494e8 0xb66c5d 0x642211 0x64e359 0x7a3f59 0x6406fd 0x7f2837102099 0x6445e8 0xfffffffffffffffe
```

tgingold, 2021-04-07:
I fear this is an issue due to the association with expression.

Note: `std_logic_vector'(a)` is a type qualification.  It should work with type conversion `std_logic_vector (a)`
(the quote is important).

abyszuk, 2021-04-07:
Hi @tgingold,
Yes, I can confirm it works with type conversion.
I can't understand how this is triggered only in presence of `sig'stable` attribute _anywhere_ in the code. Parsing error?

tgingold, 2021-04-08:
Now fixed.  It was an incorrect interaction between anonymous signals and implicit signals created for attributes.

commit 92745454 (2021-04-08): vhdl-sem_decls.adb: handle both anonymous signal and signal attribute.  Fix #1715
closed 2021-04-08 by tgingold