ghdl/ghdl #1717: Subtype on a access-type.

roynil, 2021-04-08:
**Description**
I have generic list defined in a generic package. Each list-object contains an access pointer to an element.

**Expected behaviour**
I have this code working in Questa simulator.

**How to reproduce?**

```vhd :file: generic_list_mwe_pkg.vhd
package generic_list_mwe_pkg is
  generic (
    type g_element_t);

  type elementp_t is access g_element_t;

  type g_list is protected

    impure function pull return g_element_t;

  end protected;

end package generic_list_mwe_pkg;

package body generic_list_mwe_pkg is

  type g_list is protected body
    type list_obj_t;
    type listp_t is access list_obj_t;

    type list_obj_t is record
      element : elementp_t;             -- pointer to element                                                                                                                                               
      nxt     : listp_t;                -- pointer to next list_obj                                                                                                                                         
    end record;

    variable list : listp_t;

    impure function pull
      return g_element_t is
      variable v_list : listp_t;
      variable v_elementp : elementp_t;
      impure function delete_and_return
        return g_element_t is
        variable ret : v_elementp.all'subtype;
      begin  -- function delete_and_return                                                                                                                                                                  
        ret := v_elementp.all;    -- Create a constrained copy of last element in list                                                                                                                      
        list := list.nxt;
        deallocate(v_list.element);
        deallocate(v_list);
        return ret;
      end function delete_and_return;
    begin
      assert list/=null report "Tried to pull element from empty list" severity error;
      v_list     := list;
      v_elementp := list.element;
      return delete_and_return;
    end function;
  end protected body;
      
end generic_list_mwe_pkg;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 package.vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-18.04-gcc.tgz`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
../src/generic_list_mwe_pkg.vhd:34:39:error: prefix must denote an object
        variable ret : v_elementp.all'subtype;
                                      ^
../src/generic_list_mwe_pkg.vhd:34:39:error: a type mark must be a simple or expanded name
        variable ret : v_elementp.all'subtype;
                                      ^
../src/generic_list_mwe_pkg.vhd:40:16:error: can't match 'ret' with type interface type "g_element_t"
        return ret;
               ^
/home/ronnil/ghdl-new/bin/ghdl:error: compilation error
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2021-04-09:
But instead of

```
  variable ret : v_elementp.all'subtype;
```

you could write:
```
  constant ret : g_element_t := v_elementp.all;
```

I will check if subtype attribute on a deference name is ok.

roynil, 2021-04-09:
Hi.

You are absolutely right. That compiles.

Regards,
Ronny

commit 2f908c24 (2021-04-11): vhdl-sem_names.adb: fix check for object prefix of subtype attribute.  Fix #1717
closed 2021-04-11 by tgingold