ghdl/ghdl #1718: CLI support for specifying library along with primary unit

GCHQDeveloper560, 2021-04-09:
**Is your feature request related to a problem? Please describe.**

Many tools allow running a top level design in a library using a dot notation like ``my_lib.my_top``. GHDL doesn't support this CLI syntax and instead requires something like
```
ghdl -e --work=my_lib --workdir=./my_lib my_top
```

**Describe the solution you'd like**

It would be simpler and more similar to other tools if GHDL supported
```
ghdl -e my_lib.my_top
```

**Additional context**
This was discussed by @m-kru and @tgingold in [an Edalize issue](https://github.com/olofk/edalize/issues/229).

m-kru, 2021-04-09:
I guess `my_lib` can imply `--work=my_lib`, however the question is whether it can also imply the `--workdir` value.

umarcor, 2021-04-09:
Overall, I think that the feature request is very nice. I want it! Yet, I'm afraid the proposal is not fully correct. As @m-kru said, see the following example:

```vhdl
entity ent is
end ent;

architecture arch of ent is
begin
process begin report "Hello"; wait; end process;
end;
```

```sh
# ghdl -a --work=my_lib ent.vhd 
# ghdl -e ent
error: cannot find entity or configuration ent
C:\msys64\mingw64\bin\ghdl.exe: compilation error
# ghdl -e --work=my_lib ent
```

That is, sources analyzed to specific libraries do not need to be located in separated directories. Those are different features. You can have one or multiple libraries and one or multiple locations for `*.cf` files (which is what `--workdir` is about).

So, I think it is correct to "*`my_lib` can imply `--work=my_lib`*", and leave the `--workdir` option aside (i.e. use `-P` for that).

commit 849a25e0 (2021-04-10): ghdldrv,configure: allow LIB.UNIT name for -e/-r commands.  Fix #1718
closed 2021-04-10 by tgingold