ghdl/ghdl #1720: Problems compiling UVVM examples with `llvm` and `gcc` back-ends

jeras, 2021-04-11:
**Description**

I tried to compile three UVVM examples which have scripts for building using GHDL.
I can compile them well using the `mcode` backend, but `llvm` and `gcc` back-ends fail for different reasons.

**How to reproduce?**

My OS is Ubuntu 20.04.

For UVVM I used [my branch](https://github.com/jeras/UVVM/tree/560395ff20aff7b330db167adec4235fd36e0f59), which only fixes an [issue with `-frelaxed` argument order](https://github.com/UVVM/UVVM/issues/137) compared to upstream.

For GHDL I tried to build with [nightly builds](https://github.com/ghdl/ghdl/releases/tag/nightly) with all three back-ends `mcode`/`lvvm`/`gcc`.
The `mcode` version is working well, the `llvm` version has problems with multiple object files for `transaction_pkg`, while the `gcc` version starts with a `Segmentation fault` in GHDL.
The choice of each back-end was made by setting the `PATH` to point at the extracted tarball of the nightly build.

I tried to build three UVVM examples, the only three which provide a GHDL build script.
* `bitvis_irqc`
* `bitvis_uart`
* `bitvis_vip_uart`
To execute an example do:
1. clone UVVM repo with my `-frelaxed` fix:
    ```
    git clone https://github.com/jeras/UVVM.git -b 560395ff20aff7b330db167adec4235fd36e0f59
    cd UVVM
    ```
2. `cd` to `example-name/scripts`,
3. setup `PATH` to desired back-end,
3. source the script.
   ```
   . ghdl_compile_all_and_simulate.sh
   ```

The results are:

| example | MCODE | LLVM | GCC |
|---|---|---|---|
| `bitvis_irqc` | works | works | Segmentation fault |
| `bitvis_uart` | works | multiple definition | Segmentation fault |
| `bitvis_vip_uart` | works | multiple definition | Segmentation fault |

Logs for all 9 combinations are attached, I have not double checked the log files, if the name and contents match.

[bitvis_irqc-ghdl-gha-ubuntu-20.04-gcc.log](https://github.com/ghdl/ghdl/files/6292379/bitvis_irqc-ghdl-gha-ubuntu-20.04-gcc.log)
[bitvis_irqc-ghdl-gha-ubuntu-20.04-llvm.log](https://github.com/ghdl/ghdl/files/6292380/bitvis_irqc-ghdl-gha-ubuntu-20.04-llvm.log)
[bitvis_irqc-ghdl-gha-ubuntu-20.04-mcode.log](https://github.com/ghdl/ghdl/files/6292381/bitvis_irqc-ghdl-gha-ubuntu-20.04-mcode.log)
[bitvis_uart-ghdl-gha-ubuntu-20.04-gcc.log](https://github.com/ghdl/ghdl/files/6292382/bitvis_uart-ghdl-gha-ubuntu-20.04-gcc.log)
[bitvis_uart-ghdl-gha-ubuntu-20.04-llvm.log](https://github.com/ghdl/ghdl/files/6292383/bitvis_uart-ghdl-gha-ubuntu-20.04-llvm.log)
[bitvis_uart-ghdl-gha-ubuntu-20.04-mcode.log](https://github.com/ghdl/ghdl/files/6292384/bitvis_uart-ghdl-gha-ubuntu-20.04-mcode.log)
[bitvis_vip_uart-ghdl-gha-ubuntu-20.04-gcc.log](https://github.com/ghdl/ghdl/files/6292385/bitvis_vip_uart-ghdl-gha-ubuntu-20.04-gcc.log)
[bitvis_vip_uart-ghdl-gha-ubuntu-20.04-llvm.log](https://github.com/ghdl/ghdl/files/6292386/bitvis_vip_uart-ghdl-gha-ubuntu-20.04-llvm.log)
[bitvis_vip_uart-ghdl-gha-ubuntu-20.04-mcode.log](https://github.com/ghdl/ghdl/files/6292387/bitvis_vip_uart-ghdl-gha-ubuntu-20.04-mcode.log)

umarcor, 2021-04-11:
Building UVVM with GHDL's vendor scripts is tested in [ghdl/extended-tests](https://github.com/ghdl/extended-tests). There was an issue with the compilation order files which Patrick reported some month ago. The fix was released a couple of days ago and we updated UVVM.

I can confirm the segmentation fault with GCC when using this approach: https://github.com/ghdl/extended-tests/runs/2317761456?check_suite_focus=true#step:4:18. However, building with LLVM is working in extended-tests. Hence, I guess you are getting the problem at runtime.

jeras, 2021-04-12:
The simplest failing example is `bitvis_uart` compiled with LLVM back-end.
It contains two distinct packages named `transaction_pkg` with each of them compiled in its own `work` library.

https://github.com/jeras/UVVM/blob/master/bitvis_uart/script/ghdl_compile_all_and_simulate.sh

In UVVM there is a `transaction_pkg.vhd` for each protocol VIP and they are compiled into `work` libraries with the VIP name (see script above):
```
./bitvis_vip_axi/src/transaction_pkg.vhd
./bitvis_vip_spi/src/transaction_pkg.vhd
./bitvis_vip_rgmii/src/transaction_pkg.vhd
./bitvis_vip_axilite/src/transaction_pkg.vhd
./bitvis_vip_i2c/src/transaction_pkg.vhd
./bitvis_vip_axistream/src/transaction_pkg.vhd
./bitvis_vip_gmii/src/transaction_pkg.vhd
./bitvis_vip_gpio/src/transaction_pkg.vhd
./bitvis_vip_uart/src/transaction_pkg.vhd
./bitvis_vip_sbi/src/transaction_pkg.vhd
./bitvis_vip_avalon_st/src/transaction_pkg.vhd
./bitvis_vip_ethernet/src/transaction_pkg.vhd
./bitvis_vip_avalon_mm/src/transaction_pkg.vhd
```

To me it seems they are compiled by GHDL into object files, then the linker complaints, since there are duplicate objects when it tries to link multiple object files. Some kind of separation for distinct work libraries seems to be missing in the LLVM build process.

The script for compiling UVVM you provide [`scripts/vendors/compile-uvvm.sh`](https://github.com/ghdl/ghdl/blob/master/scripts/vendors/compile-uvvm.sh) is a bit too structured for me to read it quickly, but it is probably doing something the script provided by UVVM does not.

If the difference is perhaps obvious to you, please point it out so I can try to update the UVVM example code and send them a pull request.

tmeissner, 2021-04-12:
@eine IIRC, wasn't there a problem with LLVM when files of same name compiled into the same library? I don't have the issue number at hand, but I remember such a problem.

jeras, 2021-04-12:
I found this https://github.com/ghdl/ghdl/issues/539 and this https://github.com/ghdl/ghdl/issues/1622.

jeras, 2021-04-12:
While the focus of issue #1622 is with same named files with unrelated contents, the similar issue I described here, has the extra problem, that the files have related content (contain entity or package with the same name). Due to related content, this files can only be compiled into separate `work` libraries.

I noticed, objects (functions) inside this object files already have the name of the `work` library used as a prefix.
The simplest fix would be to use this same prefix for object file names.