ghdl/ghdl #1721: PSL: Bugs in supported constructs

Blebowski, 2021-04-12:
**Description**
This bug report summarizes several bugs found in PSL implementation, in features which
are either supported, or should be supported by current state of the implementation
(no "unimplemented features" like parameter sequences or multiple clock assertions are reported).

**Expected behaviour**
Explained for each case below.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: top.vhd
entity top is
end entity;

architecture a of top is
    signal a,b,c,d : std_logic := '0';
    signal clk_sys, clk1, clk2 : std_logic;
begin

  -- Following throws bug occured with: "psl.sem_property: cannot handle N_CLOCKED_SER"
  -- Clocked SERE shall be allowed according to 6.1.1.1 of PSL LRM 2003
  -- psl my_seq : assert never {a;b;c} @ rising_edge(clk_sys);

  -- Following throws bug occured with: "build_sere_fa: cannot handle N_IMP_SEQ"
  -- This is strange because with "always" this is working. According to PSL LRM 2003
  -- FL_Property production for "always" is the same as for "never":
  --   FL_Property ::=
  --        always FL_Property
  --      | always Sequence
  --
  --   FL_Property ::=
  --        never FL_Property
  --      | never Sequence
  -- Therefore I think if sequence implication works with one, it shall work also with other.
  -- psl my_seq : assert never {a} |=> {b; c};

  -- Following error is thrown: "translate_psl_expr: cannot handle N_IMP_BOOL"
  -- Combination of "never" + implication is not very usefull, since implication is always
  -- true apart from case where 1 -> 0, therefore it will mostly fail, however, it should not
  -- crash tool (the same goes for previous case too)
  -- psl my_seq : assert never (a -> b);

  -- Following error is thrown:
  --  "top.vhd:43:28:error: operand of a 'never' operator must be a boolean or a sequence"
  -- While for "assert always", this works. IMHO this should work for both, equally since
  -- productions for never|always are the same.
  -- It is not true that operand of never must be a boolean or a sequence. It can be FL_property
  -- or sequence, which is much wider set.
  -- psl my_seq : assert never (a) -> eventually! (b);
  
  -- Following error is thrown: "top.vhd:43:58:error: left-hand side operand of logical '->' must be a boolean"
  -- This is not true, LHS and RHS of implication shall be FL_Property, not just boolean (6.2.1.6.1)
  -- psl my_seq : assert always (a -> next b -> next c -> next d);

end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a top.vhd
ghdl --elab-run top
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `d6a8f93b543ede48005000d90612060ef05cfd8c`

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tmeissner, 2021-04-12:
AFAIK some or all of your "bugs" are caused because GHDL implements the simple subset of PSL, as most of other tools.

Some of the rules:

* For `->`LHS operator has to be boolean
* For `eventually` operand must be boolean or sequence

etc. I think the simple subset is also defined in the LRM, I don't know where, as I don't have it at hand at the moment.

tmeissner, 2021-04-12:
I found the definition of the simple subset in my copy of the PSL 1.1 LRM in section 4.4:

> ...
> The simple subset of PSL is a subset that conforms to the notion of monotonic advancement of time, left to right through the property, which in turn ensures that properties within the subset can be simulated easily. The simple subset of PSL contains any PSL FL property meeting all of the following conditions:
>
> — The operand of a negation operator is a Boolean.
> — The operand of a never operator is a Boolean or a Sequence.
> — The operand of an eventually! operator is a Boolean or a Sequence.
> — The left-hand side operand of a logical and operator is a Boolean.
> — The left-hand side operand of a logical or operator is a Boolean.
> — The left-hand side operand of a logical implication (->) operator is a Boolean.
> — Both operands of a logical iff (<->) operator are Boolean.
> — The right-hand side operand of a non-overlapping until* operator is a Boolean.
> — Both operands of an overlapping until* operator are Boolean.
> — Both operands of a before* operator are Boolean.

umarcor, 2021-04-12:
> I really appreciate the ongoing discussions about PSL in GHDL. The more dynamic and interest in PSL the better.

Amen. Kudos to both Blebowski and you for that :wink:

Blebowski, 2021-04-13:
Hello @tmeissner, thanks for quick reply. I was trying to test various examples from [PSL cheat-sheet](https://uobdv.github.io/Design-Verification/Quick-References/PSL_quickref_vhdl.pdf), along with LRM,
that is how I got to this situation :)

I ran across the definition of simple sub-set in the standard, however, I did not find in GHDL documentation,
that GHDL supports only simple subset of PSL (I only saw "partial support"). Therefore, I assumed that it
attempts to implement everything, but it is unfinished. Now its clearer. Anyway, I would expect GHDL to
throw error like:  " <construct> is not allowed in simple subset of PSL. See <chapter number> of PSL LRM."
instead of crashing with GHDL bug occured...

I understand that trying to support full PSL might be infeasible, especially, if we would need to keep
track about past events (e.g. assert always ((a & next[3] b) -> c)), which requires storing "c" value from
first cycle for two more cycles when "b" is finally sampled, and only then compared. This clearly hampers
the notion of "linear time flow"...

However, operators of sequence implications ( |=>, |-> ) have LHS and RHS defined only on "sequence"
(not FL_property), which should be fully supported by "never" even in simple subset.
Therefore, I think it should not require that much more effort to "negate" the NFA, yielded by the implication
(or maybe I am wrong). RHS sequence already needs to "scan" for any condition making the sequence fail.
See e.g.
-- psl my_seq : assert always {a;b} |=> {c;d;c;d};
with NFA:
[image]
Tracking of all states in RHS is needed, so that anytime it does "not" match, NFA goes to fail state.

Also, when I try to run this example:
-- psl assert never {a;b} |=> {c;d;c;d};
in Aldec simulator, it is able to compile and simulate it.

Last but not least, IMHO following are equivalent:
-- psl my_seq : assert always {a} |=> {b;c;d};
-- psl my_seq : assert always (a -> next b -> next c -> next d);
First one is supported by GHDL, second is not. I can simulate both in Aldec. Final NFA (in principle similar to the one above) will be the same, therefore I don't really see a reason why both should not simulate (although the second one
corrupts rules of simple subset).

Question is, whether it is easy to recognize "property which corrupts linear time flow" inside the code...
E.g. how to distuguish property above from:
-- psl my_seq : assert always (((a -> next b) -> eventually c) -> next d);
Which would be really crazy to implement...

tmeissner, 2021-04-13:
Nice graphical representation of the NFA smiley

The restriction to the simple subset is somehow hidden in the documentation of GHDL's PSL support: https://ghdl.github.io/ghdl/using/ImplementationOfVHDL.html#id2 It should be more prominent.

Reading the LRM very strict, a combination of two sequences via implication operators isn't a sequence anymore, but a property. So, using it with never would violate the simple subset.

The simple subset is an arbitrarily restriction is some ways, I agree. But it's a restriction which has a clear definition. So before we implement stuff which goes beyond the simple subset, in my opinion, we maybe should try to complete support of the subset itself. However, I'm not strictly against such things which goes beyond the simple subset. I assume that Aldec implements some easy implementable extras beyond the simple subset.

Blebowski, 2021-04-13:
About adding more extensions, I agree that first full simple subset is needed, and extending beyond
only then. OTOH, I think all of the cases mentioned should throw error saying: Unsupported due to
simple subset anyway... (instead of crashing)

NFA vizualization:
I use [Graphwiz online](https://dreampuf.github.io/GraphvizOnline/)
and feed results of "Disp_NFA" there on custom build.

"Eventually", I understand as if it was left associative (just like you described it)
So the meaning of:
-- psl assert always (a -> next b -> eventually! c -> next d);
in free speech is like:
If there is a in cycle 1, then check there is b in cycle 2.
If b is not in cycle 2, then abort the check (as LHS of implication is 0, therefore whole implication will hold).
If b is in cycle 2, then at some future time, c must hold.
If c does not hold at some future cycle, then assertion fails at end of simulation.
If yes (c holds in e.g. in cycle 10), then check that d holds in cycle 11.

Is it correct, or do you think that "d" should then hold at cycle 2. In other words, does the last "next" correspond
to time of starting the whole LHS, or to time of its completion?

tmeissner, 2021-04-13:
Mhm, tricky. I read it in that way that all RHS from the first implication has to hold. So, if b isn't valid in the 2nd. cycle, I would expect the assertion to fail. I really has to look up in my PSL book if if my assumption is right.

Blebowski, 2021-04-13:
Thats interesting point, if it is like you say, then

a -> next b -> next c -> next d

 is somehow equivalent to 

> {a} |=> {b;c;d}

If it is not, then, intermediate steps like "b" and "c" can be like dont-cares, only creating preconditions of
final check "d".

As always, the devil is in the details :D Pity (or maybe good), that there is no "PSL compliance testing",
that would resolve the discussion.

tmeissner, 2021-04-13:
Hehe, I currently read in my PSL book. One tip under "Common errors" is:

> Don’t apply the never operator to a logical implication or to a suffix implication.
 
And:
`always a -> next b -> next c` is equivalent to `always (a -> ((next b) -> (next c)))` because of operator precedence and associativity. So, `b` and `c` has to be valid in the same 2nd cycle. It is NOT equivalent to `{a; b; b}`. These complex meanings are one of the reasons why I mostly use SERE instead of LTL operators. LTL operators are very powerful and seemingly easy to read, but often one read them wrong.

Blebowski, 2021-04-13:
Hi, thanks, for explanation, I have to buy the book. I use SEREs mostly too.
So in free speech it is equivalent to:
If a holds in cycle 1, then check that if b holds in cycle 2, c must also hold in cycle 2, so:

>assert always {a} |=> {not (b and (not c))};

?
since this is illegal syntax:

> {a} |=> {b -> c}

because internals of "{ }" should be only SERE, not FL_Property.

Blebowski, 2021-04-13:
OK, so if I may summarize it, the outcome of the discussion is:
1. assert never {a;b;c} @ rising_edge(clk) is TBD.
2. Invalid situations due to simple subset, could throw more explanatory error (instead of crashing).
3. **Wishlist** (but not important now) : Handle assert never {a} |=> {b} and {a} |-> {b} ?

Does that make sense?

umarcor, 2021-04-13:
> that there is no "PSL compliance testing", that would resolve the discussion.

😊[vhdl/Compliance-Tests](https://github.com/vhdl/Compliance-Tests)

There is no specific test suite for PSL. But I believe that would be the place if someone wanted to drop some PSL related test cases, using the LRM as a reference.

tmeissner, 2021-04-13:
Yeah, something like that (as I understand). They claim, that `always a -> next b -> next[2] (c)` works as intended by the user, instead of `always a -> next b -> next c`. But both aren't equal to `{a; b; c}`.

The book "a practical introduction to PSL" is a very good one. It is clear, has good examples and tips, and was written by two engineers who worked at IBM on the Sugar and later PSL language.

tgingold, 2021-04-14:
The issues are now fixed.  Thank you for the report.
We might still have to improve the error message (maybe with a tag like 'simple subset').

I think it is always better to create multiple issues when there are multiple problems (or crash).  It is simply easier to discuss.

closed 2021-04-14 by tgingold

Blebowski, 2021-04-14:
@tgingold thanks.
I will try separating everything I find into stand-alone issue.