ghdl/ghdl #1722: Got raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2046 while synthesizing a simple fifo

jeandet, 2021-04-13:
**Description**
I'm trying to synthesize a FIFO with ghdl-yosys-plugin for an ice40 device. I don't use explicitly any RAM cell but GHDL+Yosys seems to detect one from an array of ```std_logic_vector```.

**Expected behaviour**
I would expect it to work or find a way to just use FPGA R-cells as a workaround.

**How to reproduce?**

I use:
```
podman run  --rm -t -v .:/src -w /src hdlc/ghdl:yosys  yosys -v 9 -m ghdl -p 'ghdl --std=08 fifo.vhd -e fifo; synth_ice40 -json fifo.json'
```

```vhd :file: fifo.vhd
library ieee ;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity fifo is
generic(
    DEPTH    : integer := 256;
    WDTH     : integer := 16;
    TRESHOLD : integer := 32
);
port(
    reset   : in std_logic;
    clk     : in std_logic;

    data_in    : in std_logic_vector(WDTH-1 downto 0);
    data_out   : out std_logic_vector(WDTH-1 downto 0);
    empty      : out std_logic;
    full       : out std_logic;
    half_full  : out std_logic;
    wr         : in std_logic;
    rd         : in std_logic
);
end entity;

architecture ar_fifo of fifo is
    type rcells_memory_t is array(0 to DEPTH-1) of std_logic_vector(WDTH-1 downto 0);
    signal rcells_memory : rcells_memory_t;
    signal write_ptr  : integer range 0 to DEPTH-1 :=0;
    signal read_ptr   : integer range 0 to DEPTH-1 :=0;
    signal data_count : integer :=0;
    signal full_sig   : std_logic :='0';
    signal empty_sig   : std_logic :='0';
begin

    empty <= empty_sig;
    full  <= full_sig;
    half_full  <= '1' when data_count >= TRESHOLD else '0';
    empty_sig <= '1' when data_count = 0 else '0';
    full_sig <= '1' when data_count = DEPTH-1 else '0';

    process(clk, reset)
    begin
        if reset = '0' then
            data_count <= 0;
        elsif clk'event and clk = '1' then
            if wr = '1' and rd = '0'  and data_count /= DEPTH then
                data_count <= data_count + 1;
            elsif rd = '1' and wr = '0' and data_count /= 0 then                
                data_count <= data_count - 1;
            end if; 
        end if; 
    end process;

    process(clk, reset)
    begin
        if reset = '0' then
            write_ptr <= 0;
        elsif clk'event and clk = '1' then
            if wr = '1' and full_sig = '0' then
                write_ptr <= (write_ptr + 1) mod (DEPTH-1);
            end if; 
        end if; 
    end process;
    rcells_memory(write_ptr) <= data_in;

    process(clk, reset)
    begin
        if reset = '0' then
            read_ptr <= 0;
        elsif clk'event and clk = '1' then
            if rd = '1' and empty_sig = '0' then
                read_ptr <= (read_ptr + 1) mod (DEPTH-1);
            end if; 
        end if; 
    end process;
    data_out <= rcells_memory(read_ptr);

end architecture ar_fifo;
```

**Context**
Please, provide the following information:

- OS: 
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

The current GHDL Yosys docker image ID is **c8f0e7331611** but I have this issue for few month now. 

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-163-gd6a8f93b) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/src/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2046
******************************************************************

```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tmeissner, 2021-04-13:
I assume that the bug roots in your async write to the memory array. I moved the write into the write pointer process and synthesis runs without error.

```vhdl
process(clk, reset)
begin
  if reset = '0' then
    write_ptr <= 0;
  elsif clk'event and clk = '1' then
    if wr = '1' and full_sig = '0' then
      write_ptr <= (write_ptr + 1) mod (DEPTH-1);
      rcells_memory(write_ptr) <= data_in;
    end if;
  end if;
end process;
 --rcells_memory(write_ptr) <= data_in;
```

jeandet, 2021-04-14:
@tmeissner, thank you for your help! That was one my guesses but I was too lazy to make it work with sync ram ;). Is there a way to force GHDL+Yosys to use register cells instead hard macro such as RAM cells or multipliers? Usually this is not what one would what but might help for debug purposes.

umarcor, 2021-04-14:
@jeandet, in the Tip of https://ghdl.github.io/ghdl/using/Synthesis.html#synthesis-options there is a reference to https://github.com/ghdl/ghdl/blob/master/src/ghdldrv/ghdlsynth.adb#L210-L215. Maybe `-dm2` will work for you?

tmeissner, 2021-04-14:
I you left the memory read async, you get a non-BRAM implementation. I assume that the target FPGA platforms BRAM only allows sync write & read. You can also advice yosys to not use BRAMs during the [`synth_ice40`](http://www.clifford.at/yosys/cmd_synth_ice40.html) command: the `-nobram` option is the way to go.

jeandet, 2021-04-14:
@tmeissner , @umarcor, Thank you, -dm2 seems to work with my code, I'll also investigate the synchronous option ASAP.

tmeissner, 2021-04-14:
BTW: Your use of the `mod` operator seems to be pretty expensive. I have a similar design using this function for increment the pointers:

```vhdl
  function incr_pnt (data : t_fifo_pnt) return t_fifo_pnt is
  begin
    if (data = t_fifo_mem'high) then
      return 0;
    end if;
    return data + 1;
  end function incr_pnt;
```
 getting this resource summary:
```
   Number of cells:                112
     SB_CARRY                       26
     SB_DFFER                       17
     SB_DFFES                        1
     SB_DFFR                         2
     SB_LUT4                        65
     SB_RAM40_4K                     1
```
using `mod` instead I get following ressources:
```
   Number of cells:                264
     SB_CARRY                      130
     SB_DFFER                       17
     SB_DFFES                        1
     SB_DFFR                         2
     SB_LUT4                       113
     SB_RAM40_4K                     1
```
It would be interesting to know, if vendor synthesize tools has similar behavior.

dalex78, 2026-08-25:
# Issue #1722 — crash synthesizing a FIFO with an asynchronous memory write

Analysis by an AI agent.

## High-level summary

A FIFO whose memory array is written by a plain concurrent — unclocked — signal assignment, `rcells_memory(write_ptr) <= data_in;` outside any process, with `write_ptr` updated in a separate clocked process, crashes `ghdl --synth`:

```
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2046   (as reported, 2021)
raised ADA.ASSERTIONS.ASSERTION_ERROR   : netlists.adb:886             (on master today)
```

tmeissner identified the trigger on the thread within hours — *"I assume that the bug roots in your async write to the memory array"* — and moving the write into the clocked process makes it synthesize. His closing line is the actual request: *"Nevertheless, GHDL shouldn't crash with your code."*

## What is wrong in GHDL's implementation

Memory inference runs `Validate_RAM_Multiple` (`src/synth/netlists-memories.adb`), which calls `Validate_RAM_Simple` for each write branch. `Validate_RAM_Simple` walks the chain from the write gate forward to the `Signal`/`Isignal` gate, stepping over `Dff`/`Idff`/`Mdff`/`Midff` gates on the way — but it never requires one to be there. A chain that goes straight from `Dyn_Insert` to the signal, with no register at all, validates just as happily.

So `Convert_To_Memory` proceeds and `Create_RAM_Ports` builds an `Id_Mem_Wr_Sync` gate for the write, with its clock input left as `No_Net`: the code that connects a clock only runs when a `Dff`-family gate was actually found. Later, `Netlists.Disp_Vhdl.Disp_Memory` builds the memory process's sensitivity list and does

```ada
S := Get_Input_Net (Port_Inst, 2);            --  the clock input
S := Get_Input_Net (Get_Net_Parent (S), 0);   --  strip the edge  -> No_Net
```

and `Get_Net_Parent (No_Net)` is the assertion that fires.

There is no asynchronous write-port gate to route it to instead: the gate catalogue (`src/synth/netlists-gates.ads`) has both `Id_Mem_Rd` and `Id_Mem_Rd_Sync` for reads, but only `Id_Mem_Wr_Sync` for writes. Nor would one be right here. With no register in the write chain, the elements that `write_ptr` does not select would have to combinationally hold their previous value — a feedback loop, not an asynchronous write port. The pattern is simply not memory-shaped hardware.

## What the fix does

`Validate_RAM_Simple` now tracks whether the chain passes through a register, and rejects the signal as a RAM candidate if it reaches the signal gate without having seen one — the same `No_Instance` result it already returns for every other pattern it declines.

The array is then synthesized the way any other combinationally-fed-back array signal is, as ordinary `dyn_insert`/`dyn_extract` logic. On the report's FIFO, `rcells_memory` comes out as a plain 4096-bit vector with per-element mux logic, `--synth` completes, and the misleading `found RAM "rcells_memory", width: 16 bits, depth: 256` note is gone.

## Testing

`testsuite/synth/issue1722/` synthesizes the reporter's FIFO unchanged and asserts that a netlist is produced, that no assertion or bug box appears, and that no RAM is inferred. It fails on every backend without the fix (memory inference lives in `src/synth`, which all three share) and passes with it.