ghdl/ghdl #1723: GHDL Bug occured

m-kru, 2021-04-13:
**Expected behaviour**
Print an error message if having nested arrays with different length is not supported. However, it crashes even if the length of the nested arrays is the same.

**How to reproduce?**

```vhd :file: my_pkg.vhd
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

package my_pkg is

   type my_vector is array (natural range <>) of integer;
   type my_vector_vector is array (natural range <>) of my_vector;

   constant C_1 : my_vector(1 downto 0) := (others => 1);
   constant C_2 : my_vector(2 downto 0) := (others => 2);

   constant C_FOO : my_vector_vector(1 downto 0) := (
      0 => C_1,
      1 => C_2
   );

end package my_pkg;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 my_pkg.vhd
```

**Context**

- OS: Linux arch 5.11.6
- GHDL info:
```
GHDL 2.0.0-dev (1.0.0.r163.gd6a8f93b) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r163.gd6a8f93b) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/home/mkru/workspace/VHDL_examples/foo/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 --workdir=work -P/usr/local/lib/ghdl/vendors/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o work/my_pkg.o my_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x5590528c7336 0x5590529c32c5 0x5590529c4133 0x5590529c6bd7 0x5590529c4f5d 0x5590529c5271 0x559052a0097a 0x559052a01984 0x559052a00d7c 0x559052a0350f 0x559052a035e2 0x559052a37501 0x559052a3793e 0x559052a37a50 0x559052a40e6a 0x559052a2ba25 0x559052a2b253 0x559052a2b313 0x559052a139a5 0x559052a76d02 0x559052a7885c 0x559052a7a0fb 0x55905281e85e 0x7f5c3aca4b23 0x55905281d99c 0xfffffffffffffffe
******************************************************************
```

diogratia, 2021-04-14:
IEEE Std 1076-2008

6.4.2.2 Constant declarations  

>A constant declaration declares a _constant_ of the specified type. Such a constant is an _explicitly declared constant_.  
>  
>constant_declaration ::=  
&nbsp;&nbsp;&nbsp;&nbsp;**constant** identifier_list : subtype_indication [ := expression ] ;  
>  
>If the assignment symbol “:=” followed by an expression is present in a constant declaration, the expression specifies the value of the constant; the type of the expression shall be that of the constant. The value of a constant cannot be modified after the declaration is elaborated.

14.4.2 Elaboration of a declaration  
14.4.2.5 Object declarations  

> The rules of this subclause apply only to explicitly declared objects (see 6.4.2.1). Generic declarations, port declarations, and other interface declarations are elaborated as described in 14.3.2 through 14.3.5 and 14.6.  
>  
>Elaboration of an object declaration that declares an object other than a file object or an object of a protected type proceeds as follows:  
>>a) The subtype indication is first elaborated; this establishes the subtype of the object.  
>>b) If the object declaration includes an explicit initialization expression, then the initial value of the object is obtained by evaluating the expression. It is an error if the value of the expression does not belong to the subtype of the object; if the object is a composite object, then an implicit subtype con- version is first performed on the value unless the object is a constant whose subtype indication denotes an unconstrained type. Otherwise, any implicit initial value for the object is determined.  
>>c) The object is created.  
>>d) Any initial value is assigned to the object.  
>  
>The initialization of such an object (either the declared object or one of its subelements) involves a check that the initial value belongs to the subtype of the object. For a composite object declared by an object declaration, an implicit subtype conversion is first applied as for an assignment statement, unless the object is a constant whose subtype is an unconstrained type.

The bug shows up prior to elaboration during analysis.

5.3.2 Array types  
5.3.2.1 General  

> An array object is a composite object consisting of elements that have the same subtype. 

9.3.3.3 Array aggregates  

>**_For an aggregate of a one-dimensional array type, each choice shall specify values of the index type, and the expression of each element association shall be of either the element type or the type of the aggregate._** If the type of the expression of an element association is the type of the aggregate, then either the element association shall be positional or the choice shall be a discrete range.

(emphasis added)
 
Checking compliance with this -2008 semantic requirement would be missing or deficient  in commit d6a8f93b.

JimLewis, 2021-05-15:
VHDL-2008 does not support ragged arrays as you have tried to do.

However, you can implement ragged arrays by having the outer array be an array of pointers to this simply by the outer array being an array of pointers to the inner array.   But then you will have to manage the allocation.

commit 75b92a00 (2021-08-06): vhdl-sem_expr.adb: check matching subtype of array aggregate elements.

When the subtype of the aggregate is not known by the context.
Fix #1723
closed 2021-08-06 by tgingold