ghdl/ghdl #1724: Constant not supported in PSL vunit

GCHQDeveloper560, 2021-04-14:
**Description**
While signals appear to work correctly in PSL vunits, attempts to use constants result in an error

**Expected behaviour**
Expected constants to work like signals, but instead:
```
sem_psl_verification_unit: cannot handle IIR_KIND_CONSTANT_DECLARATION (const_test_vunit.vhd:7:14)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-163-gd6a8f93b) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/jksmith-home/ghdl/formal/constant_error/
Command line:
/opt/fpga/ghdl-nightly-llvm/bin/ghdl1-llvm --std=08 -P/opt/fpga/ghdl-nightly-llvm/lib/ghdl/ieee/v08/ -P/opt/fpga/ghdl-nightly-llvm/lib/ghdl/ -c -fpic -o const_test_vunit.o const_test_vunit.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x4ab475 0x535e95 0x57bc34 0x53211e 0x64303b 0x644dea 0x646454 0x4091f1 0x7f1f4e509bf5 0x4084d8 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

**How to reproduce?**

```vhd :file: const_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity const_test is
    generic (
        addr_width : positive := 5
    );
    port (
        clk : in std_logic
    );
end entity;

architecture rtl of const_test is
begin
end architecture rtl;
```

```vhd :file: const_test_vunit.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

vunit const_test_vunit (const_test(rtl))
{
    constant depth : positive := 2**addr_width;
}
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 const_test.vhd const_test_vunit.vhd
```

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04
- Origin:
  - GHDL Nightly: `https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-18.04-llvm.tgz`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
sem_psl_verification_unit: cannot handle IIR_KIND_CONSTANT_DECLARATION (const_test_vunit.vhd:7:14)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-170-g2acfe88a) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/ghdl/constant_error/
Command line:
/opt/ghdl-nightly-llvm/bin/ghdl1-llvm --std=08 -P/opt/ghdl-nightly-llvm/lib/ghdl/ieee/v08/ -P/opt/ghdl-nightly-llvm/lib/ghdl/ -c -fpic -o const_test_vunit.o const_test_vunit.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x4ab475 0x535ecf 0x57bc6e 0x53211e 0x643209 0x644fb8 0x646622 0x4091f1 0x7fe765054bf5 0x4084d8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2021-04-15:
I am fixing this crash, but currently vunit cannot be simulated.

GCHQDeveloper560, 2021-04-15:
Thank you! I'm certainly a PSL beginner and have been experimenting with assertions in the code vs in submodules vs in a vunit. I've mostly been experimenting with SymbiYosys, so the synthesis restriction is okay.

It also appears that attempting to declare a type isn't supported in a vunit. Would you prefer a separate issue for this?
```
sem_psl_verification_unit: cannot handle IIR_KIND_TYPE_DECLARATION (type_test_vunit.vhd:6:10)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-170-g2acfe88a) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/ghdl/formal/type_error/
Command line:
/opt/fpga/ghdl-nightly-llvm/bin/ghdl1-llvm --std=08 -P/opt/fpga/ghdl-nightly-llvm/lib/ghdl/ieee/v08/ -P/opt/fpga/ghdl-nightly-llvm/lib/ghdl/ -c -fpic -o type_test_vunit.o type_test_vunit.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x4ab475 0x535ecf 0x57bc6e 0x53211e 0x643209 0x644fb8 0x646622 0x4091f1 0x7faa05db2bf5 0x4084d8 0xfffffffffffffffe
******************************************************************
```

umarcor, 2021-04-15:
> Would you prefer a separate issue for this?

Yes, please.

commit df9f4b18 (2021-04-15): vhdl: handle constant declarations in PSL vunit.  Fix #1724