ghdl/ghdl #1725: Vendor libraries don't compile on linux

stfarley, 2021-04-15:
**Description**
In Linux the vendor library compilation scripts don't work

**Expected behaviour**
Succesful compile

**How to reproduce?**
Run any compilation script eg,

./compile-osvvm.sh --all
Loading environment...
/tools/ghdl_gcc/1.0.0/lib/ghdl/vendors/shared.sh: line 71: declare: -n: invalid option
declare: usage: declare [-aAfFgilrtux] [-p] [name[=value] ...]

**Context**

- OS:Centos 7
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [2acfe88 ] Built from sources: `commit SHA`

Paebbels, 2021-04-15:
What Bash version is used to execute the shell scripts. Looks like Bash is outdated.

stfarley, 2021-04-15:
Bash version is 4.2.46

stfarley, 2021-04-15:
FYI after upgrading bash on my Centos 7 machine to 5.0 the scripts seem to work properly.
Thanks for the quick response Paebbels

Xiretza, 2021-04-15:
For the record, namerefs were added in bash 4.3: http://git.savannah.gnu.org/cgit/bash.git/tree/NEWS?id=f3a35a2d601a55f337f8ca02a541f8c033682247#n665