ghdl/ghdl #1731: Resizing signed produces IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_SGN_SGN when synthesizing

umarcor, 2021-04-21:
Resizing signed seems not to be supported for synthesis yet:

```vhd
--:file: fifo.vhd
library ieee;
context ieee.ieee_std_context;

entity fifo is
  generic (
    data_width : positive := 8;
    fifo_depth : positive := 8
  );
  port (
    clkw : in std_logic;
    clkr : in std_logic;
    rst  : in std_logic;
    wr   : in std_logic;
    rd   : in std_logic;
    d    : in std_logic_vector(data_width-1 downto 0);
    e    : out std_logic;
    f    : out std_logic;
    q    : out std_logic_vector(data_width-1 downto 0)
  );
end fifo;

architecture arch of fifo is

  type fifo_t is array (0 to 2**fifo_depth-1)
    of std_logic_vector(data_width-1 downto 0);
  signal mem : fifo_t;

  signal rdp, wrp : unsigned(fifo_depth downto 0);

begin

-- Assertions
  process(clkw, clkr)
    constant dx : std_logic_vector(d'left downto 0) := (others => 'X');
    constant du : std_logic_vector(d'left downto 0) := (others => 'U');
  begin
    if rising_edge(clkw) then
      if ( wr and ( d?=dx or d?=du ) ) then
        assert false report "wrote X|U to fIfO" severity failure;
      end if;
      if (f and wr) then
        assert false report "wrote to fIfO while full" severity failure;
      end if;
    end if;
    if rising_edge(clkr) then
      if (e and rd) then
        assert false report "Read from fIfO while empty" severity failure;
      end if;
    end if;
  end process;

--

  process(clkw) begin
    if rising_edge(clkw) then
      if wr then
        mem(to_integer(wrp(fifo_depth-1 downto 0))) <= d;
      end if;
    end if;
  end process;

  process(clkw) begin
    if rising_edge(clkw) then
      if rst then
        wrp <= (others => '0');
      else
        if wr then
          wrp <= wrp+1;
        end if;
      end if;
    end if;
  end process;

  f <= rdp(fifo_depth-1 downto 0)?=wrp(fifo_depth-1 downto 0)
       and (rdp(fifo_depth) xor wrp(fifo_depth));
  e <= rdp ?= wrp;

  process(clkr) begin
    if rising_edge(clkr) then
      if rst then
        q <= (others => '0');
      elsif rd then
        q <= mem(to_integer(rdp(fifo_depth-1 downto 0)));
      end if;
    end if;
  end process;

  process(clkr) begin
    if rising_edge(clkr) then
      if rst then
        rdp <= (others => '0');
      else
        if rd then rdp <= rdp+1; end if;
      end if;
    end if;
  end process;

end arch;
```

```vhdl
--:file: axis_conv1d9x1.vhd
library ieee;
context ieee.ieee_std_context;

entity axis_conv1d9x1 is
  generic (
    fifo_depth : integer := 0 -- ceiling of the log base 2 of the desired FIFO length
  );
  port (
    s_axis_clk   : in  std_logic;
    s_axis_rstn  : in  std_logic;
    s_axis_rdy   : out std_logic;
    s_axis_data  : in  std_logic_vector(31 downto 0);
    s_axis_valid : in  std_logic;
    s_axis_strb  : in  std_logic_vector(3 downto 0);
    s_axis_last  : in  std_logic;

    m_axis_clk   : in  std_logic;
    m_axis_rstn  : in  std_logic;
    m_axis_valid : out std_logic;
    m_axis_data  : out std_logic_vector(31 downto 0);
    m_axis_rdy   : in  std_logic;
    m_axis_strb  : out std_logic_vector(3 downto 0);
    m_axis_last  : out std_logic
  );
end axis_conv1d9x1;

architecture arch of axis_conv1d9x1 is

  constant data_width : integer := 32;

  signal r, e, f, wr, rd, valid : std_logic;
  signal d, q : std_logic_vector(31 downto 0);

  signal acca, accb, acca_n, accb_n : signed(31 downto 0);
  signal rlast : std_logic;

begin

  r <= (s_axis_rstn nand m_axis_rstn);

  fifo: entity work.fifo
    generic map (
      fifo_depth => fifo_depth,
      data_width => 32
    )
    port map (
      CLKW => s_axis_clk,
      CLKR => m_axis_clk,
      RST => r,
      WR => wr,
      RD => rd,
      E => e,
      F => f,
      D => d,
      Q => q
    );

  process(s_axis_clk) begin
    if rising_edge(s_axis_clk) then
      if not s_axis_rstn then
        acca <= (others=>'0');
        accb <= (others=>'0');
      elsif s_axis_valid and s_axis_rdy then
        acca <= acca_n;
        accb <= accb_n;
      end if;
    end if;
  end process;

  acca_n <=
    acca + signed(s_axis_data(15 downto 0)) when rlast='0' else
    resize(signed(s_axis_data(15 downto 0)), acca) when ??(s_axis_valid and s_axis_rdy) else
    (others=>'0');
  accb_n <=
    accb + signed(s_axis_data(31 downto 16)) when rlast='0' else
    resize(signed(s_axis_data(31 downto 16)), acca) when ??(s_axis_valid and s_axis_rdy) else
    (others=>'0');

  process(s_axis_clk) begin
    if rising_edge(s_axis_clk) then
      if not s_axis_rstn then
        rlast <= '0';
      else
        rlast <= s_axis_last and (s_axis_valid and s_axis_rdy);
      end if;
    end if;
  end process;

  wr <= rlast;
  d <= std_logic_vector(accb(18 downto 3)) & std_logic_vector(acca(18 downto 3));

-- AXI4 Stream Slave logic

  s_axis_rdy <= not f;

-- AXI4 Stream Master logic

  rd <= (not e) and (valid nand (not m_axis_rdy));

  process(m_axis_clk) begin
    if rising_edge(m_axis_clk) then
      if ((not m_axis_rstn) or ((valid and E) and m_axis_rdy))='1' then
        valid <= '0';
      elsif rd then
        valid <= '1';
      end if;
    end if;
  end process;

  m_axis_valid <= valid;
  m_axis_last <= q(d'left);
  m_axis_strb <= (others=>'1');
  m_axis_data <= q(data_width-1 downto 0);

end architecture;
```

```shell
# ghdl synth --std=08 --out=raw-vhdl -gfifo_depth=3 fifo.vhd axis_conv1d9x1.vhd -e axis_conv1d9x1
axis_conv1d9x1.vhd:72:5: unhandled function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_SGN_SGN
axis_conv1d9x1.vhd:76:5: unhandled function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_SGN_SGN
fifo.vhd:26:10:note: found RAM "mem", width: 32 bits, depth: 8
```

Tested on MSYS2:

```
# ghdl version
GHDL 2.0.0-dev (1.0.0.r122.g1c17875a) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I can work around it by not using resize, since the dimensions are hardcoded anyway. Yet, rezising would come handy.

tmeissner, 2021-04-21:
If I read the sources correctly, `function RESIZE (ARG : UNRESOLVED_SIGNED; NEW_SIZE : NATURAL) return UNRESOLVED_SIGNED` is supported, but not `function RESIZE (ARG, SIZE_RES : UNRESOLVED_SIGNED) return UNRESOLVED_SIGNED`.

However, I doubt if your use is valid for synthesis, as the size parameter isn't constant. If I cast the second parameter to `integer` GHDL complains about the non-constant size parameter:

```
resize.vhd:38:5:error: size must be constant
resize.vhd:43:5:error: size must be constant
```

tmeissner, 2021-04-21:
I added the two left `resize()` functions in a branch of GHDL: b0f7e64196c8c884d8f7d17b5d448eddd5b0a4c0

But I wonder how `signed` and values outside ot the `integer'range` should be handled. Furthermore I get a crash when I use something like `resize(a, to_signed(12, 7))`.

umarcor, 2021-04-21:
@tmeissner, in this case, I can use a generic intead of `acca`, because the length of `acca` is defined through a generic. That works ok.
However, I wanted to also try using the target signal as the argument to the resize. My understanding is that it should be constant too. Because neither the size nor the dimension of `acca` are variable, everything is fixed.

tgingold, 2021-04-21:
Did you forget a `'length` ?

closed 2021-04-21 by tgingold

umarcor, 2021-04-21:
@tgingold, as Torsten pointed out, there are two functions in the library:

https://github.com/ghdl/ghdl/blob/45f80d929899d8c16754736eb5cecd508f3ec85b/libraries/ieee2008/numeric_std.vhdl#L1104-L1110

https://github.com/ghdl/ghdl/blob/45f80d929899d8c16754736eb5cecd508f3ec85b/libraries/ieee2008/numeric_std.vhdl#L1123-L1124

So, my understanding is that `resize(signed(s_axis_data(31 downto 16)), acca'length)` and `resize(signed(s_axis_data(31 downto 16)), acca)` should be equivalent. Even though the content/value of the signal is not fixed, `acca'length` is constant (not type constant). I don't see where is the risk of "having a variable number of wires" coming from. Maybe the overload for the resize is not defined for synthesis (as Torsten suggested)?

tgingold, 2021-04-21:
Indeed, my mistake.

umarcor, 2021-04-21:
I created this minimal reproducer, both for signed and unsigned:

```vhdl
library ieee;
context ieee.ieee_std_context;

entity ent is
  port (
    dsin  : in signed(15 downto 0);
    dsout : out signed(31 downto 0);
    duin  : in unsigned(15 downto 0);
    duout : out unsigned(31 downto 0)
  );
end axis_conv1d9x1;

architecture arch of ent is

begin

  dsout <= resize(signed(dsin), dsout);
  duout <= resize(unsigned(duin), duout);

end architecture;
```

```sh
ghdl synth --std=08 --out=raw-vhdl ent.vhd -e ent
```

tmeissner, 2021-04-21:
Uhhhh, I find the `function RESIZE (ARG, SIZE_RES : UNRESOLVED_SIGNED) return UNRESOLVED_SIGNED` function very misleading. I thought the value of size would be the resulting length.

umarcor, 2021-04-22:
I can confirm that the issue is fixed in artifacts of `da51617` for MSYS2. Tristan, Torsten, thank you so much for another astonishing fast response/fix.

commit 37f4aea4 (2021-04-21): synth-vhdl_oper.adb: handle resize sgn/sgn.  Fix #1731

With an hint from T.Meissner