ghdl/ghdl #1742: GHDL-mcode: exception raised: stack overflow

bahm-lgs, 2021-04-23:
**Description**
I was just trying to run a simple UVVM testbench with GHDL for one of my IPcores developped for Vivado. I am using the UVVM VCCs for axilite and axistream. Due to Xilinx not providing VHDL behavioral models for latest Xilinx Fifo Generator cores anymore I had to use structural models for the FIFOs of my core and had to compile the Xilinx Unisim library.

As a first test I made my testbench program some axilite registers (which works fine) and afterwards issue an axistream transaction. When the last transaction is simulated GHDL crashes with the message
`exception raised: stack overflow`

This happens only with the Windows mcode version of GHDL 2.0.0-dev (1.0.0.r163.gd6a8f93b). **With the 1.0.0 Linux release everything seems to work fine and the simulation finishes**.

**Expected behaviour**
Normal simulation execution without crash.

**How to reproduce?**
Next week I will try to provide a minimal working example to reproduce, I still have to remove some unsharable code and strip the example down as much as possible.
**Context**
Please, provide the following information:

- OS: Windows10 1909 64bit
- Origin: Released binaries: nightly build 2.0.0-dev (1.0.0.r163.gd6a8f93b).

**Additional context**
I would appreciate if someone could give me instructions on how to get more debug information from ghdl instead of just the stack overflow exception.

umarcor, 2021-04-23:
I don't know if `ulimit` (https://ghdl.github.io/ghdl-cosim/vhpidirect/notebook/mistakes.html) can help here...

For debugging, you need to build your simulartion with `-g`. That needs to be done with either LLVM or GCC backends. Then, you run the simulation on gdb. Is the exception reproducible with the MINGW64 version (LLVM), instead of the MINGW32 (mcode)?

See https://ghdl.github.io/ghdl/development/Debugging.html#gnu-debugger-gdb.

bahm-lgs, 2021-04-23:
> I don't know if `ulimit` (https://ghdl.github.io/ghdl-cosim/vhpidirect/notebook/mistakes.html) can help here...

Uh, I thought ulimit is a Linux-only thing?

> For debugging, you need to build your simulartion with `-g`. That needs to be done with either LLVM or GCC backends. Then, you run the simulation on gdb. Is the exception reproducible with the MINGW64 version (LLVM), instead of the MINGW32 (mcode)?

I still cannot build on Windows. All our company PCs have SentinelOne installed which keeps on deleting mingw from my disk. I will have to wait for our IT to get that fixed, will takes some days, then I'll give it a try.

Thanks

umarcor, 2021-04-23:
> Uh, I thought ulimit is a Linux-only thing?

Indeed... my bad.

bahm-lgs, 2021-04-26:
Decreasing the stack limit under Linux (with "ulimit") from the default 8MB to 4MB resulted in "Segmentation fault".
Increasing the stack limit under Windows (by patching ghdl.exe using VisualStudio's "editbin" command) from the default 1MB to 8MB makes my simulation complete successfully.

closed 2021-04-26 by bahm-lgs

umarcor, 2021-04-26:
@bahm-lgs would you mind providing any reference about how to patch ghdl with editbin? I'd like to document how to do it.

Moreover, maybe that should be a configuration/build option? So, instead of patching an existing binary, users can built it with the desired stack size.

bahm-lgs, 2021-04-26:
> @bahm-lgs would you mind providing any reference about how to patch ghdl with editbin? I'd like to document how to do it.

@umarcor , it's quite straightforward:
- in VisualStudio open Tools --> Visual Studio Developer Command Prompt
- run the command `editbin /stack:8388608 <ghdl-bin-path>/ghdl.exe`

> Moreover, maybe that should be a configuration/build option? So, instead of patching an existing binary, users can built it with the desired stack size.

Maybe, but I'm not sure on how to do that for the mcode version.

umarcor, 2021-04-27:
@tgingold, any thought?

tgingold, 2021-04-27:
There is the `--stack` option for the GNU linker:

https://sourceware.org/binutils/docs/ld/Options.html#Options-Specific-to-i386-PE-Targets

bahm-lgs, 2021-04-27:
> There is the `--stack` option for the GNU linker: https://sourceware.org/binutils/docs/ld/Options.html#Options-Specific-to-i386-PE-Targets

Thanks, I will give it a try.

Just in case someone else might want to reproduce the issue, here is a stripped-down minimal example:
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_axistream;
use bitvis_vip_axistream.vvc_methods_pkg.all;
use bitvis_vip_axistream.axistream_bfm_pkg.all;

entity axistream_uvvm_tb is
end entity axistream_uvvm_tb;

architecture func of axistream_uvvm_tb is

    constant C_TB_SCOPE                 : string    := "TESTBENCH";
    constant C_CLK_PERIOD               : time      := 10 ns;
    constant GC_VCCI_STREAMGEN          : natural   := 0;
    constant GC_STREAMGEN_DATA_WIDTH    : natural   := 16;
    constant GC_STREAMGEN_USER_WIDTH    : natural   := 3;
    constant GC_STREAMGEN_ID_WIDTH      : natural   := 0;
    constant GC_STREAMGEN_DEST_WIDTH    : natural   := 0;

    signal clk                          : std_logic  := '0';

    signal axistream_vvc_master_if      : t_axistream_if(
                                            tdata(GC_STREAMGEN_DATA_WIDTH-1 downto 0),
                                            tkeep((GC_STREAMGEN_DATA_WIDTH/8)-1 downto 0),
                                            tuser(GC_STREAMGEN_USER_WIDTH -1 downto 0),
                                            tstrb((GC_STREAMGEN_DATA_WIDTH/8)-1 downto 0),
                                            tid(GC_STREAMGEN_ID_WIDTH-1 downto 0),
                                            tdest(GC_STREAMGEN_DEST_WIDTH-1 downto 0)
                                        ) := init_axistream_if_signals(true,
                                            GC_STREAMGEN_DATA_WIDTH,
                                            GC_STREAMGEN_USER_WIDTH,
                                            GC_STREAMGEN_ID_WIDTH,
                                            GC_STREAMGEN_DEST_WIDTH
                                        );
begin

    i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;
    p_clk: clk <= not clk after C_CLK_PERIOD;

    i_axistream_vvc_master : entity bitvis_vip_axistream.axistream_vvc
    generic map(
        GC_VVC_IS_MASTER    => true,
        GC_DATA_WIDTH       => GC_STREAMGEN_DATA_WIDTH,
        GC_USER_WIDTH       => GC_STREAMGEN_USER_WIDTH,
        GC_ID_WIDTH         => GC_STREAMGEN_ID_WIDTH,
        GC_DEST_WIDTH       => GC_STREAMGEN_DEST_WIDTH,
        GC_INSTANCE_IDX     => GC_VCCI_STREAMGEN
    )
    port map(
        clk                 => clk,
        axistream_vvc_if    => axistream_vvc_master_if
    );

    p_main: process
        variable s_data     :   t_slv_array(0 to 1)(GC_STREAMGEN_DATA_WIDTH-1 downto 0);
        variable s_user     :   t_user_array(0 to 1);
    begin
        await_uvvm_initialization(VOID);

        s_data := (x"0000", x"FFFF");
        s_user := (x"01", x"02");
        axistream_transmit(AXISTREAM_VVCT, GC_VCCI_STREAMGEN, s_data, s_user, "Send data", C_TB_SCOPE);

        std.env.stop;
        wait;

    end process p_main;

end func;
```