ghdl/ghdl #1745: GHDL compilation failure

vdkeybus, 2021-04-26:
Regularly, I check out the latest version of GHDL and build it.

Currently I'm facing a compile issue:

`clang++ -c `llvm-config --cxxflags` -g -o llvm-cbindings.o src/ortho/llvm6/llvm-cbindings.cpp
src/ortho/llvm6/llvm-cbindings.cpp:1257:20: error: call to non-static member function without an object argument
         DebugLoc::get(DebugCurrentLine, 0, DebugCurrentScope),
         ~~~~~~~~~~^~~
src/ortho/llvm6/llvm-cbindings.cpp:1600:18: error: call to non-static member function without an object argument
      (DebugLoc::get(DebugCurrentLine, 0, DebugCurrentScope));
       ~~~~~~~~~~^~~
src/ortho/llvm6/llvm-cbindings.cpp:1615:20: error: call to non-static member function without an object argument
         DebugLoc::get(DebugCurrentLine, 0, DebugCurrentSubprg),
         ~~~~~~~~~~^~~
src/ortho/llvm6/llvm-cbindings.cpp:2762:18: error: call to non-static member function without an object argument
      (DebugLoc::get(DebugCurrentLine, 0, DebugCurrentScope));
       ~~~~~~~~~~^~~
4 errors generated.
make[1]: *** [src/ortho/llvm6/Makefile:20: llvm-cbindings.o] Error 1
make[1]: Leaving directory '/home/vdkeybus/Src/ghdl'
make: *** [Makefile:360: ghdl1-llvm] Error 2
`

I'd like to know where to start fixing this. Should I look into downgrading my tools or is this a genuine bug ?

Thansk

Xiretza, 2021-04-26:
This is an LLVM 12 incompatibility, caused by https://github.com/llvm/llvm-project/commit/41c3b271399229976d9885233db79ebbbe0383c6.

vdkeybus, 2021-04-26:
Submitted a fix in #1746

closed 2021-04-27 by tgingold