ghdl/ghdl #1748: Handling of unallocated line

avelure, 2021-04-27:
**Description**
When the simulator encounters a variable of `line` type that is used before it is allocated, it exits with error code 3, but without any error message to what caused the fault.

**Expected behaviour**
Something like a segfault message with the file and line number that caused it would be useful for tracking down the problem.

**How to reproduce?**
Following is an MWE. I provide two examples as the first will print the `....(report note):` before exiting, while the second exits before, they were from different parts of my code and I don't know if it is the same cause.
Uncommenting the `write(` line makes it all work.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity ent is
end entity ent;
architecture beh of ent is
begin
  process
    variable v_line : line;
    procedure test_print_1(
      constant msg : in string
    ) is
    begin
      report msg;
      report "Test 1 pass";
    end procedure;
    procedure test_print_2 is
      constant v_msg : string := v_line.all(1 to v_line'length - 1);
    begin
      report v_msg;
      report "Test 2 pass";
    end procedure;
  begin
    --write(v_line, string'(""));
    test_print_1(v_line.all);
    test_print_2;
    report "Test passed";
    wait;
  end process;
end architecture beh;
```

```sh :image: ghdl/ghdl:buster-llvm
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent
```

**Context**
Please, provide the following information:

- OS: Win 10
- Origin:
  - [x] Released binaries: `Nightly LLVM GHDL 2.0.0-dev (1.0.0.r163.gd6a8f93b)`

diogratia, 2021-04-27:
the only part of the context clause you need is the use clause for std.textio.
```none
Macbook: ghdl -a --std=08 ent.vhdl
Macbook: ghdl -e --std=08 ent
Macbook: ghdl -r ent
ent.vhdl:17:7:@0ms:(report note): Macbook:
```
Note the empty string printed out in the report statement without removing the comment delimiter from 
`--write(v_line, string'(""));`.

```none
 Macbook: ghdl --version
GHDL 2.0.0-dev (1.0.0.r163.gd6a8f93b) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Macbook:
```
This is from a daily build download. Presumably compiled with llvm-11.

The variables would be the platform, tool versions or build.

avelure, 2021-04-28:
Thanks for investigating, I updated to remove the extra context.

I did a quick test on some dockers with the Ubuntu nightly releases for `GHDL 2.0.0-dev (v1.0.0-116-g9ab4fd18) [Dunoon edition]`
LLVM and GCC version has same printout as on the mac (ie. it prints nothing).
Mcode gives a stack trace
```
ent.vhd:13:7:@0ms:(report note):
Execution of ghdl terminated by unhandled exception
raised CONSTRAINT_ERROR : grt-errors.adb:115 access check failed
Call stack traceback locations:
0x55aa340c6f11 0x55aa3411b750 0x55aa3411bbfb 0x7f6a906cc20e
```

umarcor, 2021-04-28:
@avelure, note that we provide multiple containers for testing purposes: https://hub.docker.com/r/ghdl/run/tags?page=1&ordering=last_updated. All of those (Debian, Ubuntu, Fedora...) are equivalent to the nightly packages, but you avoid having to download, and extract/install them. In fact, the images are generated in https://github.com/ghdl/docker. Buster images are triggered by the CI workflows in this repo: https://github.com/ghdl/docker#-buster-3-jobs--max-3--6-images-triggered-by-ghdlghdl-nightly-job. Others are updated weekly: https://github.com/ghdl/docker#-test-15-jobs--max-3--30-images-weekly.

avelure, 2021-04-28:
I knew there were official docker scripts, but I did not know about the docker hub containers. Thanks, I'll keep it in mind for later.