ghdl/ghdl #1751: Force/release of alias to signal

avelure, 2021-05-03:
**Description**
Forcing or releasing an alias to a signal gives an error `target (alias "alias_name") is not a signal`.
From what I understand of the LRM (6.6) an alias to a signal should be equivalent to a signal.

**Expected behaviour**
No error reported.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
end entity ent;

architecture beh of ent is
  signal sig_1 : std_logic;
  alias a_sig_1 is sig_1;
begin
  process
  begin
    a_sig_1 <= force '1';
    a_sig_1 <= release;
    wait;
  end process;
end architecture beh;
```

```sh :image: ghdl/ghdl:buster-llvm
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent
```
Output:
```
ent.vhd:14:5: target (alias "a_sig_1") is not a signal
    a_sig_1 <= force '1';
    ^
ent.vhd:15:5: target (alias "a_sig_1") is not a signal
    a_sig_1 <= release;
    ^
```

**Context**

- OS: win10
- Origin: Released binaries: Nightly build `2.0.0-dev (1.0.0.r163.gd6a8f93b)`

tgingold, 2021-05-04:
I will have a look but currently force/release are not supported.

avelure, 2021-05-06:
Ah, I see, thanks.
By support I assume you mean from the development side, as it seems it is supported from the software side looking at some older issues, and in fact testing without alias works.

tgingold, 2021-05-06:
I am fixing this issue: aliases will be correctly handled, but force/release not yet...

commit 09f3b6ad (2021-05-06): vhdl-sem_stmts: handle aliases for force/release assignment.  Fix #1751
closed 2021-05-06 by tgingold