ghdl/ghdl #1753: [pyGHDL] help appreciated

LeFl0w, 2021-05-07:
Hi everyone, 
I try to use libGDHL feature to perform request to the AST of my elaborated VHDL design but I might admit I'm quite lost.

I started with [this post](https://github.com/ghdl/ghdl/issues/1156) which is pretty much what I want to try.
I got to [the comment](https://github.com/ghdl/ghdl/issues/1156#issuecomment-597239513) stating that  `vhdl-nodes.ads` is the kind of specification of everything and this might be addressed with libghdl and its python binding.

I found this [old repository](https://ohwr.org/project/vhdl-style) which might be of some help but I think is outdated because it reference a `configure --enable-python` option for ghdl which doesn't exist anymore. I supposed it evolved as `pyGHDL`.

Therfore I installed pyGHDL with command `pip3 install .` in the ghdl source folder.
I can launch python and import the pyGHDL library with command `import pyGHDL.libghdl as libghdl`
Then I'm quite lost and cannot find how to load an elaborated design and iterate through the AST.

I saw [the testfile](https://github.com/ghdl/ghdl/blob/master/testsuite/pyunit/libghdl/Initialize.py) but it seems to work on a single file and not on an elaborated design.

Any help would be appreciated.
thanks.
Florent

tgingold, 2021-05-07:
No, currently there is no elaborated design available in libGHDL.

That's something that will arrive, but not yet.

LeFl0w, 2021-05-09:
Thanks for the feedback. I'll keep trying with a single file.

closed 2021-05-09 by LeFl0w

umarcor, 2021-05-09:
@LeFl0w see label [Interface: Python](https://github.com/ghdl/ghdl/labels/Interface%3A%20Python):

- https://ghdl.github.io/ghdl/quick_start/python/index.html
- #1557 (#1556)
  - #1553 ([vhdl/pyVHDLModel](https://github.com/vhdl/pyVHDLModel))
  - #1590
- #1492 (#1437)
  - #1671
- #1519 (follow-up of #111)

LeFl0w, 2021-05-17:
Thanx a lot for these information.
If I do understand correctly:
* `pyghdl/libghdl`  will implement direct access to libghdl and the hdl-nodes.ads . Rough examples are located in : `testsuite/pyunit/libghdl/initialize.py` and show how to access architecture name and entity name.
* `pyghdl/dom` implement higher access to item like entity,package, configuration  names  with associated examples in `testsuite/pyunit/dom/SimpleEntity.py`

If I am correct only access to this names are already implemented into to python API as examples). If we want (for example) a signal instanciation name in the architecture body part , this will have to be developped and added to the libghdl part of the pyGHDL.

Paebbels, 2021-05-17:
I think only iterating structures of GHDL via `libghdl` is needed, to fill out the fields in `dom`.

umarcor, 2021-05-19:
Hi Florent! Let me be direct: do you have "some time" to devote into making `pyGHDL.dom` usable? If that is the case, "we are almost there". If that is not the case, you/we will need to wait until someone can allocate some time.

Plase, see #1449. That is something I did in Aug-Sep last year. Back then, pyGHDL did not exist. libghdl was available only. I wrote an example that took a file and listed the entities and architectures in it. For the entities, it would show the list of ports: name, mode and type. If the type was `std_logic_vector`, it also showed the range (limits and direction). See https://github.com/umarcor/ghdl/tree/9a2181109880a172a87fa318bd4be95d703734a2/testsuite/python/units01. `show_units.py` was the existing test by Tristan, and `show_ports` is the version I wrote on top of that. As you can see in #1449, I stopped there because we considered that features such as https://github.com/umarcor/ghdl/blob/9a2181109880a172a87fa318bd4be95d703734a2/testsuite/python/units01/show_ports.py#L18-L57 should be better written in a reusable module.

In Dec-January this year, we did the pyGHDL rework. We created a "proper" Python package with submodules, and Patrick split pyVHDLModel from pyVHDLParser, in order to use it as the missing abstraction layer between libghdl and the show_ports example. Unfortunately, those pyGHDL changes broke the show_ports example, and we did not have time for focusing on fixing it according to the new sources.

Hence, if you are willing to contribute, I suggest we start with reproducing the show_ports example with pyGHDL.dom. Then, it should be trivial to extend it for iterating over the signals in the architecture, similarly to the iteration over the ports.

umarcor, 2021-05-19:
> this will have to be developped and added to the libghdl part of the pyGHDL.

pyGHDL.libghdl is "complete" because most of it is auto-generated. The missing part is to convert that unpythonic (Ada/C alike) Python sources, into the more Pythonic pyGHDL.dom.

LeFl0w, 2021-06-01:
Hi, thanks for pointed me to these example. I will try to use them with pyGHDL.dom .
I'm not sure I will succeed as I'm more like a FPGA designer than a software/python one. We'll see

umarcor, 2021-06-01:
@LeFl0w, the vast majority of participants in this repo/project are more of hardware (RTL) designers than software/python people. That's precisely why advancing in these features is hard for us. We know what we have, we know what we want to achieve, but, as Donald Knuth said (emphasis added by me):

> https://en.wikiquote.org/wiki/Donald_Knuth
> In fact, my main conclusion after spending ten years of my life working on the TEX project is that **software is hard. It’s harder than anything else I’ve ever had to do**.

Therefore, take it easy, and don't be overwhelmed. Let's try to move one little step at a time. Do not hesitate to ask if you don't understand something/anything. Talking about a problem/challenge helps you, me and others understand the complexity around it. If only, that alone is a contribution to the project.

LeFl0w, 2021-06-09:
I tried to do a simple test based on the libghdl example about entity and architecture listing:    
   - display the location of the entity in the original file. All the source code are available on [github](https://github.com/LeFl0w/Test_pyGHDL)

First I used the dump ast command to see where I was going : `ghdl -a -dp adder.vhd`
 Here is an extract below
```
  chain: [854]
  library_unit: entity_declaration "adder" [844]
    location: adder.vhd:4:8
    identifier: "adder" [1046]
    has_begin: false
    visible_flag: false
    is_within_flag: false
    end_has_reserved_id: false
    end_has_identifier: true
    parent: design_unit "adder" [836]
```

In the example we are looking to a node called *entity_declaration* with  `nodes.Get_Kind(libraryUnit) == nodes.Iir_Kind.Entity_Declaration:`
Which I think is supposed to be the one above

I added the command `print(nodes.Get_Location(libraryUnit))` based on `vhdl-nodes.ads` I should be okay: 
```
 -------------------------------------------------
   -- General methods (can be used on all nodes): --
   -------------------------------------------------

   --  Get the location of the node: ie the current position in the source
   --  file when the node was created.  This is a little bit fuzzy.
   --
```

The command is accepted by python by the return is `58` instead of the expected `4`

Did I miss something? Is there any king of transfert fonction like  `Get_Name_Ptr to ` be applied ? Or maybe the line is the line in the AST?
Thx

tgingold, 2021-06-09:
The location is an internal number.  To extract file, line or column from a location, use functions from files_map.
(First get the file, then from file and location get the line and then from file, line and location get the column).

The file is an identifier (NameId), so you need to call Get_Name_Ptr to get the corresponding file name as a string.

Many things are represented by numbers in ghdl.  That works well for strong typed language, much less for not so strongly
typed languages like python!

LeFl0w, 2021-06-10:
thanks for the tips.  This is definitivelly working:
```
  name=getIdentifier(libraryUnit)

  #internal AST conversion tables up to line
  loc = nodes.Get_Location(libraryUnit)
  fil = files_map.Location_To_File(loc)
  pos = files_map.Location_File_To_Pos(loc, fil)
  line = files_map.Location_File_To_Line(loc, fil)
  col = files_map.Location_File_Line_To_Offset(loc, fil, line)

  print("name:"+ str(getIdentifier(libraryUnit)) + " line "+ str(getNodeLineInFile(libraryUnit)) + " column "+str(getNodeColumInFile(libraryUnit)))
```