ghdl/ghdl #1754: Synth of generic array of std_logic_vector in record leads to crash

siesse, 2021-05-09:
**Description**
The ghdl --synth crashes on the following MWE. 

**Expected behaviour**
No crash :-)

**How to reproduce?**
```vhd :file: my_entity.vhd
library ieee;
use ieee.std_logic_1164.all;

entity my_entity is
end;

architecture behaviour of my_entity is
    type std_logic_vector_array_t is array (natural range <>) of std_logic_vector;
    type reg_t is record
        x : std_logic_vector_array_t(15 downto 0)(15 downto 0);
    end record;

begin

end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 my_entity.vhd -e my_entity
```

**Context**
Please, provide the following information:

- OS: Windows 10
- Origin:
  - [ ] Package manager: `version`
  - [*] Released binaries: `https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst`
  - [ ] Built from sources: `c69c9c51c9235835bb876e4373bb4c784cf7c96d`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

** Bugreport**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r257.g82665d42) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: i686-w64-mingw32
C:\Temp\
Command line:
C:\GHDL\nightly-mingw32-mcode\bin\ghdl.exe --synth --std=08 my_entity.vhd -e my_entity
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-vhdl_context.adb:234 access check failed
Call stack traceback locations:
0x855081 0x8553e2 0x854527 0x85469a 0x854c89 0x856da2 0x856ef6 0x871d10 0x871ec2 0x86b1d2 0x899b46 0x7d6472 0x93b978 0x5d2cd6 0x5d1394 0x765d6357 0x77158962 0x77158932
******************************************************************
```

**Additional context**
Note that this code simulates perfectly with the same ghdl version. 
Hopefully this is not a pure duplicate, i love to contribute a bit on this wonderful project.

dalex78, 2026-08-20:
# Issue #1754 — Crash: generic array of `std_logic_vector` in a record

Analysis by an AI agent.

## High-level summary

A record field declared as a constrained array of unconstrained `std_logic_vector` elements (`type std_logic_vector_array_t is array (natural range <>) of std_logic_vector; type reg_t is record x : std_logic_vector_array_t(15 downto 0)(15 downto 0); end record;` — merely declared, never instantiated or used) crashed `ghdl --synth` with an internal `CONSTRAINT_ERROR` at `synth-vhdl_context.adb:234`.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth --std=08 my_entity.vhd -e my_entity` now produces a clean (empty, since the type is never used) netlist instead of crashing. This must have been fixed as part of general improvements to nested-unconstrained-array-of-vectors type handling in `src/synth` since this was reported (originally against a 2021 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1754/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this ticket be closed ? (PR merged)

tgingold, 2026-08-24:
Yes indeed.  Thank you for your patience, I was away for a few days.

closed 2026-08-24 by tgingold

dalex78, 2026-08-24:
> Yes indeed. Thank you for your patience, I was away for a few days.

Thank you for this project !