ghdl/ghdl #1756: GHDL and GTKwave install ghwdump

Paebbels, 2021-05-09:
**Description**
Normally, GTKwave was installing `ghwdump`. Now, also GHDL brings `ghwdump`.

In Arch Linux, a package collision is reported, because both packages install the same file.  
According to @umarcor this also applies to MSYS2.

**Expected behavior**
Both packages should be installable in parallel.

**How to reproduce?**

**Context**
- OS: Arch Linux

umarcor, 2021-05-09:
That is correct. On MSYS2, GTKWave does install ghwdump by default (along with some other helpers: https://github.com/gtkwave/gtkwave/tree/master/gtkwave3-gtk3/src/helpers). Now, GHDL builds it by default too, because it is used in the testsuite.

GTKWave does not use libghw, which was added very recently. It has ghw sources vendored in the codebase. Therefore, at a given point in time, it is not guaranteed that GTKWave and GHDL are using exactly the same source. However, since GHDL is the only known GHW generator and GTKWave is the only known GHW reader, we might consider such situations a coordination issue between GHDL and GTKWave to be solved ASAP.

I see two solutions:

- Rename the binary installed by either GTKWave or GHDL, for avoiding the name collision and for allowing potentially different versions to coexist. ghwdump is a very small binary, so size duplication is not relevant.
- Distribute libghw and ghwdump separatedly from GHDL. On MSYS2, we can build everything together and then create two packages. At the same time, remove ghwdump from the GTKWave package. Then, make GHDL and GTKWave packages dependent on the ghw package. This would allow users of GTKWave only or GHDL only to get ghwdump, or to install both tools without conflicts. However, strictly, GTKWave might be internally using a version of GHW sources that does not match the ghwdump built by GHDL.

/cc @lazka for feedback about building an additional package containing just one binary and one DLL.
/cc @gtkwave for getting his point of view.
/cc @FFY00 @marzoul @Xiretza as Arch/AUR users/packagers.

gtkwave, 2021-05-10:
Most gtkwave users probably don't care about ghwdump unless they have GHW files to debug...which in pretty much nearly all cases would mean GHDL is already installed. I can either remove this (which seems to make more sense) or rename it to something like ghwdump_gtkwave or something similar.

-Tony

marzoul, 2021-05-10:
For now, I made the ArchLinux AUR packages ghdl-*-git not install the ghwdump binary (the package can still perform checks with the non-installed binary).

If main development of ghw lib migrates to GTKWave repo, it would be fine to have the ghdl package require the package gtkwave as a check-only dependency. ghdl developers may as well want to have checks with other waveform formats in the future, so the other waveform tools in the GTKWave repo may also become useful.

Suggestion of @umarcor also makes sense, to have a split package : separate the waveform viewer (which depends on gtk etc), and the waveform creation/manipulation libraries and binaries (these don't depend on gtk). And the make ghdl package (and code) explicitly depend on those libs.
But, this may not be worth the effort if the only advantage is to have a ghdl package that is not pulling the gtk dependency.

Xiretza, 2021-05-10:
> * Distribute libghw and ghwdump separatedly from GHDL. On MSYS2, we can build everything together and then create two packages. At the same time, remove ghwdump from the GTKWave package. Then, make [...]

The extension of this would be to develop and distribute libghw as an actual independent library, and have GTKWave link against it instead of vendoring the sources. The packages would then look like this:

* libghw provides libghw.so
* either libghw provides ghwdump as an example implementation, or ghwdump is a separate package that depends on libghw
* gtkwave depends on libghw.so at buildtime and runtime (if GHW functionality is desired in the package)
* ghdl depends on libghw/ghwdump for testing only

@gtkwave, would that be an option for you or do you want to avoid another library dependency?

gtkwave, 2021-05-10:
It makes sense from a software development point of view, but there are probably a dozen Linux distributions that package gtkwave. I don't have the time to hunt all users down. I could simply force breakage to ferret them out, but that's not the way to do things.

I think the best way to do it would be a phased in approach: libghw if found, otherwise fallback to the sources found in gtkwave. Eventually the sources are removed. I wouldn't do such a thing unless Tristan himself split off a reader library. (Writing, as far as I know, is done all in Ada code inside GHDL.) As far as I know, gtkwave has been the only user since 2005 so this might not be worth the effort.

-Tony

tgingold, 2021-05-11:
Yes, I think it is not worth the effort.  The ghw library is just a single file.

The easiest would be indeed to not install ghwdump in gtkwave (as GHDL users already have it).

umarcor, 2021-05-14:
As both Tony and Tristan agreed, I proceeded with temporarily patching the MSYS2 package upstream, and I submited a PR for removing ghwdump from GTKWave. Tony, there is no rush for merging or tagging a new GTKWave release on my side. We can wait. 

@marzoul @Xiretza, while I do agree with your vision, I do also think it's not worth the effort yet. In the mid term, I'd like to explore the possibility of building a "waveform toolkit", by combining GTKWave's helpers plus libghw, plus some bindings (say Python, Rust...). That would justify the creation of a new package since it would be usable standalone. However, that's just a desirable idea for now. I would need time for properly studying how to split the "backend" from GTKWave.