ghdl/ghdl #1757: Choice subtype is not locally static error within case statement

treforsouthwell, 2021-05-13:
**Description**
The code below fails to compile

 ./ghdl -a -fsynopsys testm.vhd
testm.vhd:43:9:error: choice subtype is not locally static
   case (fred(1 downto 0)) is

**Expected behaviour**
The code compiles fine with Cadence XRun, Synopsys VCS and Synopsys Design Compiler

**How to reproduce?**
See code below:
 ./ghdl -a -fsynopsys testm.vhd

testm.vhd:

entity testm is

port (clk : in std_logic;
      data : in std_logic_vector(2+1 downto 0);
      q1 : out std_logic_vector(3 downto 0)
);
end testm;

architecture rtl of testm  is

-- img_log2 function
   function tlog2(d : positive) return natural is
      variable tmp : positive;
   begin
      tmp := 1;
      for count in 0 to d loop 
         if (tmp >= d) then
            return count;
         end if;
         tmp := tmp*2;
      end loop;
      return d;
   end;

constant SBITS : integer := tlog2(16);

signal fred : std_logic_vector(SBITS - 1 downto 0);

begin

fred <= data;

process (fred)
begin
   case (fred(1 downto 0)) is
        when "00" =>
         q1 <= data;

        when "01" =>
         q1 <= "0000";

        when "10" =>
         q1 <= data;

        when others  =>
         q1 <= "1111";

   end case;
end process;

end rtl;

tgingold, 2021-05-13:
I think GHDL is correct.  You are analyzing using the 93 revision of the VHDL standard and the restrictions of section 8.8 applies.

Use `--std=08` to select the 2008 revision which relaxes the restrictions and accept your design.

treforsouthwell, 2021-05-14:
Hi, thanks for getting back to me, that seemed to work.

It's interested that the commerical vendors build this code in vhdl 93 mode, maybe they aren't so strick.

closed 2021-05-14 by treforsouthwell

Paebbels, 2021-05-14:
@treforsouthwell Check if your commercial tool offers a `--pedantic` switch and have fun :).