ghdl/ghdl #1758: Writing multiple reals to a binary file produces a different output from writing a real_vector

umarcor, 2021-05-14:
I would expect the following tests to produce the same binary output. In both cases, 10 real values are to be saved. The first test declares a file of type `real` ; `write()` is executed 10 times. In the second test, the file is declared as `real_vector(0 to 9)`; write is executed once.

```vhdl
--:file: tb_real.vhd

entity tb_real is
end entity;

library ieee;
context ieee.ieee_std_context;

architecture arch of tb_real is
begin

  p_main: process
    variable y: real;

    type real_file is file of real;
    file fptr : real_file;
    variable fstat : file_open_status;
  begin

    file_open(fstat, fptr, "tb_real.bin", write_mode);
    assert fstat = open_ok report "file_open failed" severity failure;

    y := 0.0;

    for k in 0 to 9 loop
      write(fptr, y);
      y := y + 1.2;
    end loop;

    file_close(fptr);
    wait;
  end process;

end arch;
```

```vhdl
--:file: tb_real_vector.vhd

entity tb_real_vector is
end entity;

library ieee;
context ieee.ieee_std_context;

architecture arch of tb_real_vector is
begin

  p_main: process
    variable y: real;

    subtype y_vec is real_vector(0 to 9);
    variable yv : y_vec;

    type realvec_file is file of y_vec;
    file fptr : realvec_file;
    variable fstat : file_open_status;
  begin

    y := 0.0;

    for k in 0 to 9 loop
      yv(k) := y;
      y := y + 1.2;
    end loop;

    file_open(fstat, fptr, "tb_real_vector.bin", write_mode);
    write(fptr, yv);

    file_close(fptr);
    wait;
  end process;

end arch;
```

```vhdl
#!/bin/sh

set -e

cd $(dirname $0)

ghdl -a --std=08 ./*.vhd

ghdl --elab-run --std=08 tb_real
ghdl --elab-run --std=08 tb_real_vector

printf "\nod tb_real:\n"
od -tfD tb_real.bin

printf "\nod tb_real_vector:\n"
od -tfD tb_real_vector.bin

printf "\nhexdump tb_real:\n"
hexdump tb_real.bin

printf "\nhexdump tb_real_vector:\n"
hexdump tb_real_vector.bin

#:file: run.sh
```

To my surprise, the output is very different:

```
# ./run.sh 

od tb_real:
0000000                        0                      1.2
0000020                      2.4       3.5999999999999996
0000040                      4.8                        6
0000060                      7.2                      8.4
0000100                      9.6       10.799999999999999
0000120

od tb_real_vector:
0000000    8.107188373263167e+44   3.1729656390642953e+59
0000020    8.569253938161024e+30             3.29655e-318
0000040    4.667261455498039e-62   4.6672614591174805e-62
0000060   -9.255963124396812e+61     4.66726145912314e-62
0000100             2.07469e-317   -9.255963143559087e+61
0000120   -9.255963143560489e+61   4.6672614591275615e-62
0000140  -2.353437366349645e-185            2.077017e-317
0000153

hexdump tb_real:
0000000 0000 0000 0000 0000 3333 3333 3333 3ff3
0000010 3333 3333 3333 4003 cccc cccc cccc 400c
0000020 3333 3333 3333 4013 0000 0000 0000 4018
0000030 cccd cccc cccc 401c cccd cccc cccc 4020
0000040 3333 3333 3333 4023 9999 9999 9999 4025
0000050

hexdump tb_real_vector:
0000000 4723 4448 2d4c 4942 414e 5952 462d 4c49
0000010 2d45 2e30 0a30 465b 2e5d 000a 0000 0000
0000020 0000 3300 3333 3333 f333 333f 3333 3333
0000030 0333 cc40 cccc cccc 0ccc 3340 3333 3333
0000040 1333 0040 0000 0000 1800 cd40 cccc cccc
0000050 1ccc cd40 cccc cccc 20cc 3340 3333 3333
0000060 2333 9940 9999 9999 2599 0040
000006b
```

The binary file produced by the first test has the expected content: 10*8=80bytes. However, `tb_real_vector.bin` has 27 additional bytes. In fact, removing the first 27 bytes achieves the expected content:

```
# hexdump -s 27 tb_real_vector.bin

hexdump tb_real_vector:
000001b 0000 0000 0000 0000 3333 3333 3333 3ff3
000002b 3333 3333 3333 4003 cccc cccc cccc 400c
000003b 3333 3333 3333 4013 0000 0000 0000 4018
000004b cccd cccc cccc 401c cccd cccc cccc 4020
000005b 3333 3333 3333 4023 9999 9999 9999 4025
000006b
```

tgingold, 2021-05-14:
Yes, that's correct.

The file content is not defined by the LRM, and GHDL only write raw values for scalar types.

umarcor, 2021-05-14:
1. Did I already ask you about this either publicly or privately? I vaguely remember that I knew that was your reply, but I was searching through issues and the chat and I could not find where we talked about it.
2. I'm good with the file content not being defined in the LRM, as long as I understand the format that GHDL uses.
  2.1. The use case is VHDL-Matlab/Octave co-execution using plain files/pipes, without using co-simulation interfaces. I'm not interested in transferring binary files accross VHDL simulators. I want to generate binary files with GHDL and read them from Matlab/Octave.
   2.2. I am aware that the binary format might be platform dependent. I'm not interested in transferring files from one CPU/OS to another.
3. Currently, I can `fread('tb_real.bin', 'double')`. I can also `fseek` for skipping the first 27 bytes in `tb_real_vector.bin`. However, where is that 27 coming from? What additional "metadata" are you writing in the "header" of the file?

Paebbels, 2021-05-14:
I would not expect any metadata in a file, it it's writing a sequence of scalar values. If this is technically required (array length, byte order information, element size information etc.), we could discuss about a generic format (header, data, footer) and bring it into the standard.

tgingold, 2021-05-14:
The header in the file is mostly ascii, and specifies the type of the data.

umarcor, 2021-05-14:
Thanks!

```
# hexdump -C control/test/tb_real_vector.bin 
00000000  23 47 48 44 4c 2d 42 49  4e 41 52 59 2d 46 49 4c  |#GHDL-BINARY-FIL|
00000010  45 2d 30 2e 30 0a 5b 46  5d 2e 0a 00 00 00 00 00  |E-0.0.[F].......|
00000020  00 00 00 33 33 33 33 33  33 f3 3f 33 33 33 33 33  |...333333.?33333|
00000030  33 03 40 cc cc cc cc cc  cc 0c 40 33 33 33 33 33  |3.@.......@33333|
00000040  33 13 40 00 00 00 00 00  00 18 40 cd cc cc cc cc  |3.@.......@.....|
00000050  cc 1c 40 cd cc cc cc cc  cc 20 40 33 33 33 33 33  |..@...... @33333|
00000060  33 23 40 99 99 99 99 99  99 25 40                 |3#@......%@|     
0000006b
```

That is: `#GHDL-BINARY-FILE-0.0.[F].\n`

@tgingold, what do you think about reserving 32 bytes for the header (or more) so that the actual content is aligned?

tgingold, 2021-05-14:
The header describes the content.  So because of records, it is not bounded.

The simpler would be to skip the first two lines (after the second \n).

closed 2021-08-01 by tgingold