ghdl/ghdl #1759: Make vcd dump files compliant to IEEE 1364-1995

pstrueb, 2021-05-14:
Dear ghdl community, Im working with vcd files to compare simulation output with other simulators. 

**The Problem**
Im using - a commercial graphical vcd compare solution that is not willing to accept "u" written by ghdl to the vcd dump as a valid value.

**Describe the solution you'd like**
In the IEEE 1364-1995 where the vcd format is described along with verilog is stated that only {0, 1, x, X, z, Z} are valid values. This means that u should be mapped to x.

**Describe alternatives you've considered**
It would also be a nice solution if this mapping can be introduced optionally as a switch to be compatible to previous ghdl revisions. e.g.:
`--vcd-map-u-x `

**Additional context**
Also the documentation for vcd seems a bit outdated because it says 
"Currently, there is no way to select signals to be dumped: all signals are dumped, which can generate big files."
Which does not seem to be the case anymore as I was able to select a subset of signals.
 
[Link to the standard](http://ce.sharif.edu/courses/80-81/1/ce493/references/ieee-std-1364.pdf) - pdf page 221

Thank you in advance for considering my issue.

umarcor, 2021-05-14:
> to be compatible with previous ghdl revisions

What do you mean? Previous revisions of GHDL did not write 'U'?

I found this problem with sigrok-cli/Pulseview too. A contributor of those projects "fixed" it similarly to how you suggest: https://github.com/sigrokproject/libsigrok/commit/358105152aaceb9065e4a18b2b0e0a1fc118eecb

> The previous implementation already mapped L/H/U/- literals for input
data values to the 0/1/0/0 logic levels which sigrok can handle.

---

> "Currently, there is no way to select signals to be dumped: all signals are dumped, which can generate big files."
> Which does not seem to be the case anymore as I was able to select a subset of signals.

I'm not sure to understand:

1. That comment might apply to GHDL's capability for filtering which signals to dump. It is not an statement about what the Verilog Std states.
  1.1 A similar comment seems to be repeated in https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-vcdgz and https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-wave.
2. It might be outdated, indeed, since option [`--read-wave-opt=`](https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-read-wave-opt) exists.
3. I think that Verilog supports specifying which signals to dump in the HDL code. I don't think that is supported in VHDL. Therefore, it might not be possible for GHDL to be strictly compliant with VCD.

pstrueb, 2021-05-14:
> > to be compatible with previous ghdl revisions
> 
> What do you mean? Previous revisions of GHDL did not write 'U'?

Triing to make more clear what I meant: Since all previous revisions of ghdl write 'U' it might be unexpected for some people that ghdl is certainly writing 'X'. This could be prevented by introducing the mapping as an optional switch.

> I'm not sure to understand:
>
> 1. That comment might apply to GHDL's capability for filtering which signals to dump. It is not an statement about what the Verilog Std states.
>    1.1 A similar comment [...]

Number 2. is what I was triing to say.

umarcor, 2021-05-14:
I'm writing a PR for fixing the docs.
I will let Tristan decide about 'U'. If 'U' is supported by GHDL and GTKWave, but it's not in the Verilog Std, maybe both tools should have a similar switch. OTOH, there is EVCD. Maybe that's relevant?

Blebowski, 2021-05-14:
Hello all, AFAICT Cadence tools have option to "reduce" VHDLs 9 state logic to Verilogs 4-state.

So, if VCD is exported with this option "off", U will be written, giving non-standard VCD.
If the option is "on", 'U' will be reduced to 'X', which makes sense.
What do you think?

umarcor, 2021-05-14:
@Blebowski, similarly to how different non-compliant implementations of the IEEE libs are supported, I think it would make sense to support multiple specifications of the VCD/VCD format. However, I'm afraid that there are more than 2-3 non-standard extensions. Hence we should first understand which specific variant we want/need to support. I believe that is better than a very specific option such as `--vcd-map-u-x`. It'd be interesting to know whether the tool used by @pstrueb claims to be fully compliant with the Std.

Blebowski, 2021-05-14:
Hi, @umarcor , you are right, broader picture is needed first. In the end, it might be something like:
`--vcd-export-policy=RAW|REDUCE_TO_VCD|EVCD|<ANYTHING_ELSE_WHAT_OCCURS>`

tgingold, 2021-05-14:
I would propose `--vcd-format=` option.

Choices would be: `vcd` or `vhdl`.  As far as I understand, `evcd` is slightly different than the current output.

closed 2021-09-30 by tgingold