ghdl/ghdl #1761: Some runtime storage limit reached around 52.4GB of memory used?

pbreuer, 2021-05-14:
All runtimes (ghdl-gcc/llvm/mcode) crash (seemingly while doing innocuous signal declarations, as far as I can track) at about 52.4GB of memory used. I have plenty of swap - around 100GB - and I have checked there are no process limits (I can run a single process claiming and dirtying memory via malloc till at least 64GB).

I have changed the VHDL source a lot, trying to make it smaller, and the crash happens at different places in my code, but always at about the same total memory usage (45GB+7GB). The compiled  executable is about 300MB, 650MB with debugging symbols.  Simulation never gets past a first pass through all units, setting up signals. I imagine that means allocating heap memory and initialising it.

The crash is presently reported by the debugger as here in my source code, which should be of no interest (to me neither, as it is generated code - by Clash 1.2.0):

898  signal \c$case_alt_7\    : signed(31 downto 0);
899  signal \c$case_scrut_3\  : signed(63 downto 0);   <-- here
900  signal result_12         : signed(63 downto 0);

The full backtrace shows memory being claimed via __ghdl_malloc, some kind of fail, then into a Ada backend signal raising routine:

#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000155555355537 in __GI_abort () at abort.c:79
#2  0x0000555562ed71ee in __gnat_last_chance_handler ()
#3  0x0000555562ec0f6e in grt.signals.sig_table.resize ()
#4  0x0000555562ec101c in grt.signals.sig_table.increment_last ()
#5  0x0000555562ec30d4 in grt.signals.create_signal () 
#6  0x0000555562ec5dc9 in __ghdl_create_signal_e8 ()
(from then then on back it's my code).

That is the first use of __ghdl_create_signal_e8().

What is an "E8" signal? It should be 64 bit.

I do not read Ada - or VHDL - but that  is part of a Create_Signal() call in the ghdl backend (see below).

I am running ghdl 1.0.0 from debian unstable on a 4-core amd64 architecture. I have recompiled (with gcc 10.2.1) and it makes no difference. I have tricked out your __ghdl_malloc() (and realloc) to catch and report successes and errors, but it reports success all the way until death. It never fails. Here is the end-sequence from ghdl-mcode -r

malloc got 40B in 1 parts for fn with return addr 0x14f70cd72028
malloc got 2560B in 64 parts for fn with return addr 0x14f70cd720eb
malloc got 1280B in 32 parts for fn with return addr 0x14f70cd721ae
malloc got 2560B in 64 parts for fn with return addr 0x14f70cd72271
malloc got 1280B in 32 parts for fn with return addr 0x14f70cd72334

Execution of ghdl-mcode terminated by unhandled exception
raised STORAGE_ERROR : grt-table.adb:58 explicit raise
Call stack traceback locations:
0x560c33a98931 0x560c33a989c0 0x560c33a9aa0e 0x560c33a9d819 0x14f70cd72332

I have no clue what goes on. As far as I can guess, there is something like a table limit being exceeded somewhere in an Ada back-end (Ada source is greek to me).

Nevertheless, I can see the generic __ghdl_create_signal function call in the runtime source;

      --  function __ghdl_create_signal_e8 (init_val : ghdl_i32_type)
      --                                    return __ghdl_signal_ptr;
      --  procedure __ghdl_signal_simple_assign_e8 (sign : __ghdl_signal_ptr;
      --                                            val : __ghdl_integer);
      Create_Signal_Subprograms ("e8", Ghdl_I32_Type,
                                 Ghdl_Create_Signal_E8,         <-- here
                                 Ghdl_Signal_Init_E8,
                                 Ghdl_Signal_Simple_Assign_E8,
                                 Ghdl_Signal_Start_Assign_E8,
                                 Ghdl_Signal_Next_Assign_E8,
                                 Ghdl_Signal_Associate_E8,
                                 Ghdl_Signal_Add_Port_Driver_E8,
                                 Ghdl_Signal_Driving_Value_E8,
                                 Ghdl_Signal_Force_Drv_E8,
                                 Ghdl_Signal_Force_Eff_E8);

The actual "E8" function is:

function Ghdl_Create_Signal_E8 (Val_Ptr : Ghdl_Value_Ptr;
                                   Resolv_Func : Resolver_Acc;
                                   Resolv_Inst : System.Address)
                                  return Ghdl_Signal_Ptr is
   begin
      return Create_Signal
        (Mode_E8, Val_Ptr, Get_Current_Mode_Signal, Resolv_Func, Resolv_Inst);
   end Ghdl_Create_Signal_E8;

and that's all she wrote. WHAT CAN CAUSE THAT TO FAIL? What does it even mean that it calls Create_SIgnal, and as far as I can vaguely see, Create_Signal calls it?

I am out of ways to attack this. Does someone please have a clue for the clueless? Maybe tell me how to fiddle with the Ada source to make it speak more?

I can even see where  the Create_Signal calls Increment_Last, which is nearer to the death point in the stack trace. It's the first line of Create_Signal:

   function Create_Signal
     (Mode : Mode_Type;
      Value_Ptr : Ghdl_Value_Ptr;
      Mode_Sig : Mode_Signal_Type;
      Resolv_Proc : Resolver_Acc;
      Resolv_Inst : System.Address)
     return Ghdl_Signal_Ptr
   is
      Res : Ghdl_Signal_Ptr;
      Resolv : Resolved_Signal_Acc;
      S : Ghdl_Signal_Data (Mode_Sig);
      Init_Val : Value_Union;
   begin
      Sig_Table.Increment_Last;      <- here
      ...

After that, I lose it. How does one read Ada! I am guessing as to where a process starts and ends. Does the  mixed case  signify anything?. I can't tell what is a header file and what isn't. I can't tell what is a declaration and what isn't. I have no success in figuring out where grt.signals.sig_table.resize() is in order to look at it ... what are all those dots in the name? Do processes have subprocesses? How do I declare a static inline malloc so I can get it embedded in place in the source code at call sites and have it report the function name that called it instead of having to try and figure it out later?

How do I even figure which units are using all that memory? The generated assembler contains an INST_SIZE declaration in each function. Is that it? Is that frame size (locals/stack use) or heap memory claimed?  I don't think it's the same thing as the .size (code size).

For completeness, here is the full back trace behind that call to create_signal_e8 that sort-of fails, apprently:

#0  0x0000555562ec5d94 in __ghdl_create_signal_e8 ()
#1  0x000055555db597b2 in testbench__crypto_encrypt8__ARCH__structural__DECL_ELAB (INSTANCE=0x15550118b010) at crypto_encrypt8.vhdl:23
#2  0x0000555556afbdcf in testbench__crypto_encrypt8__LASTARCH__DECL_ELAB (
    INSTANCE=0x15550118b010) at e~testbench:1
#3  0x000055555dabfec0 in testbench__crypto_stage8__ARCH__structural__crypto_encrypt8_result_3__COMP_ELAB (INSTANCE=0x1555094d4010) at crypto_stage8.vhdl:18
#4  0x000055555dad222a in testbench__crypto_stage8__ARCH__structural__DECL_ELAB
    (INSTANCE=0x1555094d4010) at crypto_stage8.vhdl:64
#5  0x0000555556afbce5 in testbench__crypto_stage8__LASTARCH__DECL_ELAB (
    INSTANCE=0x1555094d4010) at e~testbench:1
#6  0x000055555a73577b in testbench__write_stage__ARCH__structural__crypto_stage8_result_11__COMP_ELAB (INSTANCE=0x155530c80010) at write_stage.vhdl:18
#7  0x000055555a786723 in testbench__write_stage__ARCH__structural__DECL_ELAB (
    INSTANCE=0x155530c80010) at write_stage.vhdl:461
#8  0x0000555556afb23b in testbench__write_stage__LASTARCH__DECL_ELAB (
    INSTANCE=0x155530c80010) at e~testbench:1
#9  0x000055555a6bc2c9 in testbench__pipe__ARCH__structural__write_stage_result_3__COMP_ELAB (INSTANCE=0x15553129e010) at pipe.vhdl:49
#10 0x000055555a6cc50c in testbench__pipe__ARCH__structural__DECL_ELAB (
    INSTANCE=0x15553129e010) at pipe.vhdl:177
#11 0x0000555556afb19f in testbench__pipe__LASTARCH__DECL_ELAB (
    INSTANCE=0x15553129e010) at e~testbench:1
#12 0x0000555556e10e8a in testbench__testbench__ARCH__structural__pipe_result_43__COMP_ELAB (INSTANCE=0x155553b69010) at TestBench.vhdl:35
#13 0x000055555713a8ab in testbench__testbench__ARCH__structural__DECL_ELAB (
    INSTANCE=0x155553b69010) at TestBench.vhdl:7350
#14 0x0000555556afd52c in __ghdl_ELABORATE () at e~testbench:11
#15 0x0000555562ed6ae7 in grt.main.ghdl_elaborate_wrapper ()
#16 0x0000555562ed7011 in __ghdl_run_through_longjump ()
#17 0x0000555562ed6c4f in grt_main_elab ()
#18 0x0000555562ed6d83 in grt.main.run ()
#19 0x0000555562ef0de2 in ghdl_main ()
#20 0x0000555562ed74f4 in main ()

What does "elaboration" even mean? Dynamic linking? Initialization of constants? Allocation of  heap memory for globals? How about writing the manual page in standard computer science language! I didn't even figure that "mcode" meant "interpreter" till now.

Any hints? Anyone?

PTB

Xiretza, 2021-05-14:
`grt-table.adb:58` is [here](https://github.com/ghdl/ghdl/blob/029c8cbb77e8a83bfa39ef7f1281c0d40141e800/src/grt/grt-table.adb#L58), and the raise occurs if `realloc(3)` fails. There may be some possible workarounds on ghdl side here, but the actual failure is caused by libc.

pbreuer, 2021-05-14:
Thank you! But why do you think the Ada code calls C realloc()? It says:

Table := Realloc (Table, New_Size);

and Realloc seems to be defined/declared locally just above as

 function Realloc (T : Table_Ptr; Size : size_t) return Table_Ptr;
      pragma Import (C, Realloc);                                                <------ here?

Why does that import realloc() from libc? Further up, I think a path is defined:

 with Grt.C; use Grt.C;

I can't see what that would be, but I guess that it should define Realloc.

I have __gnat_realloc() in grt-cgnatrts.c changed to make noise if it is called, and it is not. What is that for if not to stand in as some kind of foreign  interface function.

More to the point, what do I change the Table := Realloc to in order to be able to find out what is happening in realloc. I don't much fancy loading a realloc() replacement via LD_LIBRARY_gubbins, but I suppose I could.

Thanks again

PTB

pbreuer, 2021-05-15:
I'll add that running under gdb and setting a breakpoint in "realloc" never triggers that breakpoint at runtime (by contrast, setting one in "malloc" does trigger, often). So I am very, very interested in locating what "Realloc" in the Ada file really resolves to. Then I can attack it

On a guess, I have set a breakpoint for __GI___libc_realloc and that does trigger. There are two calls, one to resize something to 330M, and one that does the damage:

__GI___libc_realloc (oldmem=0x1554b77ff010,     bytes=18446744071562067968)

That is silly. Ghdl has counted wrong, or wrapped, or something. Yes, it is a signedness problem. The bytesize is this in Hex:

FFFFFFFF80000000

That is 2^31 sign-extended to 64 bits. It should be 2GB being resized (I have it!) and instead a load of gazillionbytes is being asked for.

The call is from #1  0x0000555562ec0f44 in grt.signals.sig_table.resize ()

The VHDL source code trigger is harmless:

899                       signal \c$case_scrut_3\             : signed(63 downto 0);    <- here

So I think it is back to you. I can't grok the Ada so my only recourse is to locate the actual call to realloc and replace it with a bit of defensive programming that removes a whole load of leading 1s in a 64-bit byte count.
I would like help with locating that call, or modifying it.

Thanks!

PTB

pbreuer, 2021-05-15:
Further ... the memory segment being realloced to 2GB in size is at that point size 1GB (approx), so I would guess that the Ada source is trying to dynamically extend a table by doubling the memory reserved for it and then carrying on writing new entries ...

The actual size of the memory pointed to by the pointer to be realloced is 1073745904 bytes, equal to 0x40000FF0. It's being doubled.

The safest way to double the size would be to do what I just did, and call for the current size with malloc_usable_size(ptr), which returns a size_t (which is 64-bit signed here), which you can safely double
as a size_t and pass to realloc.

Instead you must be keeping the current size of the table as a 32-bit value, and things are going wrong in casting it to a 64-bit size_t for realloc (if Ada does casting ..). You should be keeping the size as a size_t value, but does Ada know about that? If you insist on a 64-bit size you may have problems casting it down to 32-bit in a 32-bit environment.

The defensive solution is as indicated above, to call realloc(ptr,2*malloc_usable_size(ptr)), and don't trouble yourselves with keeping count of the table size, since you don't really know what bit size of counter will work in the deployment architecture and best not to get involved.

Alternatively, if you know the table row size and are counting rows, you may be suffering the classical "multiplicative overflow". Your 32-counts of row size and number will be right and translate OK to 64-bit numbers, but when you multiply them to get a byte-size for the table, the 32-bit space overflows. So "don't do that", if so. That is what reallocarray() array is for.  It takes the two arguments separately.

I would really appreciate a quick fix, but if you have 32-bit stuff hanging around there, it may be goodness-knows everywhere, and take ages to get rid of.

Still, I'm hoping ...

PTB

tgingold, 2021-05-15:
You probably have declared too many signals.  GHDL always scalarizes signals and a signal has many metadata with it (last value, last change time)...

Usually, it is due to memories represented using signals (while variables are much lighter).  But as you are generating the VHDL, I suppose you don't really have control on that.

That's something we can investigate if you provide your design.

pbreuer, 2021-05-15:
There is no such thing as "too many signals" (and what does "scalarize" mean?). I am not writing the VHDL and know nothing about it - it is generated automatically by Clash 1.2 from a higher level functional description of a complete 19-stage pipelined processor, working encrypted, in 84 subunits. It has the number of signals it has. If GHDL can't cope, it needs to say what it can't cope with, with specificity.

I have prepared a patch. It compiles. I have not tried it yet, but it will fix that particular issue, but you will have more like that if you think there is such a thing as "too many signals"! This replaces the Ada that multiplies the number of rows by the size of entries in 32 bits (which overflows at the 2GB mark) and then casts it to size_t to pass to libc "realloc" (I hope!) for a number of bytes with Ada that keeps the two numbers separate, casts them separately, and passes them to libc "reallocarray" (I hope!) as separate size_t numbers.

.
--- ghdl-1.0.0+dfsg.orig/src/grt/grt-table.adb
+++ ghdl-1.0.0+dfsg/src/grt/grt-table.adb
@@ -40,19 +40,22 @@ package body Grt.Table is

    --  Resize and reallocate the table according to LAST_VAL.
    procedure Resize is
-      function Realloc (T : Table_Ptr; Size : size_t) return Table_Ptr;
-      pragma Import (C, Realloc);
+      function ReallocArray (T : Table_Ptr; Rows : size_t; Entry_Size : size_t)
+          return Table_Ptr;
+      pragma Import (C, ReallocArray);
+
+      New_Table_Rows : size_t;
+      Table_Row_Size : size_t;

-      New_Size : size_t;
    begin
       while Max < Last_Val loop
          Max := Max + (Max - Table_Low_Bound + 1);
       end loop;

-      New_Size := size_t ((Max - Table_Low_Bound + 1) *
-                            (Table_Type'Component_Size / Storage_Unit));
+      New_Table_Rows := size_t (Max - Table_Low_Bound + 1);
+      Table_Row_Size := size_t (Table_Type'Component_Size / Storage_Unit);

-      Table := Realloc (Table, New_Size);
+      Table := ReallocArray (Table, New_Table_Rows, Table_Row_Size);

       if Table = null then
          raise Storage_Error;

This is not the right fix, IMO. It's local, and you need to go through everywhere looking for this kind of problem and fix it in a global way via an overall strategy. I offered some discussion in the post above of some options. "Going through everything and fixing it everywhere" is what must be done, but via what approach?

Why I haven't tried the patched GHDL yet is because I am continuing overnight to run the original under ghdl, just intervening personally at calls to "realloc" to fix the number passed to it. I have had to do it only that once so far. My guess from what I see is that there are TWO tables that are being extended dynamically, and that is being done by doubling their size when they need more rows.

Doubling is a bit excessive! Maybe instead keep some constant margin? I grant you that the new part may not actually be backed by buffers - and won't be on Linux - until you dirty it, and then only gradually, dirtied block by dirtied block (since at large sizes that will be backed by an anonymous mmap on a block device), but it's the principle of the thing. That's just way over the top. Making the number bigger slower will delay hitting an arithmetic overflow again. But really you need to detect and report it too. One needs to check that size_t(num_rows * row_size) == size_t(num_rows) * size_t(row_size) and REPORT non-equality.

(one table has ben doubled to 2GB, the other has been doubled to 600KB, so far - If they want to be doubled again, that is fine by me!)

A bigger question for me is WHY. Please let me have diagnostics on which unit needs "too many signals". I have been struggling to understand what is going on with simulation for weeks. That has been so not-funny.

Regards

PTB

pbreuer, 2021-05-15:
Shucks - that patch presented badly. I'll try again.

```
--- ghdl-1.0.0+dfsg.orig/src/grt/grt-table.adb
+++ ghdl-1.0.0+dfsg/src/grt/grt-table.adb
@@ -40,19 +40,22 @@ package body Grt.Table is

    --  Resize and reallocate the table according to LAST_VAL.
    procedure Resize is
-      function Realloc (T : Table_Ptr; Size : size_t) return Table_Ptr;
-      pragma Import (C, Realloc);
+      function ReallocArray (T : Table_Ptr; Rows : size_t; Entry_Size : size_t)
+          return Table_Ptr;
+      pragma Import (C, ReallocArray);
+
+      New_Table_Rows : size_t;
+      Table_Row_Size : size_t;

-      New_Size : size_t;
    begin
       while Max < Last_Val loop
          Max := Max + (Max - Table_Low_Bound + 1);
       end loop;

-      New_Size := size_t ((Max - Table_Low_Bound + 1) *
-                            (Table_Type'Component_Size / Storage_Unit));
+      New_Table_Rows := size_t (Max - Table_Low_Bound + 1);
+      Table_Row_Size := size_t (Table_Type'Component_Size / Storage_Unit);

-      Table := Realloc (Table, New_Size);
+      Table := ReallocArray (Table, New_Table_Rows, Table_Row_Size);

       if Table = null then
          raise Storage_Error;

```

pbreuer, 2021-05-15:
I can confirm the above patch "works", in the sense of not triggering the 32-64 bit signed conversion problem on resizing the ghdl internal table that is at issue. I have now run the simulation past the previous crash point, recording the calls to realloc (reallocarray, now), and that explains what was happening:

```
__GI___libc_reallocarray (optr=0x155542fd8010, nmemb=524288, elem_size=8)
__GI___libc_reallocarray (optr=0x155509710010, nmemb=1048576, elem_size=8)
__GI___libc_reallocarray (optr=0x155508f0f010, nmemb=2097152, elem_size=8)
__GI___libc_reallocarray (optr=0x15550341e010, nmemb=4194304, elem_size=8)
__GI___libc_reallocarray (optr=0x1554b77ff010, nmemb=268435456, elem_size=8)  <--HERE

```
Marked is the call that used to fail and now does not. It is a call to raise the size of a table to 256M rows of 8B each, for a total of 2GB. Since table sizes are being at most doubled on resize by the existing algorithm [max = max + (max - foo) in the Ada], that does not look like the same table as was just previously resized to 400K rows of 8GB each ...  but maybe, somehow. I could find out for sure, but won't. It already looks too much for sure like TWO tables being managed.

What are these tables and why are they growing? If someone knows, please let me know. [If you do not know, please say "I do not know" and give some reasons why not and pointers to how to find out! That is more helpful than silence or guesses glossed as fact!]

The simulation is still in an "elaboration" phase, as the execution is hanging off __ghdl_ELABORATE(). What is "elaboration", please? Can someone tell me? I am really exhausted at having to guess what the introverted language used in the documents means! Please remenber that documents are to explain for people who do not already know what you mean what you do mean, as well as an aide-memoir for you. Otherwise you remain isolated.

WHAT do I have to do to get some printed output from a simulation run, if it ever leaves "elaboration"? All I want is a cycle by cycle printout of the generated signal from the testbench, which in itself requires no input signals.  Can someone please tell me? I would be really grateful for some point of entry to this.

The testbench executable is producing signals declared as follows in the high level HDL from which the vhdl code for testbench is being generated:

```
{-# ANN testbench32
  (Synthesize { t_name   = "TestBench"
              , t_inputs = [ ]
              , t_output = PortName "trace"
              }) #-}

```

and that is all I know about the VHDL. There's a recent block diagram of the structure at https://www.researchgate.net/profile/Peter-Breuer-2/project/Encrypted-Computing/attachment/6052234e0f39c7000139ff5d/AS:1002372880945152@1615995726112/download/KPU.pdf

HOW do I get some printout from what now seems to be a running simulation for the generated vhdl? When I run the description from which the vhdl is derived in the high level HDL (clash 1.2.0), I see output like:

```
0:      0.1:    A_sub (16,0,0)  !*t0(80)=0.0 
1:      0.2:    A_addi (16,16,254)      !*t0(80)=0.254 
2:      20.0:   A_mtspr (16,0,41)       !*esr(105)=0.254 
3:      24.0:   A_sub (16,0,0)  !*t0(80)=0.0 
4:      24.1:   A_addi (16,16,474)      !*t0(80)=0.474 
5:      44.0:   A_mtspr (16,0,42)       !*epcr(106)=0.474 
6:      48.0:   A_sub (16,0,0)  !*t0(80)=0.0 
...
```
That stuff is just a pretty print of what the testbench signals mean, each cycle (with boring cycles elided), and the vhdl should be producing those signals in the simulation. I'll settle for the raw binary, but how do I get something VISIBLE out of a vhdl simulation?

The documentation seems to neglect to say that sort of thing, instead assuming that I am in charge of and control the vhdl and know all about it, which I am not and do not (although I am the author of a book on vhdl and other simulation semantics! That's much higher level).

Regards

PTB

My guess is it means "memory allocation and initialisation". Is it ever going to end? How can I get a breakout of how much is being allocated and for what, and which parts of the process have been completed (what units) and how much is still to go?

tgingold, 2021-05-17:
Clearly, your design declares too many signals (at least to exhaust memory).

Unfortunately, it is not easy to know where in your design it occurs.  The best you could do is to print a backtrace when the memory allocation fails and guess where using the backtrace.  There are calls each time it progress in the hierarchy.

The design hierarchy is built during elaboration.

pbreuer, 2021-05-17:
You really need to get this notion of "too many signals" out of your head because there no such thing and the  JOB of ghdl is to (a) run the design whatever it is and (b) if it can't, to say so, with appropriate specificiity about what where and why.

Actually, (c) the design runs fine, in a half-gigabyte of memory in simulation at high level at an acceptable realtime speed of about  400 cycles a minute, (d) now having hacked around that internal 32-64-bit cast problem for realloc in ghdl,  it also runs in ghdl now, requiring about 92GB of actual memory allocation, which is _nothing_.

However, (e), it is much slower than at high level, and obviously takes much more memory. it took hours to even get to  cycle zero. There appear to be computationally inefficient internal algorithms ("ordering signals"?), and (f) there is a lack of  feedback from ghdl on what it is doing, in particular how much memory is it allocating to what, and why.

Does "analysis" mode give that sort of information?

I note there are still no answers offered to really elementary questions like "what does elaboration mean" and what does ghdl do for it. I presume therefore that the answers are not known. I would be grateful if you could concentrate on providing those answers.

 My guess is still that the INST_SIZE value that ghdl embeds into the assembler of e~testbench.s is an estimate of the total static heap memory allocation for each computational unit. It should be known to a compiler? Can you confirm that?

If so, what are the counting units ... the native word size (64 bits here)?

BTW I haven't looked for more problems like that particular 32/64 signedness messup in the Ada code because I don't speak Ada, but I'm sure there will be more. Where there is one there is two - SE lore. Have you scanned for more? A quick scan ... shows 14 .adb files in src/ that use size_t. There's something suspicious in ./src/grt/grt-sdf.adb, for example:

Res := fread (Buf (Pos)'Address, 1, size_t (Read_Size), Sdf_Stream);

I presume that is a file read. Files can be more that 2GB in size.

 
Regards

PTB

Xiretza, 2021-05-17:
@pbreuer: please tone it down a bit. While this issue doesn't really concern me directly, I am subscribed to the ghdl repository and your comments in my inbox are starting to get on my nerves.

If I may direct your attention to section 11 of [the license](https://github.com/ghdl/ghdl/blob/c873b5f3f7d98fbecec8f9005788f8434f8f9c7e/COPYING.md):

> BECAUSE THE PROGRAM IS LICENSED FREE OF CHARGE, THERE IS NO WARRANTY FOR THE PROGRAM, TO THE EXTENT PERMITTED BY APPLICABLE LAW. EXCEPT WHEN OTHERWISE STATED IN WRITING THE COPYRIGHT HOLDERS AND/OR OTHER PARTIES PROVIDE THE PROGRAM "AS IS" WITHOUT WARRANTY OF ANY KIND, EITHER EXPRESSED OR IMPLIED, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE. THE ENTIRE RISK AS TO THE QUALITY AND PERFORMANCE OF THE PROGRAM IS WITH YOU. SHOULD THE PROGRAM PROVE DEFECTIVE, YOU ASSUME THE COST OF ALL NECESSARY SERVICING, REPAIR OR CORRECTION.

> You really need to get this notion of "too many signals" out of your head because there no such thing and the JOB of ghdl is to (a) run the design whatever it is and (b) if it can't, to say so, with appropriate specificiity about what where and why.

To be blunt, nobody here owes you anything. If ghdl doesn't work for you for one reason or another, that's unfortunate - if you bring these issues up nicely, I'm sure people would be interested in helping you out. But no one has any *obligation* to make sure ghdl either works for every conceivable use case or specifically states what it can and can't do, I suggest you keep that in mind and adjust your attitude about the problem accordingly.

umarcor, 2021-05-17:
@pbreuer, see [ghdl.github.io/ghdl/quick_start](https://ghdl.github.io/ghdl/quick_start):

> *Since this is the user and reference manual for GHDL, it does not contain an introduction to VHDL. Thus, the reader should have at least a basic knowledge of VHDL. A good knowledge of VHDL language reference manual (usually called LRM) is a plus*.

---

It seems that you are trying to use Clash for generating a whole chip design without minimal knowledge about the field.  GHDL and Verilator are rather similar in this regard, so I would expect the outcome of this issue to mimick verilator/verilator#2867. I strongly recommend reading/learning about hardware description languages and event-driven simulations. Moreover, if you want to simulate a whole chip in less than some months, I recommend to hire a hardware designer.

pbreuer, 2021-05-17:
I suggest everyone actually concentrate on the issues ...  You had a bug, you claimed it was "libc" (!!), I found it, I fixed it, I hacked a patch. I have pointed only to the code. Please add to my patch something that complains out loud when the newly hacked limits (which are only 8 times bigger than the old limits) are exceeded. I know nothing about Ada programming, not even enough to do that.

I hope that is not up for discussion.

I would be happy to point out more problems as I find them, but I expect cooperation in tracking them down, not hostility. The whole point of Ada is to be correct, and the  point of open source is so that people can see problems with your code, and fix them. If you don't want that, take it private.

Regards

PTB

[As to your insults about MY ignorance, I guess my PhD in EE from Cambridge doesn't really count, nor do any of my papers on VHDL, or book on it, or leadership of NATO and other projects on VHDL, and many other such fineries of scholarship and practice! Luckily, I claim only to know nothing about everything - which is how it should be - so I am very happy to embrace ignorance, if you call it that. It is more helpful than believing that what you know is correct, and that is the way it should be in science! There is no excuse for egotism.

So, what does "elaboration" do, then? Lack of answer always has meant "I don't know", in my experience! It's amazing to me that no one answers that.

PS. I've had a look at what might have been suspicious uses of size_t in grt-files.adb, and everything looks alright (what does "grt" stand for? Come to that, what does "adb" stand for!). You are using sequential read and write, without any use of offsets, so so long as the lumps you read and write are less than 2GB in size, that is alright. I guess it is possible, however, to define a signal with a  width that will exceed that when printed, and I see nothing that will complain about that.

There is also a clearly vulnerable definition of Malloc() in ghdlrun.adb

 Argv := Malloc
        (size_t (Argc * (Ghdl_C_String'Size / System.Storage_Unit)));

That is a buffer overflow attack vulnerability. It is reading arguments supplied by the user. Overflow can smash the stack, overwrite the return address, and  control the calling utility, with its rights and permissions, and so on ...

umarcor, 2021-05-17:
@pbreuer, I did like reading about your debugging abilities. You seem to have a pretty good knowledge about C and memory management, which is a great asset for this project. However, you need to adapt to an open source project where you expect to contribute, or where you expect to receive help for free. As valuable as your knowledge might be, we (users and developers of GHDL) do not have to permit you passive-aggressively questioning our knowledge, capabilities or interest. Very precisely, you were explicitly asked to provide a reproducer. As much as you wrote, you didn't address THE request of the main developer, which is also explicitly explained in the Bug Report template (which you removed and ignored).

There are 5-10 questions you raised in this issue which are explicitly answered either in the GHDL documentation or in the VHDL Language Reference Manual. No one is claiming your ignorance, but just reading your statements. I personally couldn't care less about your CV, for good and bad.  Should you know the answers to those all elementary questions given your background, I would be so glad if you could answer them yourself by reading the documents which you should already be familiar with, instead of expecting us to do so. If you don't, then something does not match in the picture. 

As @Xiretza said, tone it down. Relax, take it easy, and let's learn/collaborate. Which of the elementary questions you did were rhetoric and which were real things your don't know?

marzoul, 2021-05-17:
GHDL usage seems to broaden so the issue of what "chip size" is too much will inevitably be hit and raised by other people.
High-level synthesis tools are just accelerators to reach this from non-hardware people, but not even required, I have already hit these kinds of limits in the past with simple generate statements.

Some guiding message about how many signals were memory allocated, and are about to be allocated, would be helpful for most users to try on their side to reduce the design size while keeping that size meaningful.

Would it be possible to catch allocation errors (return NULL) and gracefully launch a small handler function to count and print some statistics ? Or, continue elaboration in some form of "reduced mode" : free already-allocated signals to relieve memory, and continue counting signals and memory, for the purpose of gathering information.
Example of message could look like :

Error : A memory allocation operation failed while allocating memory for signals.
You should reduce your design size and/or try to use a machine with more RAM. 
Number of signals already allocated : xxx
Total number of signals to allocate : xxx
Requested memory : +xxx GB
Already used memory : xxx GB
Total available memory : xxx GB (if that makes sense the system)
Estimated size required for all signals : xxx GB

Of course, the case of crashes due to historical code using signed "short" integers (leading to undesired sign extension when reaching malloc) is a different type of issue that we can safely say is bug.