ghdl/ghdl #1762: Assertion failure in trans-chap4.adb:305

zero9178, 2021-05-15:
**Description**

I was trying to compile a testbench of mine. I initially used the ghdl version shipped with my distro (Fedora 34), when that started failing I compiled current trunk. I use the LLVM backend and the llvm-config installed is 12.0.0

**Expected behaviour**

Compilation success, or a proper error message if there is something wrong with my code

**How to reproduce?**
I created following `test.vhd` file:

```vhdl

library ieee;
use ieee.numeric_bit.unsigned;

package isa is

    type alu_op is (alu_abus,alu_add,alu_and,alu_neg);
    
    type cond_op is (cond_no_jump,cond_neg,cond_zero,cond_none);
    
    type shifter_op is (shifter_noop,shifter_left,shifter_right);
    
    type bus_operand is ('0','1',neg_one,pc,r0,r1,r2,r3,r4,r5,r6,r7,r8,r9,r10,ac);
    
    constant data_word_size : natural := 16;
    
    subtype data_word is unsigned(data_word_size - 1 downto 0);
    
end package;

library ieee;
use ieee.numeric_bit.all;

entity reg is
    port(input : in bit_vector;
         clk : in bit;
         rst : in bit := '1';
         output : out bit_vector);
end entity;

architecture behaviour of reg is
    signal mem : bit_vector(input'RANGE);
begin

    changes : process (clk,rst) is
    begin
        if rst = '0' then
            mem <= (others => '0');
        elsif rising_edge(clk) then
            mem <= input;
        end if;
    end process;
    
    output <= mem;

end architecture;

library ieee;
use work.isa.all;

entity register_bank is
    port(s_result : in data_word;
         s_select, b_select, a_select : in bus_operand;
         clk, write_enable : in bit;
         rst : in bit := '1';
         a_out, b_out : out data_word);
end entity;

architecture behavioural of register_bank is

    type reg_clocks_t is array (r0 to ac) of bit;
    signal reg_clocks: reg_clocks_t;
    
    subtype bv is bit_vector(data_word'RANGE);
    
    type reg_outputs_t is array (r0 to ac) of bv;
    signal reg_outputs: reg_outputs_t;

begin

    registers : for i in r0 to ac generate
        reg : entity work.reg
                    port map(input => bv(s_result),
                             clk => reg_clocks(i),
                             rst => rst,
                             output => reg_outputs(i));
    end generate;

    read_process : process (b_select, a_select, clk, rst) is
    
        procedure assign(signal bus_out : out data_word; reg_select : bus_operand) is
        begin
            case reg_select is
                -- pc is currently undocumented
                when '0' | pc => bus_out <= (others => '0');
                when '1' => bus_out <= (0 => '1',others => '0');
                when neg_one => bus_out <= (others => '1');
                when others =>
                    bus_out <= data_word(reg_outputs(reg_select));
            end case;
        end procedure;
    
    begin
        assign(a_out, a_select);
        assign(b_out, b_select);
    end process;
    
    write_process : process (s_select, clk, write_enable) is
        variable enable : bit;
    begin
        for i in r0 to ac loop
            if i = s_select then
                enable := '1';
            else
                enable := '0';
            end if;
            reg_clocks(i) <= clk and write_enable and enable;
        end loop;
    end process;

end architecture;
```
It is the combination of the files isa.vhd, reg.vhd and register_bank.vhd from a project of mine:
https://github.com/zero9178/Micro16/tree/b74862ee43270620fb98e9f85c8b081a3f330d31

For compilation I used `ghdl-llvm -a test.vhd` where `ghdl-llvm` is a symlink to the just compiled `ghdl` with the llvm backend.

**Context**
Please, provide the following information:

- OS: Fedora 34
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `029c8cbb77e8a83bfa39ef7f1281c0d40141e800`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r277.g029c8cbb) [Dunoon edition]
Compiled with GNAT Version: 11.1.1 20210428 (Red Hat 11.1.1-
Target: x86_64-redhat-linux
/home/markus/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -c -o test.o test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:305
Call stack traceback locations:
0x7fdf3abb9712 0x6005ca 0x60d202 0x60d976 0x5a01ef 0x5a15c8 0x5a208a 0x635342 0x631005 0x631708 0x63151f 0x631787 0x5edf60 0x5e11b2 0x63c769 0x63e177 0x63f794 0x40b42f 0x7fdf3a40fb73 0x40a5bc 0xfffffffffffffffe
******************************************************************
ghdl-llvm:error: compilation error
```

tgingold, 2021-05-15:
I can reproduce the crash.  It is due to the type conversions in reg instantiation.

dalex78, 2026-08-20:
# Issue #1762 — Crash on a type conversion in a port association

Analysis by an AI agent.

## High-level summary

The design instantiates `reg` with a type conversion on the actual:

```vhdl
reg : entity work.reg
  port map (input => bv(s_result), ...);
```

`input` is an unconstrained `bit_vector` port and `s_result` is a `data_word`, i.e. `unsigned(15 downto 0)`. Elaborating that association crashed the gcc and llvm backends with an internal `ASSERT_FAILURE` in `trans-chap4.adb`. mcode is unaffected: it does not use the `src/vhdl/translate` translator this lives in.

## What is actually wrong in GHDL's implementation

`Elab_Conversion` (`src/vhdl/translate/trans-chap4.adb`) elaborates the conversion. `Elab_In_Conversion` passes the actual as its input and the formal as its output, so here the output is the unconstrained `input` port — unbounded — and the input is the bounded `data_word`. When the output is unbounded the code asserted that the input must be unbounded too:

```ada
if Out_Tinfo.Type_Mode in Type_Mode_Unbounded then
   --  The only reason why the output is unbounded is type conversion
   --  between two unbounded ports.
   pragma Assert (In_Tinfo.Type_Mode in Type_Mode_Unbounded);
```

That is not the only reason the output can be unbounded. A conversion between two unbounded ports is one case; an unconstrained formal that takes its constraint from the conversion is another, and that is this design. The premise fails and the assertion fires.

The assertion guards the branch that builds the destination bounds, and with assertions disabled GHDL does not stop — it continues into that branch with the precondition violated.

## What the fix does

Removes the assertion and corrects the comment to state both ways the output can be unbounded.

Nothing else is needed, because the branch already handles a bounded source. `Chap3.Get_Composite_Bounds` returns the bounds from the static layout for a `Type_Mode_Bounded_Arrays` operand, and `Chap7.Translate_Type_Conversion_Array_Bounds` works on Mnodes and types generically. The assumption was over-strict, not the code — checked before removing the assertion rather than deleting it because it was in the way.

## Testing

`testsuite/gna/issue1762/` covers two cases; both crash on master without the fix.

`test.vhd` is the design from the report. It is elaborated and simulated rather than only analyzed, because analysing alone cannot tell a correct conversion from one built on a violated precondition — which is what a `--disable-checks` build does here. `tb.vhd` writes `16#BEEF#` into `r1` and `16#1234#` into `r5` through the converted port and reads them back:

```
tb.vhd:41:5:@70ns:(report note): PASS a_out=48879 b_out=4660
```

`conv_range.vhdl` covers what the report's design cannot show. Its conversion targets `bit_vector(data_word'RANGE)`, which has exactly the range of the actual, so it cannot reveal where the bounds of the formal come from. The added case converts an `unsigned(15 downto 0)` actual to a `bit_vector(0 to 15)` target — a different range *and* direction — and checks the value that arrives, read from `'LEFT` to `'RIGHT` so the result does not depend on how the bounds are labelled:

```
pos(left..right)=1111000000000001 len=16
```

`16#F001#` is not a palindrome, so a reversal or misalignment would show. The bounds labels themselves are deliberately not asserted: the backends disagree about them for this construct, which is a separate matter from this crash.

Both cases pass on mcode, llvm and gcc, and with this fix alone applied to master. Full `sanity gna vests synth` pass on all three backends.

closed 2026-08-24 by tgingold