ghdl/ghdl #1763: Support passing multiple '--work' options between files for commands -a, -s, -i and synth

umarcor, 2021-05-15:
> [May 11, 2021 3:59 PM](https://gitter.im/ghdl1/Lobby?at=609a8dcffbcd854d3a193617)
> Hey there ! When using ghdl -s is there a way to specify several libs ? In exemple :
> 
> `ghdl -s f1.vhd f2.vhd --work="another_lib" f3.vhd`
> 
> (which doesn't work afaik)

As discussed in https://github.com/ghdl/ghdl/pull/1733#issuecomment-825054618, the same feature might apply to `-a`, `-s`, `-c`, `-i` and `synth`.

It seems that "*passing multiple `--work` between files*" is supported for -c already: #1362.

olofk, 2021-12-11:
Should this be closed? I got the impression from @tgingold that this was fixed now.

tgingold, 2021-12-11:
I think it needs to be implemented for -s and -i

umarcor, 2021-12-11:
It is supported for some (sub)commands only, which makes it inconsistent. This issue is for tracking consistency. See also #1204.