ghdl/ghdl #1764: Simulation Crash on OSVVM Axi4Master

JimLewis, 2021-05-15:
**Description**
Simulation Crash on OSVVM Axi4Master.   Exits with `child process exited abnormally`

**Expected behaviour**
A simulation that finishes properly.  

**How to reproduce?**
I have test cases that I will push to my new ghdl_issues repository - when I have an issue number.   

The current OSVVM Axi4Master.vhd exhibits the error.    I also have a reduced capability model that I was able to create by fiddling around that runs.   Unfortunately having a work around does not help me generalize what the issue is so the MWE is not small.   In fact, the model has two similar sections of code (WriteAddressHandler and ReadAddressHandler) and only the WriteAddressHandler has issues.    This makes me worry that the bug is not so stable.   

- OS:  Windows 10
- Origin:
  - [ ] GHDL 2.0.0-dev (0.37.0.r1673.g75a3e05b) [Dunoon edition].  Compiled with GNAT Version: 10.3.0.   llvm code generator

If a `GHDL Bug occurred` block is shown in the log, please paste it here:   `child process exited abnormally`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

umarcor, 2021-05-15:
FTR, the GHDL version that Jim used is not shown in the history of this repository. It was created from https://github.com/umarcor/ghdl/commits/75a3e05ba9b4ae17d75e103e2f7679a834377c8e, which is a temporary CI fix on top of https://github.com/ghdl/ghdl/commit/029c8cbb77e8a83bfa39ef7f1281c0d40141e800.

JimLewis, 2021-05-16:
The provided examples show the code that fails and the modifications that I did to get the code working.

Code that breaks GHDL
```vhdl
  WriteAddressHandler : process
    alias    AW    : AxiBus.WriteAddress'subtype is AxiBus.WriteAddress ;
    variable Local : AxiBus.WriteAddress'subtype ;
    variable WriteAddressReadyTimeOut : integer ;
  begin
    -- AXI4 Lite Signaling
    AW.Valid  <= '0' ;
    AW.Addr   <= (Local.Addr'range   => '0') ;
    AW.Prot   <= (Local.Prot'range   => '0') ;
    -- AXI4 Full Signaling
    AW.Len    <= (Local.Len'range    => '0') ;
    AW.ID     <= (Local.ID'range     => '0') ;
    AW.Size   <= (Local.Size'range   => '0') ;
    AW.Burst  <= (Local.Burst'range  => '0') ;
    AW.Lock   <= '0' ;
    AW.Cache  <= (Local.Cache'range  => '0') ;
    AW.QOS    <= (Local.QOS'range    => '0') ;
    AW.Region <= (Local.Region'range => '0') ;
    AW.User   <= (Local.User'range   => '0') ;

    WriteAddressLoop : loop
      -- Find Transaction
      if WriteAddressFifo.Empty then
         WaitForToggle(WriteAddressRequestCount) ;
      end if ;
      (Local.Addr, Local.Len, Local.Prot, Local.ID, Local.Size, Local.Burst, Local.Lock, Local.Cache, Local.QOS, Local.Region, Local.User) := WriteAddressFifo.Pop ;

      WaitForClock(Clk, integer'(Params.Get(Axi4OptionsType'POS(WRITE_ADDRESS_VALID_DELAY_CYCLES)))) ; 

      -- Do Transaction
      AW.Addr   <= Local.Addr      after tpd_Clk_AWAddr   ;
      AW.Prot   <= Local.Prot      after tpd_clk_AWProt   ;
      -- AXI4 Full
      AW.Len    <= Local.Len       after tpd_clk_AWLen    ;
      AW.ID     <= Local.ID        after tpd_clk_AWID     ;
      AW.Size   <= Local.Size      after tpd_clk_AWSize   ;
      AW.Burst  <= Local.Burst     after tpd_clk_AWBurst  ;
      AW.Lock   <= Local.Lock      after tpd_clk_AWLock   ;
      AW.Cache  <= Local.Cache     after tpd_clk_AWCache  ;
      AW.QOS    <= Local.QOS       after tpd_clk_AWQOS    ;
      AW.Region <= Local.Region    after tpd_clk_AWRegion ;
      AW.User   <= Local.User      after tpd_clk_AWUser   ;

      Log(ModelID,
        "Write Address." &
        "  AWAddr: "  & to_hstring(Local.Addr) &
        "  AWProt: "  & to_string(Local.Prot) &
        "  Operation# " & to_string(WriteAddressDoneCount + 1),
        INFO
      ) ;

      GetAxi4Parameter(Params, WRITE_ADDRESS_READY_TIME_OUT, WriteAddressReadyTimeOut) ;

      ---------------------
      DoAxiValidHandshake (
      ---------------------
        Clk            =>  Clk,
        Valid          =>  AxiBus.WriteAddress.Valid,  --!GHDL
        Ready          =>  AxiBus.WriteAddress.Ready,  --!GHDL
        tpd_Clk_Valid  =>  tpd_Clk_AWValid,
        AlertLogID     =>  BusFailedID,
        TimeOutMessage =>  "Write Address # " & to_string(WriteAddressDoneCount + 1),
        TimeOutPeriod  =>  WriteAddressReadyTimeOut * tperiod_Clk
      ) ;

      -- State after operation
[... 198 lines cut ...]
        TimeOutMessage =>  "Read Address # " & to_string(ReadAddressDoneCount + 1),
        TimeOutPeriod  =>  ReadAddressReadyTimeOut * tperiod_Clk
      ) ;

      -- State after operation
      AR.Addr   <= Local.Addr   + 4  after tpd_Clk_ARAddr   ;
      AR.Prot   <= Local.Prot   + 1  after tpd_clk_ARProt   ;
      -- AXI4 Full
      AR.Len    <= Local.Len    + 1  after tpd_clk_ARLen    ;
      AR.ID     <= Local.ID     + 1  after tpd_clk_ARID     ;
      AR.Size   <= Local.Size   + 1  after tpd_clk_ARSize   ;
      AR.Burst  <= Local.Burst  + 1  after tpd_clk_ARBurst  ;
      AR.Lock   <= Local.Lock        after tpd_clk_ARLock   ;
      AR.Cache  <= Local.Cache  + 1  after tpd_clk_ARCache  ;
      AR.QOS    <= Local.QOS    + 1  after tpd_clk_ARQOS    ;
      AR.Region <= Local.Region + 1  after tpd_clk_ARRegion ;
      AR.User   <= Local.User   + 1  after tpd_clk_ARUser   ;

      -- Signal completion
      Increment(ReadAddressDoneCount) ;
      wait for 0 ns;
    end loop AddressReadLoop ;
  end process ReadAddressHandler ;
```
If there are differences, I cannot see them.

JimLewis, 2021-05-16:
I will retest this when I get the build for 1765.   Maybe we will get lucky and they are related somehow.

JimLewis, 2021-05-16:
If you hate the OSVVM scripting, this is the set of commands that I ran to build the OSVVM Library:

```bash
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/NamePkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/OsvvmGlobalPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/VendorCovApiPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/TranscriptPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/TextUtilPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/AlertLogPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/MessagePkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/SortListPkg_int.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/RandomBasePkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/RandomPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/CoveragePkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/MemoryPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/ScoreboardGenericPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/ScoreboardPkg_slv.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/ScoreboardPkg_int.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/ResolutionPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/TbUtilPkg.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/osvvm/OsvvmContext.vhd
ghdl -a --std=08 -Wno-hide --work=OSVVM_Common --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_common/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/Common/src/StreamTransactionPkg.vhd
[... cut ...]
ghdl -a --std=08 -Wno-hide --work=osvvm_axi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_axi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/AXI4/Axi4/src/Axi4Memory.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_axi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_axi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/OsvvmLibraries/AXI4/Axi4/src/Axi4MemoryVti.vhd
```

JimLewis, 2021-05-16:
After building the OSVVM libraries, this is the set of commands that run the test case:
```bash
ghdl -a --std=08 -Wno-hide --work=osvvm_axi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_axi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_fails/Axi4Master.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_fails/TestCtrl_e_ghdl.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_fails/TbAxi4.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_fails/TbAxi4Memory.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/TestCases/TbAxi4_BasicReadWrite.vhd
ghdl --elab-run --std=08 --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev TbAxi4_BasicReadWrite
```

JimLewis, 2021-05-16:
This is the set of commands that build the test case that uses the code that works and produces the correct output:

```
ghdl -a --std=08 -Wno-hide --work=osvvm_axi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_axi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_works/Axi4Master_ghdl.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_works/TestCtrl_e_ghdl.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_works/TbAxi4.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/Axi4Testbench_works/TbAxi4Memory.vhd
ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/TestCases/TbAxi4_BasicReadWrite.vhd
ghdl --elab-run --std=08 --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev TbAxi4_BasicReadWrite

ghdl -a --std=08 -Wno-hide --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_tool_issues/ghdl_issues/GHDL_1764_Axi4Master_Sim_Fails/TestCases/TbAxi4_MemoryReadWrite1.vhd
ghdl --elab-run --std=08 --work=osvvm_TbAxi4 --workdir=/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbaxi4/v08 -P/c/SynthWorks/Dev/_tool_issues/ghdl_issues/sim/VHDL_LIBS/GHDL-2.0.0-dev TbAxi4_MemoryReadWrite1
```

The blank line in the above script shows a gap between the simulations that run.

Output produced for the first test is:
```
%% Log   ALWAYS  in Default, Write and Read with ByteAddr = 0, 4 Bytes at 110 ns
%% Log   INFO    in master_1, Write Data.  WData: 55555555  WStrb: 1111  Operation# 1 at 110 ns
%% Log   INFO    in master_1, Write Address.  AWAddr: AAAAAAA0  AWProt: 000  Operation# 1 at 110 ns
%% Log   INFO    in responder_1, Write Data.  WData: 55555555  WStrb: 1111  Operation# 1 at 120 ns
%% Log   INFO    in responder_1, Write Address.  AWAddr: AAAAAAA0  AWProt: 000  Operation# 1 at 120 ns
%% Log   INFO    in responder_1, Write Response.  BResp: 0  Operation# 1 at 120 ns
%% Log   PASSED  in Default, Responder Write Addr:  Received : AAAAAAA0 at 120 ns
%% Log   PASSED  in Default, Responder Write Data:  Received : 55555555 at 120 ns

. . . -- Skipped over many similar messages

%% Log   INFO    in master_1, Read Data: 00DDCCBB  Read Address: 11111111  Prot: 0 at 410 ns
%% Log   PASSED  in Default, Master Read Data:  Received : DDCCBB at 410 ns

%% DONE  PASSED  TbAxi4_BasicReadWrite  Passed: 60  Affirmations Checked: 60  at 430 ns
```

The second test completes as follows:
```
%% DONE  PASSED  TbAxi4_MemoryReadWrite1  Passed: 40  Affirmations Checked: 40  at 440 ns
```

JimLewis, 2021-05-16:
@tgingold    Need anything else from me?   It is high priority for me to support getting all OSVVM VC running under GHDL.   This is looking like one of the last big hurdles.

We have another minor hurdle - external names to our BurstFifos, but we have plans in the works that may change that anyway.

tgingold, 2021-05-17:
I will have a look at Axi4Master.  It might be a minor bug.

External names are very different.  They aren't currently implemented because the elaboration model is very different from vhdl 2002 and previous versions.

JimLewis, 2021-05-17:
For some applications, external names are a choice.   OTOH, when a verification component is embedded inside a design, think of a Zynq core - then we need to connect to the VC without adding ports to the design - so the VC can work with the same ports that the actual design does.    A convenient way to do this is with external names.

tgingold, 2021-05-18:
Ok, I now understand the issue.  It is not really a minor bug, but it's a backend bug that happens with gcc or llvm backends.

tgingold, 2021-05-19:
The test is now passing.  Thank you for the issue.

closed 2021-05-19 by tgingold

JimLewis, 2021-05-19:
@tgingold  thanks for the fix.

commit 17e24689 (2021-05-19): ortho/gcc, ortho/llvm: do not allocate space for unbounded fields.  Fix #1764