ghdl/ghdl #1766: Powershell vendor script for OSVVM is failing with "Missing an argument for parameter 'VHDL2008'"

umarcor, 2021-05-16:
It seems that https://github.com/ghdl/ghdl/commit/47e76528 (#1730) introduced a regression.

- Extended tests run before: https://github.com/ghdl/extended-tests/runs/2392127303
- Extended tests run after: https://github.com/ghdl/extended-tests/runs/2401859382

```
compile-osvvm.ps1: D:\a\extended-tests\extended-tests\verification\OSVVM\run.ps1:2
Line |
   2 |  & $env:GHDL_PREFIX\vendors\compile-osvvm.ps1 `
     |  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | Missing an argument for parameter 'VHDL2008'. Specify a parameter of type 'System.Boolean' and try
     | again.
```

/cc @bahm-lgs @Paebbels

bahm-lgs, 2021-05-17:
It seems like compile-osvvm.ps1 uses the function Get-VHDLVariables with parameters of type "switch" and and compile-xilinx-vivado.ps1 with params of type "boolean". That's why 47e7652 fixed it for the xilinx scipt but broke it for the OSVVM script.

This could be fixed like that:
bbb2a1220f4febe776b6a093c53fa8a2ee6535f1

Paebbels, 2021-05-17:
Thanks @bahm-lgs for narrowing it down. Yesterday, I also had a quick look and was distracted by the error message and line number. I also traced it to `Get-VHDLVariables` CmdLet using `-VHDL2008`, but unfortunately run out of time to see the differences in switch vs. boolean.

bahm-lgs, 2021-05-17:
@umarcor could you rerun the extended regression tests with the code from #1767?

umarcor, 2021-05-18:
> @umarcor could you rerun the extended regression tests with the code from #1767?

@bahm-lgs I'm trying to do so: https://github.com/umarcor/ghdl-extended-tests/actions/runs/853983847. Unfortunately, GitHub Actions is having issues today: https://www.githubstatus.com/. I will let you know as soon as I can see it executed.

With regard to the specific fix, just for my own understanding, why `bool` and not `switch`? Is there any difference between them? Why do both exist?

> Or simply omit the parameter like in the UVVM script, since VHDL-2008 is the default anyway.

Do you mean that VHDL-2008 is the default in "all the vendor scripts"? The default in GHDL is 93c, I believe.

bahm-lgs, 2021-05-18:
> With regard to the specific fix, just for my own understanding, why `bool` and not `switch`? Is there any difference between them? Why do both exist?

Because 47e7652 changed the parameters to bool so it's compatible with the usage in the compile-xilinx-vivado script.

> > Or simply omit the parameter like in the UVVM script, since VHDL-2008 is the default anyway.
> 
> Do you mean that VHDL-2008 is the default in "all the vendor scripts"? The default in GHDL is 93c, I believe.

No, it is the default in the Get-VHDLVariables function. The UVVM script simply calls it without parameters to set VHDL2008, see https://github.com/ghdl/ghdl/blob/f07792aec9c4e943978f8212e2a0f6e870294dbf/scripts/vendors/compile-uvvm.ps1#L167

umarcor, 2021-05-18:
> Because [47e7652](https://github.com/ghdl/ghdl/commit/47e765284652e2835cc03994de0d4e25d6d3c5ff) changed the parameters to bool so it's compatible with the usage in the compile-xilinx-vivado script.

This is my concern. The parameters were of type `switch`. When changing them to `bool` for Xilinx, all other scripts don't match. #1767 fixes it for OSVVM, but others might need the same change. Instead, it would make sense to revert https://github.com/ghdl/ghdl/commit/47e76528 and use `switch` in Xilinx too.

bahm-lgs, 2021-05-18:
> The parameters were of type `switch`. When changing them to `bool` for Xilinx, all other scripts don't match. #1767 fixes it for OSVVM, but others might need the same change. Instead, it would make sense to revert [47e7652](https://github.com/ghdl/ghdl/commit/47e76528) and use `switch` in Xilinx too.

With the parameters of type switch the calls in all scripts using variables for the parameters would have to be changed from `Get-VHDLVariables $VHDL93 $VHDL2008` to `Get-VHDLVariables -VHDL93:$VHDL93 -VHDL2008:$VHDL2008`

Otherwise you would get the error "A positional parameter cannot be found that accepts argument..." again which I decribed in https://github.com/ghdl/ghdl/pull/1730#issuecomment-822430017

umarcor, 2021-05-18:
@bahm-lgs thanks for clarifying. I will let @Paebbels take the final decision.

Anyway, I'd like to let you know that #1767 worked, within the scope we had defined above: https://github.com/umarcor/ghdl-extended-tests/runs/2613356406?check_suite_focus=true.

bahm-lgs, 2021-05-18:
> @bahm-lgs thanks for clarifying. I will let @Paebbels take the final decision.

Yep, he might remember that problem from https://stackoverflow.com/questions/38009106/how-to-pass-a-switch-parameter-to-another-powershell-script

Paebbels, 2021-09-22:
@bahm-lgs sorry that this got so long delayed. I summarized all proposed changes and also what was the old, new and intended code. The fix is in #1878.

I'm not sure why the code didn't contain the `-VHDL2008:$VHDL2008` syntax anymore, because 5 years ago I stumbled of this problem and wrote a question on SO: [How to pass a switch parameter to another PowerShell script?](https://stackoverflow.com/questions/38009106/how-to-pass-a-switch-parameter-to-another-powershell-script). I know, that I did the implementation somewhere in the code, but maybe when reworking the compile scripts it got "merged away".

Anyhow, I tested it on my local machine with all tools I have.

PR #1878 (2021-09-22): Fixed PoSh scripting
This fixes the vendor scripts written in PowerShell for a bug concerning `bool` vs. `switch` parameters.

Fixes #1766, fixes #1767.

-----------------
The problem is described here: [How to pass a switch parameter to another PowerShell script?](https://stackoverflow.com/questions/38009106/how-to-pass-a-switch-parameter-to-another-powershell-script)
closed 2021-09-22 by umarcor