ghdl/ghdl #1768: GHDL Bug when building GR NOEL-V generic

duncanrhamill, 2021-05-17:
**Description**
_Disclaimer:_ I am not familiar with VHDL/GHDL, I'm only trying to simulate an existing design by Gaisler.

I'm trying to build Gaisler Research's NOEL-V generic SoC to simulate, however I get a bug on both 0.37 and 1.0.0 around the code described below.

**Expected behaviour**
Others using Vivaldo seem to have build this design, so I would expect it to compile with GHDL.

**How to reproduce?**
Download [GRLIB 2021.1](https://www.gaisler.com/index.php/downloads/leongrlib), follow quick start in the library docs under `doc/grlib.pdf`, then 

```shell
cd designs/noelv-generic
make ghdl
````
The bug appears to be caused by line 3 of the extract below, with the error being `are_trees_equal: cannot handle IIR_KIND_INDEXED_NAME (./../../lib/gaisler/axi/axi.vhd:481:40)`

```vhd 
  function read_replicate (
    data_in : in std_logic_vector;
    addr    : in std_logic_vector(log2(AXIDW/8)-1 downto 0);
    hsize   : in std_logic_vector(2 downto 0)
    ) return std_logic_vector is
    variable addr_i       : integer range 0 to (AXIDW/32)-1;
    variable data_shifted : std_logic_vector(AXIDW-1 downto 0);
    variable data_out     : std_logic_vector(AXIDW-1 downto 0);
  begin
    
    addr_i := to_integer(unsigned(addr(addr_range_high(AXIDW) downto addr_range_low(AXIDW))));

    data_shifted := data_in;
    if AXIDW > 32 then
      for i in 0 to range_return(AXIDW) loop
        for j in i to (AXIDW/32)-1 loop
          if (j-i) = addr_i then
            data_shifted((i+1)*32-1 downto i*32) := data_in((j+1)*32-1 downto j*32);
          end if;
        end loop;  -- j      
      end loop;  -- i
    end if;

    data_out := data_in;
    case hsize is
      --replicate for 8-bit/16-bit/32-bit
      --always 32-bit replication
      when "000" | "001" | "010" =>
        data_out := subwidth_replicate(data_shifted(31 downto 0), AXIDW, 32);
      when "011" =>
        if AXIDW > 64 then
          --replicate for 64-bit
          data_out := subwidth_replicate(data_shifted(dswr_64(AXIDW) downto 0), AXIDW, dswr_64(AXIDW)+1);
        end if;
      when "100" =>
        if AXIDW > 128 then
          --replicate for 128-bit
          data_out := subwidth_replicate(data_shifted(dswr_128(AXIDW) downto 0), AXIDW, dswr_128(AXIDW)+1);
        end if;
      when others => null;
    end case;

    return data_out;

  end read_replicate;
```

**Context**
Please, provide the following information:

- OS: Pop!_OS 20.10
- Origin:
  - [x] Package manager: `0.37`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `2fb2384`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/drh/Development/rust-in-space/grlib-gpl-2021.1-b4265/designs/noelv-generic/
Command line:
ghdl -m -fexplicit --ieee=synopsys --mb-comments --warn-no-binding -O2 --workdir=gnu/work --work=work -Pgnu -Pgnu/grlib -Pgnu/techmap -Pgnu/eth -Pgnu/opencores -Pgnu/gaisler -Pgnu/work testbench
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x559144fd2486 0x5591450bfc87 0x5591450bf682 0x5591450bf8aa 0x5591450bf25a 0x5591450bf35a 0x5591450bf14a 0x5591450bee64 0x5591450bf066 0x5591450bfcb8 0x5591450c0b43 0x5591450b36d0 0x5591450b396f 0x5591450c2986 0x5591450c3ad2 0x5591450794f8 0x55914507a36c 0x559145125171 0x55914512556e 0x5591451251fa 0x5591451251fa 0x5591451251fa 0x559145126b67 0x55914513428a 0x55914513bab8 0x559145070f29 0x5591451fcb76 0x559144e27a67 0x7fd7cbf9ccb0 0x559144e261cc 0xfffffffffffffffe
******************************************************************
```

tgingold, 2021-05-17:
Thank you for the report.  The issue has been fixed.

closed 2021-05-17 by tgingold

diogratia, 2021-05-17:
A second issue (a GHDL Bug occurred message) showed up using the daily binary distribution with f07792a during analysis of ./../../lib/gaisler/noelv/core/iunv.vhd:

```none
ghdl -m -fexplicit --ieee=synopsys --mb-comments --warn-no-binding -O2 --workdir=gnu/work --work=work `cat ghdl.path` testbench
./../../lib/techmap/maps/syncram.vhd:498:7:warning: IN port "enable" must be connected (or have a default value) [-Wport]
      refram : generic_syncram generic map (abits, dbits)
      ^
./tb/testbench.vhd:164:5:warning: IN port "extrxclk" must be connected (or have a default value) [-Wport]
    p0: phy
    ^
analyze ./../../lib/gaisler/noelv/core/iunv.vhd
./../../lib/gaisler/noelv/core/iunv.vhd:222:33:warning: declaration of "rv" hides constant "rv" [-Whide]
  function is_rv(len : integer; rv : std_logic_vector(1 downto 0)) return boolean is
                                ^
./../../lib/gaisler/noelv/core/iunv.vhd:1175:17:warning: declaration of "r" hides signal "r" [-Whide]
  function ctrl(r : registers; stage : stage_type) return pipeline_ctrl_lanes_type is
                ^
./../../lib/gaisler/noelv/core/iunv.vhd:1187:16:warning: declaration of "r" hides signal "r" [-Whide]
  function csr(r : registers; stage : stage_type) return csr_type is
               ^
./../../lib/gaisler/noelv/core/iunv.vhd:1200:16:warning: declaration of "r" hides signal "r" [-Whide]
  function rs1(r : registers; stage : stage_type) return rfa_lanes_type is
               ^
./../../lib/gaisler/noelv/core/iunv.vhd:1216:16:warning: declaration of "r" hides signal "r" [-Whide]
  function rs2(r : registers; stage : stage_type) return rfa_lanes_type is
               ^
./../../lib/gaisler/noelv/core/iunv.vhd:1230:17:warning: declaration of "r" hides signal "r" [-Whide]
  function ctrl(r : registers; stage : stage_type; lane : lanes_range) return pipeline_ctrl_type is
                ^
./../../lib/gaisler/noelv/core/iunv.vhd:1238:15:warning: declaration of "r" hides signal "r" [-Whide]
  function rd(r : registers; stage : stage_type; lane : lanes_range) return std_logic_vector is
              ^
./../../lib/gaisler/noelv/core/iunv.vhd:1253:23:warning: declaration of "r" hides signal "r" [-Whide]
  function v_fusel_eq(r     : registers; stage : stage_type; lane : lanes_range;
                      ^
./../../lib/gaisler/noelv/core/iunv.vhd:1288:25:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq_only(r   : registers; stage : stage_type; lane : lanes_range;
                        ^
./../../lib/gaisler/noelv/core/iunv.vhd:1323:20:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq(r  : registers; stage : stage_type; lane : lanes_range;
                   ^
./../../lib/gaisler/noelv/core/iunv.vhd:1357:20:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq(r   : registers; stage : stage_type; lane : lanes_range;
                   ^
./../../lib/gaisler/noelv/core/iunv.vhd:1407:21:warning: declaration of "r" hides signal "r" [-Whide]
  function tost_rfa(r : registers; rfa : rfa_tuple) return string is
                    ^
./../../lib/gaisler/noelv/core/iunv.vhd:1418:20:warning: declaration of "r" hides signal "r" [-Whide]
  function tost_rd(r : registers; stage : stage_type; lane : lanes_range) return string is
                   ^
./../../lib/gaisler/noelv/core/iunv.vhd:1426:25:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq_xrdv(r   : registers; stage : stage_type; lane : lanes_range;
                        ^
./../../lib/gaisler/noelv/core/iunv.vhd:1453:25:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq_xrdv(r   : registers; stage : stage_type; lane : lanes_range;
                        ^
./../../lib/gaisler/noelv/core/iunv.vhd:1500:27:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq_xvalid(r     : registers; stage : stage_type; lane : lanes_range;
                          ^
./../../lib/gaisler/noelv/core/iunv.vhd:1528:27:warning: declaration of "r" hides signal "r" [-Whide]
  function v_rd_eq_xvalid(r     : registers; stage : stage_type; lane : lanes_range;
                          ^
./../../lib/gaisler/noelv/core/iunv.vhd:1572:21:warning: declaration of "inst" hides signal "inst" [-Whide]
  function csr_addr(inst : word) return csratype is
                    ^
./../../lib/gaisler/noelv/core/iunv.vhd:1578:21:warning: declaration of "r" hides signal "r" [-Whide]
  function csr_addr(r : registers; stage : stage_type) return csratype is
                    ^
./../../lib/gaisler/noelv/core/iunv.vhd:1584:19:warning: declaration of "r" hides signal "r" [-Whide]
  function csr_eq(r : registers; stage1 : stage_type; stage2 : stage_type) return boolean is
[... 329 lines cut ...]
                         dbgi           : in  nv_debug_in_type;
                         ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r286.gf07792ae) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/dio_gratia/Desktop/grlib-gpl-2021.1-b4265/designs/noelv-generic/
Command line:
/Users/dio_gratia/opt/ghdl/bin/ghdl1-llvm -fexplicit -fsynopsys --mb-comments --warn-no-binding -O2 --workdir=gnu/work --work=work -Pgnu/ -Pgnu/grlib/ -Pgnu/techmap/ -Pgnu/eth/ -Pgnu/opencores/ -Pgnu/gaisler/ -Pgnu/work/ -P/Users/dio_gratia/opt/ghdl/lib/ghdl/ieee/v93/ - [... cut ...]
Load address: 0x10288c000
Call stack traceback locations:
0x103973bd1 0x102a99f38 0x102aa78c9 0x102aa8023 0x102a36b69 0x102a37f9c 0x102a38b00 0x102acfcc7 0x102acb76a 0x102acc08f 0x102acbcbd 0x102acc10e 0x102acbcbd 0x102acc15b 0x102a880a7 0x102a7a919 0x102ad718d 0x102ad8795 0x102ad9a72 0x10288f4e6
******************************************************************
ghdl:error: compilation error
make: *** [testbench] Error 1
```

[image]

The warnings preceding the failure are in procedure trigger_module

commit f9cb7943 (2021-05-17): vhdl-sem: handle indexed and slice names.  Fix #1768