ghdl/ghdl #1770: **** GHDL Bug occurred ****(ghdl -s)

vitorstraggiotti, 2021-05-21:
**Description**
I created two descriptions for simulation, one of them being a test bench with the following names:
ex5610.vhdl
ex5610_tb.vhdl
When running the command "ghdl -s ex5610.vhdl" there were no problems, but then when I used "ghdl -s ex5610_tb.vhdl" I received "**** GHDL Bug occurred ****"

**How to reproduce?**
Files used
[ex5610.txt](https://github.com/ghdl/ghdl/files/6524862/ex5610.txt)
[ex5610_tb.txt](https://github.com/ghdl/ghdl/files/6524863/ex5610_tb.txt)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
---------------------------------------------------------------------------------------------------------------------------------------------------

ex5610_tb.vhdl:15:16:error: "end component;" expected
ex5610_tb.vhdl:12:18:error: ';' is expected instead of 'attribute'
ex5610_tb.vhdl:22:8:error: 'begin' is expected instead of 'end'
ex5610_tb.vhdl:22:11:error: missing ";" at end of architecture
ex5610_tb.vhdl:24:8:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:25:8:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:27:1:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:36:8:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:40:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:41:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:42:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:44:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:45:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:46:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:47:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:49:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:50:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:51:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:52:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:54:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:55:16:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:57:8:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:58:1:error: missing entity, architecture, package or configuration
ex5610_tb.vhdl:58:25:error: 'of' is expected instead of ';'
ex5610_tb.vhdl:58:25:error: name expected here, found ';'
ex5610_tb.vhdl:58:25:error: 'is' is expected instead of ';'
ex5610_tb.vhdl:58:25:error: ';' (semi colon) not allowed alone
ex5610_tb.vhdl:59:1:error: 'begin' is expected instead of end of file
ex5610_tb.vhdl:59:1:error: unexpected end of file, 'END;' expected
ex5610_tb.vhdl:59:1:error: 'end' is expected instead of end of file
ex5610_tb.vhdl:58:26:error: missing ";" at end of architecture
ex5610_tb.vhdl:5:25:error: entity 'ex5610' was not analysed

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/vitor/Área de Trabalho/vhdl/ex_5.6.10/
Command line:
/usr/bin/ghdl-gcc -s ex5610_tb.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-sem.adb:108
Call stack traceback locations:
0x7f560e6fad12 0x4e4631 0x4e8b2c 0x4b8632 0x4f733b 0x4f7428 0x4b3bb2 0x40848f 0x7f560e2ea0b1 0x4084fc 0xfffffffffffffffe
******************************************************************

umarcor, 2021-05-21:
Tested with a recent version of GHDL on MSYS2 (MINGW64) and on ghdl/ghdl:ubuntu18-mcode and ghdl/ghdl:bullseye-gcc-9.1.0 containers. None of them produces the GHDL Bug ocurred block. Hence, I assume this is fixed in the development version.
@vitorstraggiotti can you please try a more recent version of GHDL and confirm?

vitorstraggiotti, 2021-05-21:
ok, will try

closed 2021-08-01 by tgingold