ghdl/ghdl #1771: libraries/ieee2008/float_generic_pkg-body.vhdl differs from the file in the IEEE repository

robertgj, 2021-05-23:
The file  [https://github.com/ghdl/ghdl/blob/master/libraries/ieee2008/float_generic_pkg-body.vhdl](url) differs from
[https://opensource.ieee.org/vasg/Packages/-/commits/98ecff1a0b80d57c4eb9df7dc0288483cd46b264/ieee/fixed_generic_pkg-body.vhdl](url). In particular, note lines 922 and 954:
```
--- ghdl/libraries/ieee2008/fixed_generic_pkg-body.vhdl	2021-05-23 22:45:50.849086742 +1000
+++ Packages-master/ieee/fixed_generic_pkg-body.vhdl	2019-07-27 08:53:10.000000000 +1000
@@ -719,10 +719,9 @@
   is
     variable result     : UNRESOLVED_ufixed (minimum(l'high, r'high) downto
                                              mine(l'low, r'low));
-    constant rlow : integer := mins(r'low, r'low);
     variable lresize    : UNRESOLVED_ufixed (maximum(l'high, r'low) downto
-                                             rlow-guard_bits);
-    variable rresize    : UNRESOLVED_ufixed (r'high downto rlow-guard_bits);
+                                             mins(r'low, r'low)-guard_bits);
+    variable rresize    : UNRESOLVED_ufixed (r'high downto r'low-guard_bits);
     variable dresult    : UNRESOLVED_ufixed (rresize'range);
     variable lslv       : UNRESOLVED_UNSIGNED (lresize'length-1 downto 0);
     variable rslv       : UNRESOLVED_UNSIGNED (rresize'length-1 downto 0);
@@ -919,7 +918,7 @@
       lslv       := to_uns (lresize);
       rslv       := to_uns (rresize);
       result_slv := lslv + rslv + cx;
-      c_out      := result_slv(left_index);
+      c_out      := result_slv(left_index-right_index);
       result := to_fixed(result_slv (left_index-right_index-1 downto 0),
                          left_index-1, right_index);
     end if;
@@ -950,7 +949,7 @@
       lslv       := to_s (lresize);
       rslv       := to_s (rresize);
       result_slv := lslv + rslv + cx;
-      c_out      := result_slv(left_index);
+      c_out      := result_slv(left_index-right_index);
       result := to_fixed(result_slv (left_index-right_index-1 downto 0),
                          left_index-1, right_index);
     end if;

```

umarcor, 2021-05-23:
@robertgj, IEEE libraries were released in 2019 and are compatible with VHDL 2019 only. Tristan backported them in order to replace the non open source libs of previous versions of the standard. Therefore, it is expected for IEEE sources to differ between GHDL and opensource.ieee.org. Please, can you update the title of this issue so it's more descriptive? For instance, mention `rresize`, `c_out` and/or `result_slv` in the title.

Moreover, did you find any actual mismatch or unexpected behaviour when using GHDL's libraries?

robertgj, 2021-05-24:
My apologies for the noise. I was trying to add an ieee.fixed_pkg to the web version of the Altera/Intel Quartus compiler. I think that the add_carry function in ghdl fixed_generic_pkg-body.vhdl is correct and that that in the IEEE repository is not.
 
A simple example:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.fixed_pkg.all;

entity tf is
end entity tf;

architecture testbench of tf is
    signal sx : sfixed (3 downto 0) := "1001";
    signal sy : sfixed (3 downto -3) := "0101110";
    signal sz : sfixed (3 downto -3);
    signal sci : std_ulogic := '1';
    signal sco : std_ulogic;
begin
  tt : process is
    variable x : sfixed (3 downto 0) := sx;
    variable y : sfixed (3 downto -3) := sy;
    variable z : sfixed (3 downto -3);
    variable ci : std_ulogic := sci;
    variable co : std_ulogic;
  begin
    add_carry (L => x, R => y, c_in => ci, result => z, c_out => co);
    sz <= z;
    sco <= co;
    wait for 1 ns;
    wait;
  end process tt;
end architecture testbench;

```

tgingold, 2021-05-24:
This makes sense.  The package has probably been fixed after VHDL-2008 publication.

robertgj, 2021-05-24:
Try this example with add_carry():
```
sx(3` downto 0) <= "1101"                    lslv(7 downto 0) := 16#68#

sy(3 downto -3) <= "0111000"                 rslv(7 downto 0) := 16#38#

sci<='1'                                       cx(0 downto 0) := 16#01#
                                                                                                                                                              
sz(3 downto -3) <= "0100001"           result_slv(7 downto 0) := 16#A1#
sco <='1'
```
But when I run the testbench above with these values of sx, sy and sci, then ghdl gives sco='0'.

In addition, when I try to recompile fixed_generic_pkg-body.vhdl into the work library, in  function round_fixed, the remainder argument shadows a function name appearing later in the package:
```
  fixed_generic_pkg-body.vhdl:192:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
  fixed_generic_pkg-body.vhdl:225:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
```

tgingold, 2021-05-24:
> But when I run the testbench above with these values of sx, sy and sci, then ghdl gives sco='0'.

Isn't it the correct value ?

> In addition, when I try to recompile fixed_generic_pkg-body.vhdl into the work library, in function round_fixed, the remainder argument shadows a function name appearing later in the package:

It is not defined later, it is defined before in the package declaration.

robertgj, 2021-05-24:
If I repeat with:
```
  signal sx : ufixed (6 downto 0) := "1101000";
  signal sy : ufixed (6 downto 0) := "0111000";
  signal sz : ufixed (6 downto 0);
```
then I get sco=1. 

It seems that if I do use add_carry then I should just define my range as n downto 0.

tgingold, 2021-05-24:
Yes, but that's the correct result too.  Signed fixed doesn't behave exactly like unsigned fixed.

umarcor, 2021-05-24:
> My apologies for the noise.

Hey! No need to apologise! You did nothing wrong! 

> I think that the add_carry function in ghdl fixed_generic_pkg-body.vhdl is correct and that that in the IEEE repository is not.

You might want to bring this issue in https://gitlab.com/IEEE-P1076/VHDL-Issues. That's where the VHDL analysis and standardisation group (VASG) discusses enhancements/fixes to the LRM and standard/IEEE packages.

robertgj, 2021-05-25:
Attached is :[add_carry_ghdl_testbench.tar.gz](https://github.com/ghdl/ghdl/files/6538842/add_carry_ghdl_testbench.tar.gz)
It contains:
1. add_carry_ghdl_testbench.vhdl : a simple testbench for add_carry from the ghdl std08 fixed_pkg
2. add_carry_ieee_testbench.vhdl : a simple testbench for add_carry using a version of the ieee std08 fixed_pkg lightly edited  so that it can be compiled into the work library.
3. compile.sh : a shell script for compiling the two testbenches.

In both cases the testbench calls add_carry with three tests:
1.  x and y with range 6 downto 0
2.  x and y with range 3 downto -3
3.  x with range 3 downto 0 and y with range 3 downto -3

In each case sco_xxx is the carry out bit. My results:
```
$ tar -xf add_carry_ghdl_testbench.tar 
$ ls
compile.sh  src
$ sh ./compile.sh 
GHDL 1.0-dev () [Dunoon edition]
 Compiled with GNAT Version: 11.0.0 20210210 (Red Hat 11.0.0-
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
src/firmware/common/ieee/ieee_fixed_generic_pkg-body.vhdl:192:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
src/firmware/common/ieee/ieee_fixed_generic_pkg-body.vhdl:225:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
$ ls
bin  compile.sh  obj  src
$ bin/add_carry_ghdl_testbench 
sci=1
sx_6_0=68.0
sy_6_0=38.0
sz_6_0=21.0
sco_6_0=1

sci=1
sx_3_m3=D.0
sy_3_m3=7.0
sz_3_m3=4.2
sco_3_m3=0

sci=1
sx_3_0=D.0
sy_3_m3=7.0
sz_3_0_m3=4.2
sco_3_0_m3=0

$ bin/add_carry_ieee_testbench 
sci=1
sx_6_0=68.0
sy_6_0=38.0
sz_6_0=21.0
sco_6_0=1

sci=1
sx_3_m3=D.0
sy_3_m3=7.0
sz_3_m3=4.2
sco_3_m3=1

sci=1
sx_3_0=D.0
sy_3_m3=7.0
sz_3_0_m3=4.2
sco_3_0_m3=1

$ 
```
In each case the operation should be identical, a bitwise addition with a carry out, and only the interpretation of the result changes. It seems to me that, in fact, the ieee version of add_carry is correct.

tgingold, 2021-05-25:
Ok, so there is a bug (or several bugs) in the vhdl-2008 version of the packages that has been fixed in the 2019 version.
Should the fixes be backported to vhdl-2008 ?  Why not.

closed 2021-08-01 by tgingold