ghdl/ghdl #1772: openpower-cores/a2i crash with LLVM backend, works with mcode

umarcor, 2021-05-23:
**Description**

VUnit was added to a2i for CI purposes. When using GHDL with mcode backend, analisis/elaboration is successful. However, when using LLVM, GHDL crashes.

**Expected behaviour**

None of the backends should crash.

**How to reproduce?**

See https://github.com/umarcor/a2i/runs/2650758403 (https://github.com/umarcor/a2i/commit/e8acd83e7384bc56e87ebe9fa46810a44391ac46).

1. Install GHDL with LLVM backend.
2. Clone openpower-cores/a2i.
3. Execute VUnit: `./run.py`.

**Context**

- OS: Debian Buster container
- Origin: container image ghdl/vunit:llvm-master

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-299-gb99e2d89) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/src/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/src/vunit_out/ghdl/libraries/ibm --work=ibm --std=08 -P/src/vunit_out/ghdl/libraries/vunit_lib/ -P/src/vunit_out/ghdl/libraries/support/ -P/src/vunit_out/ghdl/libraries/ibm/ -P/src/vunit_out/ghdl/libraries/clib/ -P/src/vunit_out/ghdl/libraries/tri/ -P/src/vunit_out/ghdl/libraries/top/ -P/src/vunit_out/ghdl/libraries/unimacro/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o /src/vunit_out/ghdl/libraries/ibm/std_ulogic_support.o /src/rel/src/vhdl/ibm/std_ulogic_support.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1266
Call stack traceback locations:
0x4b062e 0x63c160 0x63a846 0x63ab24 0x5f89e0 0x5f989e 0x5f8d88 0x5fb26d 0x5fb328 0x5dd007 0x5dd3f1 0x5d456b 0x5e6197 0x6433ed 0x642c68 0x642d07 0x599c78 0x65128d 0x652ca8 0x654306 0x40c3e3 0x7f5182e98099 0x40b578 0xfffffffffffffffe
******************************************************************
```

**Additional context**

The error is produced when compiling https://github.com/openpower-cores/a2i/blob/master/rel/src/vhdl/ibm/std_ulogic_support.vhdl.

tgingold, 2021-05-23:
Just comment out the attribute specification line 216:

--  attribute pin_bit_information of tconv : function is
--           (1 => ("   ","A0      ","INCR","PIN_BIT_SCALAR"),
--            2 => ("   ","10      ","INCR","PIN_BIT_SCALAR"));

The aggregate (of unbounded strings) is not yet supported.

umarcor, 2021-05-23:
Hmm... that required some effort, because the same code structure is used 200+ times. Yet, I did it, and I can confirm that the compilation is successful. See https://github.com/umarcor/a2i/actions/runs/869214279 (https://github.com/umarcor/a2i/commit/b9a9753ba935833b13f6d850647490df9b004489).

tgingold, 2021-05-24:
But in fact the aggregate is not valid.  The type is an array of an array of a string, but the length of the strings is not the same.

This is not the reason of the crash, but even if the feature were handled, the source won't be analyzed successfully.

umarcor, 2021-05-24:
> The type is an array of an array of a string, but the length of the strings is not the same.

I think that the length of the string is the same, indeed. That's why the are space-padding them, isn't it?

```vhdl
=> ("   ","A0      ","INCR","PIN_BIT_SCALAR")
```
```vhdl
=> ("   ","IN      ","SAME","PIN_BIT_VECTOR"),
```

Items are 3, 8, 4, 14 characters wide.

tgingold, 2021-05-24:
Indeed!  3, 8, 4 and 14 are not equal.

openpowerwtf, 2021-05-24:
That's odd.  If 'string' is 'array (natural range <>) of char', seems like arrays could hold different-length strings.  If you wanted to do compares, you would have to use same-length (padded) strings in those elements or have some special handler function.

umarcor, 2021-05-24:
@openpowerwtf my understanding is that VHDL does not allow having an array with elements of different types. As soon as you constraint the strings, you don't have an array of unconstrained strings anymore, but an array of four different constrained types.

/cc @JimLewis @Paebbels

tgingold, 2021-05-25:
> That's odd. If 'string' is 'array (natural range <>) of char', seems like arrays could hold different-length strings.

Well, no.  There is only one element subtype which could be queried using the Element attribute.
Furthermore, this idea was not even pursued.  Why were the strings padded

```
 attribute pin_bit_information of tconv : function is
          (1 => ("   ","A0      ","INCR","PIN_BIT_SCALAR"),
           2 => ("   ","10      ","INCR","PIN_BIT_SCALAR"));
```

instead of having different length strings:

```
 attribute pin_bit_information of tconv : function is
          (1 => ("","A0","INCR","PIN_BIT_SCALAR"),
           2 => ("","10","INCR","PIN_BIT_SCALAR"));
```

You should use records!

dalex78, 2026-09-04:
# Issue #1772 — attribute specification with an aggregate of unbounded strings

Analysis by an AI agent. The original crash is gone; a second one, reached once the attribute is actually used, is fixed here.

## The reproducer

The report pointed at `std_ulogic_support.vhdl` in openpower-cores/a2i, reached through VUnit. The construct tgingold identified on the thread is the attribute specification at line 216, and it reduces to this:

```vhdl
package repro is
  type string_array is array (natural range <>) of string;
  type pin_bit_info is array (natural range <>) of string_array;

  attribute pin_bit_information : pin_bit_info;

  function tconv (a : bit) return bit;

  attribute pin_bit_information of tconv : function is
    (1 => ("   ", "A0      ", "INCR", "PIN_BIT_SCALAR"),
     2 => ("   ", "10      ", "INCR", "PIN_BIT_SCALAR"));
end package;
```

Two separate things are wrong with it, and tgingold said both:

> The aggregate (of unbounded strings) is not yet supported.

> But in fact the aggregate is not valid. The type is an array of an array of a string, but the length of the strings is not the same.

The four strings are 3, 8, 4 and 14 characters long, so they cannot all be the element subtype of one array. The reporter had assumed the padding made them equal; it does not.

Analysing this crashed the llvm backend with `TYPES.INTERNAL_ERROR : trans.adb:1266` while mcode accepted it.

## Status on master

**Analysis no longer crashes, and it now names the exact problem** -- one message per offending element:

```
repro.vhdl:10:19:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
    (1 => ("   ", "A0      ", "INCR", "PIN_BIT_SCALAR"),
                  ^
```

The reporter's own file confirms it: `rel/src/vhdl/ibm/std_ulogic_support.vhdl` taken from a2i as-is now analyses with exit status 0 on both mcode and llvm, with six of those warnings and no internal error.

## What was still broken

Analysing the package is not the end of it. As soon as something *reads* the attribute, the invalid aggregate has to be evaluated, and on the pre-elaborated backends that turned a clean diagnostic into an internal error:

```
repro.vhdl:10:5:error: out of bound expression
    (1 => ("   ", "A0      ", "INCR", "PIN_BIT_SCALAR"),
    ^

******************** GHDL Bug occurred ***************************
raised CONSTRAINT_ERROR : elab-vhdl_objtypes.adb:1268 access check failed
```

The error is right; the box after it is not. `Elab.Vhdl_Decls.Elab_Attribute_Specification` calls `Synth_Expression_With_Type` and then goes straight on to

```ada
         Val := Unshare (Val, Instance_Pool);
         Val.Typ := Unshare (Val.Typ, Instance_Pool);
```

`Synth_Expression_With_Type` returns `No_Valtyp` once it has reported the bound violation, and `No_Valtyp.Typ` is null, so `Unshare` dereferences it. Every other caller in that file tests for `No_Valtyp` first; this one did not.

The fix is that test: report nothing further, mark the instance in error and stop. Both backends now end with the diagnostic alone --

```
repro.vhdl:10:5:error: out of bound expression
```

on mcode and llvm-jit, and `bound check failure at repro.vhdl:10` on llvm -- and no internal error.

## Testing

`testsuite/gna/issue1772` already existed, but it only ran `repro2.vhdl` and `repro3.vhdl` -- the two *valid* variants. The file holding the actual reported aggregate, `repro.vhdl`, was in the directory and never run. This extends the test rather than replacing it:

- `repro.vhdl` (the existing, previously unused file) is now analysed and elaborated, asserting the subtype mismatch at analysis, the bound violation at elaboration, and no internal error box for either;
- `attr.vhdl` and `attr_use.vhdl` are the same aggregate as the value of an attribute specification, which is the form it takes in the a2i sources, and the form that still crashed.

On master the test fails on mcode and llvm-jit (the `CONSTRAINT_ERROR` above). It passes on llvm both before and after, since the llvm path reports the bound check from the generated code and never reaches `Elab_Attribute_Specification`; the test is kept backend-independent so it covers both.

`sanity`, `gna` and `synth` all pass with the change on mcode, llvm and llvm-jit.

## What this does not do

It does not implement the unsupported aggregate. It does not need to: the aggregate in the report is invalid, and what a user needs is to be told which elements do not match -- which is what happens now, at analysis, with a location for each one.

closed 2026-09-05 by tgingold