ghdl/ghdl #1776: Connecting signals via procedure blocks their values.

m-kru, 2021-05-30:
I am not 100 % sure this is a bug.

**Description**
Direct connecting different signals with procedure blocks updating their values. 

**Expected behaviour**
Values should change (be updated).

**How to reproduce?**

```vhd :file: pkg.vhd
library ieee;
   use ieee.std_logic_1164.all;

package pkg is

   type t_record_a is record
      a : std_logic;
      b : std_logic;
   end record;

   type t_record_b is record
      a : std_logic;
      b : std_logic;
   end record;

   procedure ab_adapter (
      signal a : inout t_record_a;
      signal b : inout t_record_b
   );

end package;

package body pkg is

   procedure ab_adapter (
      signal a : inout t_record_a;
      signal b : inout t_record_b
   ) is
   begin
      a.a <= b.a;
      b.b <= a.b;
   end procedure;

end package body;
```

```vhd :file: tb.vhd
library ieee;
   use ieee.std_logic_1164.all;

library work;
   use work.pkg.all;

entity tb is
end entity;

architecture test of tb is

   signal a : t_record_a;
   signal b : t_record_b;

begin

   -- Below works correctly.
   --a.a <= b.a;
   --b.b <= a.b;

   -- Below does not work.
   ab_adapter(a, b);

   main : process is
   begin
      wait for 1 ns;
      b.a <= '1';
      a.b <= '1';

      wait for 1 ns;
      b.a <= '0';
      a.b <= '0';

      wait for 1 ns;
      b.a <= '1';
      a.b <= '1';

      wait for 2 ns;
      std.env.finish;
   end process;

end architecture;

```

```sh :image: ghdl/ghdl:buster-mcode
#!/bin/bash

ghdl -a --std=08 pkg.vhd
ghdl -a --std=08 tb.vhd
ghdl -m --std=08 tb
ghdl -r --std=08 tb --wave=wave.ghw
```

**Context**
Please, provide the following information:

- OS: 20.04.1-Ubuntu
- GHDL: GHDL 2.0.0-dev (1.0.0.r215.g82cdb8fa) [Dunoon edition]

**Additional context**
Wave when signals are connected via procedure:
[image]

Wave when signals are connected in architecture:
[image]

tgingold, 2021-05-30:
I suppose this is just a matter of drivers.  Try to replace std_logic(_vector) with std_ulogic(_vector).

m-kru, 2021-05-30:
@tgingold it is. However, there are no multiple logical drivers here. `b.a` and `a.b` are driven in the process, and `a.a` and `b.b` are driven in the procedure.

diogratia, 2021-05-31:
There are multiple drivers.

IEEE Std 1076-2008  

11.4 Concurrent procedure call statements

>A  concurrent  procedure  call  statement  represents  a  process  containing  the  corresponding  sequential  procedure call statement.

4. Suprograms and packages  
4.2 Subprogram declarations  
4.2.2 Formal parameters  
4.2.2.3 Signal parameters  

>For a formal parameter of class signal, references to the signal, the driver of the signal, or both, are passed into the subprogram call.  
>...  
>A  process  statement  contains  a  driver  for  each  actual  signal  associated  with  a  formal  signal  parameter  of  mode **out** or **inout** in a subprogram call. Similarly, a subprogram contains a driver for each formal signal parameter of mode **out** or **inout** declared in its subprogram specification.
>  
>For  a  signal  parameter  of  mode  **inout**  or  **out**,  the  driver  of  an  actual  signal  is  associated  with  the  corresponding driver of the formal signal parameter at the start of each call. Thereafter, during the execution of  the  subprogram  body,  an  assignment  to  the  driver  of  a  formal  signal  parameter  is  equivalent  to  an  assignment to the driver of the actual signal.

There are drivers for all the elements of the parameters of procedure ab_adaptor.

Making the element types of types t_record_a and t_record_b std_ulogic won't work:

```none
ghdl -r tb
for signal: .tb(test).b.a
./tb:error: several sources for unresolved signal
./tb:error: error during elaboration
```
It does however demonstrate there are multiple *sources* (6.4.2.3 Signal declarations, 14.7.3.2 Driving values, 4.6 Resolution functions).

m-kru, 2021-05-31:
Shame that passing it to the procedure as inout signal is enough to mark it as driven from the procedure, even if some signals are not driven there. Such construct would be really useful to reuse bus connections in test bench code.
Example:
```vhdl
   procedure uvvm_to_gc_wishbone_adapter (
      signal uvvm_wb_if : inout t_wishbone_if;
      signal gc_wb_ms   : inout t_wishbone_master_out;
      signal gc_wb_sm   : inout t_wishbone_master_in
   ) is
   begin
      -- General Cores WB <= UVVM WB
      gc_wb_ms.cyc <= uvvm_wb_if.cyc_o;
      gc_wb_ms.stb <= uvvm_wb_if.stb_o;
      gc_wb_ms.adr <= uvvm_wb_if.adr_o;
      gc_wb_ms.sel <= (others => '0');
      gc_wb_ms.we  <= uvvm_wb_if.we_o;
      gc_wb_ms.dat <= uvvm_wb_if.dat_o;

      -- UVVM WB <= General Cores WB
      uvvm_wb_if.dat_i <= gc_wb_sm.dat;
      uvvm_wb_if.ack_i <= gc_wb_sm.ack;
   end procedure;
```
Right now one has to copy and paste these connections into every test bench.

tgingold, 2021-05-31:
You can simply assign unused outputs to 'Z', so that there is no conflict during resolution.

marzoul, 2021-05-31:
@tgingold If it is enough to assign to 'Z' (within the procedure) the signals that we don't intend to drive there, it would be extremely helpful if that was exactly the default ghdl behavior for undriven signals in the procedure.
Is there a reason something preventing that ?

Because as of now the behavior is often extremely difficult to understand and sticking to the strictest LRM interpretation makes a hell of a designer's life.
Notably in the @m-kru experiments, it seems the existence of multidrivers or multi-source things is not even reported to the user.

I had similar issues with wide signal vectors assigned in multiple generate statements (for ... generate), or in processes in such generate statements. Each generate statement takes race of a subrange of the vectors.
In many scalable designs, using such construct is the way to go for clarity, modularity and maintainability. But the strictest multi driver LRM interpretation requires to add layers of intermadiate signals to reach a portable code...

umarcor, 2021-05-31:
@marzoul, my understanding is that the only reason for GHDL to behave as it does is because that's what the LRM dictates. GHDL is known for being strict with the LRM, even though there are some things in the language which may not make sense. We should not break that "contract" between developers and users, because that's one of the most valuable characteristics of GHDL which all the community (hackers, industry and academia) agrees on.
Therefore, such an automatic behaviour as the one you mention would need to be handled through `frelaxed` or some other switch making it explicit that it's not LRM compliant.

marzoul, 2021-05-31:
@umarcor that would be fine :-)
For any desirable change to make it to the next VHDL language revision, it has to exist somewhere so real users can comment on it.

About the assignment of large vectors inside for...generate statements, I know for example Vivado supports it and it has been a reason for me to use Vivado simulator on a per-design basis (with a bit of heart pain ;-) ). So I wonder what Vivado does with the present procedure example. Industry may be relying on this already, without realizing it's not strict LRM compliant.

diogratia, 2021-05-31:
There are places the body of a subprogram can't be examined during elaboration such as when they are found in a different secondary unit (a package body). That would leave attributes on parameter elements or the parameter itself, a compound mode whose invocation would require a reserved word decoration as well as the compound mode definition while the present behavior is valid when counting on resolution such as Tristan's suggested use of 'Z' assignments to 'not driven' elements.  

The idea seems unlikely to pass standardization, I recommended something similar for ports of composite types where eventually interfaces were adopted in -2019.

The major distinction between instantiating a component and invoking a concurrent procedure call here would be how the the drivers are determined. The difference is caused by subprogram calls being dynamically elaborated. 

It seems worth contemplating a -2019 (202X) compliant interfaces solution although it seems unlikely to be portable in my remaining lifetime based on feature uptake rates over the last three revisions of the standard. The driving force is the idea of hiding complexity (abstraction). Meanwhile we're still waiting on a fully compliant -2008 implementation from any tool provider and the value of standardization is diluted by separate feature uptake (pick and choose or non-compliant features).

In the meantime following Tristan's recommendation to assign 'Z's to a resolved subtype seems practicable although likely to lack portability without a successor to IEEE Std 1076.6-2004 (withdrawn) for synthesis. See IEEE Std 1076-2008 16.8.2.4.10 Interpretation of the high-impedance value ('Z'), 16.8.2.2 The STD_LOGIC_1164 values and G.2.11 Using STD_ULOGIC vs. STD_LOGIC.

umarcor, 2021-05-31:
> For any desirable change to make it to the next VHDL language revision, it has to exist somewhere so real users can comment on it.

@marzoul, are you aware of [gitlab.com/IEEE-P1076](https://gitlab.com/IEEE-P1076) and, precisely, [IEEE-P1076/VHDL-Issues](https://gitlab.com/IEEE-P1076/VHDL-Issues)?

marzoul, 2021-05-31:
@umarcor I did not. Thank you for the link !

closed 2021-06-05 by tgingold