ghdl/ghdl #1777: Use library from custom location through VHPIDIRECT, without specifying the absolute path

umarcor, 2021-06-01:
I want to use the same testbench and `caux.c` source for cosimulation regardless of the backend. In order to do so, I'm building `libcaux.so` and I'm setting `"VHPIDIRECT libcaux.so handleInterrupt"` as the foreign attribute string.

If `libcaux.so` is saved to `.` (pwd), both backends work as expected, both on Linux and on MSYS2. However, if the shared library is located in any other location, they fail.

With mcode, setting `LD_LIBRARY_PATH` is enough. It works. However, with LLVM, that is not enough. Elaboration fails because it cannot find the library: `gcc: error: libcaux.so: No such file or directory`. On Windows, none of them work. LLVM fails with the same error as on Linux, and mcode cannot find the library: `tb.vhd:6:12: cannot load VHPIDIRECT shared library 'libcaux.so'`

See:

- https://github.com/umarcor/SIEAV/actions/runs/894616617
- https://github.com/umarcor/SIEAV/tree/shared-from-subdir/issue-shared

That is based on https://github.com/ghdl/ghdl-cosim/tree/master/vhpidirect/shared/shlib.

tgingold, 2021-06-01:
Can you propose a mechanism ?

For LLVM/gcc backends, the name of the library is simply passed to the linker, which does the search.
For mcode backend, dlopen (or the Windows equivalent) is used to load the library. That function does the search.

So I suppose we need to add an option to specify the search path for VHPIDIRECT.  Would that work ?

umarcor, 2021-06-01:
> For LLVM/gcc backends, the name of the library is simply passed to the linker, which does the search.

Does this mean that, for LLVM/GCC backends, the library is compiled into the elaborated design and it is not used (searched for) at runtime? That feels weird... If a library is specified, I would expect it to be loaded at runtime, regardless of the backend. The use case with LLVM/GCC backends is building the executable model once and then allowing multiple simulations with different implementations of the shared library (modifying LD_LIBRARY_PATH accordingly).

However, I do understand why you are compiling the library into the executable by passing it to the linker. I guess it should be possible to pass `-L` to the compiler. I did try `-Wl,-Lpath/to/subdir` and `-Wl,-Wl,-Lpath/to/subdir`, but none of them worked. You said (privately) that there is no such option. But, I'm not sure to understand. According to https://ghdl.github.io/ghdl/using/CommandReference.html?highlight=wl#passing-options-to-other-programs, all `-Wl*` args are passed.
Anyway, I need to test it in C, in order to be sure about it.

Looking at the big picture, I would like not to specify the extension, unless I want to do it explicitly. So `"VHPIDIRECT libcaux handleInterrupt"` would be interpreted as `dlopen("libcaux.so")` on Linux, and the equivalent on Windows. Any of the backends should honour `LD_LIBRARY_PATH` (or `PATH` on Windows) by doing so.
On top of that, I would like to optionally specify an option when using GCC/LLVM for *not* using dlopen, but embedding it. In that case, the option or envvar that the linker honours must be found.

Moreover, I did some tests for defining the name of the library in the foreign attribute string, through either a generic or a constant. None of those are supported. GHDL complains about `value of FOREIGN attribute must be locally static`. That would be another reason for supporting dlopen regardless of the backend. If the name of the library is set later in the elaboration process, the current compilation model cannot link it when `ghdl -e` is executed. Naturally, I know that this is not written in the standard, and it might not be accepted. However, it's something I'd like to propose, as long as it's technically possible.

umarcor, 2021-06-01:
> Does this mean that, for LLVM/GCC backends, the library is compiled into the elaborated design and it is not used (searched for) at runtime?

Replying to myself. In C, the library is required both during compilation (`-L./subdir -lcore`) and at runtime (`LD_LIBRARY_PATH`). If `dlopen` is used `-ldl` is required during compilation (no additional custom path) and the envvar at runtime.

```C
// core.c

#include <stdio.h>

int custom_func(int argc, char **argv) {
  printf("> custom_func\n");
  return 0;
}
```

```C
// maindl.c

#include <dlfcn.h>
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv) {
  typedef int main_t(int, char**);
  void * client_hndl;

  client_hndl = dlopen("libcore.so", RTLD_LAZY);
  if (!client_hndl){
      fprintf(stderr, "%s\n", dlerror());
      exit(1);
  }

  main_t *custom_func = (main_t*)dlsym(client_hndl, "custom_func");
  if (!custom_func){
      fprintf(stderr, "%s\n", dlerror());
      exit(2);
  }

  custom_func(1, (char*[]){"client", 0});

  return 0;
}
```

```C
// main.c

extern int custom_func(int argc, char **argv);

int main(int argc, char **argv) {
  custom_func(1, (char*[]){"client", 0});
  return 0;
}
```

```sh
#!/usr/bin/env sh

set -e

cd $(dirname "$0")

mkdir -p subdir

gcc -fPIC -shared core.c -o subdir/libcore.so

export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$(pwd)/subdir"

gcc maindl.c -o maindl -ldl
./maindl
echo "Successfully run maindl"

gcc main.c -o main -L./subdir -lcore
./main
echo "Successfully run main"
```

tgingold, 2021-06-01:
There are multiple points, so we might need different issues.

In GHDL, calls to foreign subprograms are direct.  This means the symbols must be resolved before the execution.  With LLVM/GCC backends, this is done using the regular mechanism: the linker.  Note that non-shared libraries can also be used. With mcode, the libraries are first gathered, then opened and finally the symbols are resolved using dlsym.

umarcor, 2021-06-01:
> There are multiple points, so we might need different issues.

Can we start with supporting `ghdl elab-run -Wl,-L./subdir tb` when LLVM/GCC are used?

Currently, `ghdl elab-run tb` is equivalent to `gcc main.c -ldl` for mcode and `gcc main.c -o -lcore` for GCC/LLVM. Let's just add the missing `-L./subdir`.

My perception is that's a matter of order. Can you help me find where do I need to look for seeing the order of the arguments that GHDL passes to LLVM/GCC? I believe the `-L` arg *must* be before `-lcore`. Maybe that's why `-Wl,_l./subdir` does not work at the moment, even though the option itself is accepted.

tgingold, 2021-06-01:
You can always add `-v` to see how the subprograms are called.

So yes, the -Wl options is appended after the object files.  Should we simply put them before ?

marlonjames, 2021-06-03:
Just for LRM context on VHPIDIRECT:

> 20.2.4.3 Standard direct binding

> object_library_path ::=
graphic_character { graphic_character }

> A host system interprets an object library path in a manner not defined by this standard to locate a physical object library. It is an error if the host system cannot locate the physical object library identified by an object library path.

> NOTE—A host system may interpret an object library path by appending an implementation-dependent file-name
extension, such as “.so” or “.dll,” to derive a file pathname. It is recommended that a file-name extension in an object
library path be omitted so that an implementation can append an extension that is appropriate for the host environment