ghdl/ghdl #1778: wrong in xilinx primitive

the-centry, 2021-06-03:
While compile the project mentioned in [the problem](https://github.com/ghdl/ghdl/issues/1307) and happened errors! What's wrong with this?
command as mentioned in the above problem :
`yosys -m ghdl -p 'ghdl -Punisim -fsynopsys -fexplicit -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -edif hdmi_design.edif'`
the error was:
-- Running command `ghdl -Punisim -fsynopsys -fexplicit -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -edif hdmi_design.edif' --

1. Executing GHDL.
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: no declaration for "pll_lockg" in package "vcomponents"
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_BINARY = '1' and locked_out_tmp = '0') else 'H';
                           ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: target is not a signal name
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_BINARY = '1' and locked_out_tmp = '0') else 'H';
                           ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1209:27:error: no declaration for "gsr"
        GSR_dly <= TO_X01(GSR);
                          ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/IDELAYE2.vhd:198:30:error: no declaration for "gsr"
  GSR_dly          <= TO_X01(GSR)      after 0 ps;
                             ^
./src/deserialiser_1_to_10.vhd:105:1:warning: instance "iserdese2_master" of component "iserdese2" is not bound [-Wbinding]
ISERDESE2_master : ISERDESE2
^
./src/deserialiser_1_to_10.vhd:67:14:warning: (in default configuration of deserialiser_1_to_10(behavioral))
./src/deserialiser_1_to_10.vhd:142:1:warning: instance "iserdese2_slave" of component "iserdese2" is not bound [-Wbinding]
ISERDESE2_slave : ISERDESE2
^
./src/deserialiser_1_to_10.vhd:67:14:warning: (in default configuration of deserialiser_1_to_10(behavioral))
./src/serialiser_10_to_1.vhd:73:1:warning: instance "master_serdes" of component "oserdese2" is not bound [-Wbinding]
master_serdes : OSERDESE2
^
./src/serialiser_10_to_1.vhd:66:14:warning: (in default configuration of serialiser_10_to_1(behavioral))
./src/serialiser_10_to_1.vhd:121:1:warning: instance "slave_serdes" of component "oserdese2" is not bound [-Wbinding]
slave_serdes : OSERDESE2
^
./src/serialiser_10_to_1.vhd:66:14:warning: (in default configuration of serialiser_10_to_1(behavioral))
ERROR: vhdl import failed.

tgingold, 2021-06-03:
Try the option `--vendor-library=xilinx`

It will assume that units from the library are black boxes.

the-centry, 2021-06-03:
> --vendor-library
Thanks ,but the error happened as these:
`-- Running command `ghdl -Punisim -fsynopsys -fexplicit --vendor-library=xilinx -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -blif htest.eblif' --

1. Executing GHDL.
*command line*:1:1:error: cannot find resource library "xilinx"
`

tgingold, 2021-06-03:
Ah sorry, `--vendor-library=unisims`

the-centry, 2021-06-04:
> Ah sorry, `--vendor-library=unisims`
Thanks a lot,but the error is the same.
Using command `--vendor-library=unisim`
`-- Running command `ghdl -Punisim -fsynopsys -fexplicit --vendor-library=unisim -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -blif htest.eblif' --

1. Executing GHDL.
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: no declaration for "pll_lockg" in package "vcomponents"
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_BINARY = '1' and locked_out_tmp = '0') else 'H';
                           ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: target is not a signal name
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_BINARY = '1' and locked_out_tmp = '0') else 'H';
                           ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1209:27:error: no declaration for "gsr"
        GSR_dly <= TO_X01(GSR);
                          ^
/opt/Xilinx/Vivado/2017.2/data/vhdl/src/unisims/primitive/IDELAYE2.vhd:198:30:error: no declaration for "gsr"
  GSR_dly          <= TO_X01(GSR)      after 0 ps;
                             ^
ERROR: vhdl import failed.`

Using command `--vendor-library=unisims`
`-- Running command `ghdl -Punisim -fsynopsys -fexplicit --vendor-library=unisims -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -blif htest.eblif' --

1. Executing GHDL.
*command line*:1:1:error: cannot find resource library "unisims"`

Whether it means that the precompile xilinx primitives has some ploblem?

tgingold, 2021-06-04:
So yes, it is possible that the Xilinx libraries are not fully correct (or they expect to use the verilog ones).

But, are you instantiating components or entities ?

the-centry, 2021-06-09:
Components! Year may it is the problem of Xilinx libraries, But according to the guideline that ghdl should have tried to use xilinx primitives.