ghdl/ghdl #1779: Setting of integer values using VPI not implemented

BenediktO, 2021-06-04:
**Description**
It is impossible to set integer inputs using the VPI interface

**Expected behaviour**
I use cocotb to create testbenches. Setting integer inputs does not work at all, the signal does not get updated.
Since integers are widely used I would expect this to work.

**How to reproduce?**
The following VHDL code can be used to reproduce:

vartype_test.vhd
```vhd :file: vartype_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity vartype_test is
   port (
      in_data_int        : in  integer range 0 to 255;
      in_data_stdvector  : in  std_logic_vector(7 downto 0);
      out_std2int        : out integer range 0 to 255;
      out_std2std        : out std_logic_vector(7 downto 0);
      out_int2int        : out integer range 0 to 255;
      out_int2std        : out std_logic_vector(7 downto 0)
   );
end entity;

architecture vartype_test_rtl of vartype_test is
begin
   out_int2int <= in_data_int;
   out_int2std <= std_logic_vector(to_unsigned(in_data_int, 8));
   out_std2std <= in_data_stdvector;
   out_std2int <= to_integer(unsigned(in_data_stdvector));
end architecture;

```

To run this I use:
test_vartype.py
```python :file: test_vartype.py
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def adder_basic_test(dut):
    print('input, out_std2int, out_std2std, out_int2int, out_int2std')

    for value in range(32):
        dut.in_data_int <= value
        dut.in_data_stdvector <= value

        await Timer(1, units='step')

        print('%5d, %11d, %11d, %11d, %11d' % (
            value, dut.out_std2int.value.integer, dut.out_std2std.value.integer,
            dut.out_int2int.value.integer, dut.out_int2std.value.integer
        ))
```
and Makefile
```make :file: Makefile
TOPLEVEL_LANG ?= vhdl
SIM ?= ghdl

PWD=$(shell pwd)

VHDL_SOURCES = $(PWD)/vartype.vhdl

TOPLEVEL := vartype
MODULE   := test_vartype

include $(shell cocotb-config --makefiles)/Makefile.sim
```

Instead of returning a table with 32 rows where all cells in a row are equal I get this:
```
input, out_std2int, out_std2std, out_int2int, out_int2std
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
    0,           0,           0,           0,           0
    1,           1,           1,           0,           0
    2,           2,           2,           0,           0
    3,           3,           3,           0,           0
    4,           4,           4,           0,           0
    5,           5,           5,           0,           0
    6,           6,           6,           0,           0
    7,           7,           7,           0,           0
    8,           8,           8,           0,           0
    9,           9,           9,           0,           0
   10,          10,          10,           0,           0
   11,          11,          11,           0,           0
   12,          12,          12,           0,           0
   13,          13,          13,           0,           0
   14,          14,          14,           0,           0
   15,          15,          15,           0,           0
   16,          16,          16,           0,           0
   17,          17,          17,           0,           0
   18,          18,          18,           0,           0
   19,          19,          19,           0,           0
   20,          20,          20,           0,           0
   21,          21,          21,           0,           0
   22,          22,          22,           0,           0
   23,          23,          23,           0,           0
   24,          24,          24,           0,           0
   25,          25,          25,           0,           0
   26,          26,          26,           0,           0
   27,          27,          27,           0,           0
   28,          28,          28,           0,           0
   29,          29,          29,           0,           0
   30,          30,          30,           0,           0
   31,          31,          31,           0,           0
```

Returning integers and std_vector works as expected. Setting std_vector works too, but setting integers does not work as can be seen from which columns always return 0.

**Context**
Please, provide the following information:

- OS: Debian 10 64 bit kernel: 4.19.0-16-amd64
- Origin: 
  - [x] Package manager: `GHDL version 2.0.0-dev (1.0.0.r191.gd003498a.dirty) [Dunoon edition]`

I first filed an Issue in the GHDL git repository:
https://github.com/cocotb/cocotb/issues/2556
There ktbarrett traced the problem to https://github.com/ghdl/ghdl/blob/c33f41e4612347d1fcbb9a37bc82249d7b8366eb/src/grt/grt-vpi.adb#L1068
This really useful functionality is not implemented.

tgingold, 2021-06-05:
I think it is now handled.  I haven't tested with cocotb yet.

closed 2021-06-05 by tgingold

BenediktO, 2021-06-05:
Thanks a lot for the quick fix!
I can confirm that this works with cocotb.

tgingold, 2021-06-05:
Great, thank you for the status update.

commit ce33129f (2021-06-05): vpi: handle integer32 signals for vpi_put_value.  Fix #1779