ghdl/ghdl #1780: Crash when synthesizing memories

umarcor, 2021-06-09:
The following source produces a crash when trying to synthesize:

```vhd :file: mem1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity imem is
  port (
    clk_i  : in  std_ulogic;
    rden_i : in  std_ulogic;
    wren_i : in  std_ulogic;
    ben_i  : in  std_ulogic_vector(03 downto 0);
    addr_i : in  std_ulogic_vector(31 downto 0);
    data_i : in  std_ulogic_vector(31 downto 0);
    data_o : out std_ulogic_vector(31 downto 0);
    ack_o  : out std_ulogic
  );
end entity;

architecture imem_rtl of imem is

  signal addr : std_ulogic_vector(15 downto 0);

  type ram_t is array(0 to 2**15-1) of std_ulogic_vector(31 downto 0);

begin

  addr <= addr_i(addr'left+2 downto 2); -- word aligned

  verbose: block
  begin

    process(clk_i)
      variable memory : ram_t;
    begin
      if rising_edge(clk_i) then
        if wren_i then
          for x in 0 to 3 loop
            if ben_i(x) then
              memory(to_integer(unsigned(addr)))((x+1)*8-1 downto x*8) := data_i((x+1)*8-1 downto x*8);
            end if;
          end loop;
        end if;
      end if;
    end process;

    process(clk_i)
      variable memory : ram_t;
    begin
      if rising_edge(clk_i) then
        if rden_i then
          data_o <= memory(to_integer(unsigned(addr)));
        end if;
      end if;
    end process;
  end block;

end architecture;
```

```sh
# ghdl synth --std=08 mem1.vhd -e imem
mem1.vhd:14:5:warning: no assignment for port "ack_o"
    ack_o  : out std_ulogic
    ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r257.g82665d42) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-w64-mingw32
T:\neorv32\ghdl-mem-synth\
Command line:
C:\msys64\mingw64\bin\ghdl.exe synth --std=08 mem1.vhd -e imem
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:774
Call stack traceback locations:
0x7ff603c7c0a6 0x7ff603a57188 0x7ff603a4fc8e 0x7ff603a723ce 0x7ff603a72698 0x7ff603c0b050 0x7ff603c0b0dc 0x7ff603c03519 0x7ff603c59885 0x7ff603b61d6f 0x7ff603c63aab 0x7ff603a02bca 0x7ff603a013bf 0x7ff603a014f4 0x7fffced17032 0x7fffd0cc264f
******************************************************************
```

I don't necessarily expect this description to work. It's something I found when testing multiple RAM inference sources.

tgingold, 2021-06-12:
I don't get a crash for this design.  However, the memory is not recognized.

tgingold, 2021-06-12:
But well, there is no memory.

umarcor, 2021-06-12:
> I don't get a crash for this design.

I will test it on other platforms, to find whether the crash is specific to LLVM on Windows.

> However, the memory is not recognized.

Note that I discussed this with @tmeissner. We think that the description might imply two memories to be inferred. A read-only memory which is not initialized, and a write-only memory. Therefore, it is arguable whether any synthesis outcome should be given. Maybe the sensible solution is to optimise and remove both of those?

However, I can see use cases where non-initialized read-only memories or write-only memories make sense. Some FPGAs allow BRAMs to retain data after dynamic reconfiguration. Hence, one design might initialize or read the BRAMs that another design modified earlier or later.

umarcor, 2021-06-12:
FTR,

> MemoryUsageGuideforiCE40Devices.pdf
> **Pre-loading EBR Data**
>
>The data contents for an EBR block can be optionally pre-loaded during iCE40 configuration. If not pre-loaded dur-ing configuration, then the EBR contents must be initialized by the iCE40 application before the EBR contents are valid. EBR initialization data can be done in the RTL code. Pre-loading the EBR data in the configuration bitstream increases the size of the configuration image accordingly.
> 
> **EBR Contents Preserved During Configuration**
> 
> EBR contents are preserved (write protected) during configuration, assuming that voltage supplies are maintained throughout. Consequently, data can be passed between multiple iCE40 configurations by leaving it in an EBR block and then skipping pre-loading during the subsequent reconfiguration

and

> FPGA-TN-02022-1-3-iCE40-SPRAM-Usage-Guide.pdf
> **SPRAM Content during Warmboot**
>
> During configuration through Warmboot, SPRAM  content is not loaded. Thus, previous data on the SPRAM are retained.

tgingold, 2021-06-12:
Yes but I think you'd better to directly instantiate a RAM for such use.

I don't plan to handle this use case now.

umarcor, 2021-06-12:
Fair enough. Shall I close this as `wontfix`?

tgingold, 2021-06-12:
Yes, the CI passes with this test.

closed 2021-06-13 by umarcor

tmeissner, 2021-06-13:
I also think that such memories which hold their content during re-configuration is out of scope. I think, for such specific needs one have to directly instantiate RAM macros. The synthesis tool wouldn't be able to distinguish if the code is for such a specific design or simply wrong without such intention.