ghdl/ghdl #1785: Windows Release 1.0.0 or newer

Malcolmnixon, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
I'd like to upgrade from release 0.37 to something newer; however:
- Release 1.0.0 does not include pre-built binaries for any operating system. Unlike the other releases I only see the source archives.
- Newer nightly builds archive windows builds with z-std compression which is an extremely poorly supported compression format on Windows. I've had to install Cygwin and get the command-line z-std decompressor to get the files.

**Describe the solution you'd like**
I'd like to see the addition of 1.0.0 pre-built binaries to the release; or if it's difficult to add assets; perhaps a new release. In addition I'd like the windows versions of the binaries released in a format supported by Windows utilities such as 7-Zip (E.G. tgz like the other operating systems).

**Describe alternatives you've considered**
I've considered building the 1.0.0 windows binaries from the source archives; but it would be significant effort replicating the build infrastructure currently used by GHDL.

**Additional context**
N/A

umarcor, 2021-06-14:
Hi @Malcolmnixon! As explained in https://ghdl.github.io/ghdl/#ghdl-v1-0-0rc1-was-tagged, since January we no longer provide assets for releases or snapshots. We recommend users to use package managers or nightly assets. See [hdl/packages](https://github.com/hdl/packages).

On Windows, the recommended approach is to use MSYS2. In fact, those zstd tarballs are MSYS2 packages which can be installed through `pacman -U *.zst`. However, most users should not need to install nightly packages. The versions upstream should work:

```sh
# pacman -Ss ghdl
mingw32/mingw-w64-i686-ghdl-mcode 1.0.0.r257.g82665d42-1 (mingw-w64-i686-eda) [installed]
    GHDL: the open-source analyzer, compiler, simulator and (experimental) synthesizer for VHDL (mcode backend)    (mingw-w64)
mingw64/mingw-w64-x86_64-ghdl-llvm 1.0.0.r257.g82665d42-1 (mingw-w64-x86_64-eda) [installed]
    GHDL: the open-source analyzer, compiler, simulator and (experimental) synthesizer for VHDL (LLVM backend)
    (mingw-w64)
```

Once installed through MSYS2, you can set environment variables GHDL and/or GHDL_PREFIX, in order to use it from outside MSYS2 (say powershell or cmd). That is done by https://github.com/ghdl/setup-ghdl-ci/blob/master/main.js#L102-L105, and used/tested in https://github.com/ghdl/extended-tests/blob/main/.github/workflows/Test.yml#L90-L93.

Just extracting the zst tarballs anywhere is not guaranteed to work. You might be lucky if using the mcode backend. With LLVM backend it is likely to crash due to missing DLLs. See #1560, #1599 and #1773 for discussions related to providing an standalone package/installer for Windows.

> I've considered building the 1.0.0 windows binaries from the source archives; but it would be significant effort replicating the build infrastructure currently used by GHDL.

Not really. The build infrastructure for Windows/MSYS2 is probably the easiest one. For mcode:

1. Install MSYS2 (see https://github.com/msys2/msys2-installer/releases/tag/nightly-x86_64 and [msys2.org](https://www.msys2.org/)).
2. Open an MINGW32 shell.
3. Install minimal dependencies: `pacman -S base-devel git mingw-w64-i686-toolchain`.
4. Clone GHDL: `git clone https://github.com/ghdl/ghdl`
5. Change directory and build: `cd scripts/msys2-mcode; MINGW_ARCH=mingw32 makepkg-mingw --noconfirm --noprogressbar -sCLf`
6. If successful, install the package: `pacman -U *.zst`

See https://github.com/ghdl/ghdl/blob/487e786b744d487f3f5f009d94974d13ee8e0680/.github/workflows/Test.yml#L225-L243

Malcolmnixon, 2021-06-14:
I see. We've been using the mcode version of ghdl on our developer machines and build nodes for a while without using msys2.

I did take a look at the llvm version, but we're not overly concerned with performance, and instead are more interested in having the minimal number of dependencies and a 100% repeatable build/test environment.

I think it's going to be difficult to push for an entire msys2 environment install (especially considering these machines aren't even allowed to connect to the intertubes), so I'll look into manually unpacking the mcode package and making a stand-alone zip/tgz file we can use internally.

Many thanks,
 - Malcolm

Paebbels, 2021-06-15:
@Malcolmnixon see isse https://github.com/ghdl/ghdl/issues/1560 for what DLLs from MinGW64 are needed where in the install directories of GHDL. Then you can run GHDL standalone.

umarcor, 2021-06-15:
> I see. We've been using the mcode version of ghdl on our developer machines and build nodes for a while without using msys2.

I understand. I can help you setup a CI job in a repo of yours for automatically converting the nightly GHDL packages into a zipfile, a tarball or a 7z archive, which you can more easily distribute and use. Unfortunately, I am not comfortable with providing that here, because we cannot guarantee it to work, due to all the dependencies being missing.

> I did take a look at the llvm version, but we're not overly concerned with performance, and instead are more interested in having the minimal number of dependencies and a 100% repeatable build/test environment.

It's not performance only. mcode does not work on 64 bit Windows. On GNU/Linux 32 bits and 64 bits are supported. On Windows 32 bits only. Moreover, from a cosimulation perspective, LLVM allows providing custom foreign (C/C++) code to GHDL. Conversely, mcode requires any foreign code to be compiled beforehand and made available as a shared library.

mcode does have some unique features too. Apart from less dependencies for simulation only, the time scale can be changed, which allows significant performance gains, depending on the target design. Overall, it should not be one or the other, but both are useful depending on the task. Nevertheless, I understand your point.

> I think it's going to be difficult to push for an entire msys2 environment install (especially considering these machines aren't even allowed to connect to the intertubes), so I'll look into manually unpacking the mcode package and making a stand-alone zip/tgz file we can use internally.

Please, see the issue referenced by @Paebbels and also #1773. I won't maintain a manual list of DLLs/deps/assets myself. However, I will help setup CI for using manually maintained lists and for generating standalone zipfiles/installers, as done in GTKWave (see https://github.com/gtkwave/gtkwave/tree/master/MSYS2).

Moreover, note that it should be technically possible to build GHDL on Windows without MSYS2, using AdaCore's GNAT. That was done long ago (see #1599), but it's not included in CI. If anyone updates those scripts, or finds an automated script for achieving that, we might replace the MSYS2 jobs in the daily CI. In the end, GHDL is upstreamed to MSYS2 repos, so we don't need to rebuild that here. We do it because that's the only current solution we have for Windows.

Malcolmnixon, 2021-06-15:
I expanded the latest nightly build (mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst) and tested it on a machine which didn't have MSYS2 and it worked fine as a stand-alone executable (able to run a bunch of VHDL test benches).

I then repeated the same test using the official msys2 package (https://packages.msys2.org/package/mingw-w64-i686-ghdl) and it also worked stand-alone without needing any additional DLLs.

Analyzing ghdl.exe suggests that perhaps the libraries may have been statically linked in rather than requiring the DLLs.

For now it looks like pulling the package apart provides a usable stand-alone windows executable.

Thanks,
 - Malcolm

umarcor, 2021-06-15:
> I expanded the latest nightly build (mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst) and tested it on a machine which didn't have MSYS2 and it worked fine as a stand-alone executable (able to run a bunch of VHDL test benches).

I suggest trying the testsuite(s) in this repo: `GHDL=path/to/hdl ./testsuite/testsuite.sh`

> I then repeated the same test using the official msys2 package (https://packages.msys2.org/package/mingw-w64-i686-ghdl) and it also worked stand-alone without needing any additional DLLs.
>
> Analyzing [...]

The recipe here and in MSYS2 are the same in this regard. See https://github.com/ghdl/ghdl/blob/master/scripts/msys2-mcode/PKGBUILD#L21-L23 and https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-ghdl/PKGBUILD#L32-L34.

If you are going to use GHDL for simulation of simple testbenches only, then it might work with you. However, if you try generating waveforms, synthesis through yosys, or using the language, I would expect problems. Just be aware in case you find those.

targeted, 2021-06-16:
I'd vote for this. Seriously, if building a release for Windows is as easy as you describe, why don't you do it once, spend 15 minutes in a year, produce a portable zip, save a lot of headache for your users. I don't want to follow down yet another package manager, search for files, locate missing DLLs, adjust environment variables, just to make a command-line compiler work. And I know for sure it can just work, it *was* that excellent. Instead, you [quote](https://github.com/ghdl/ghdl/pull/1773#issuecomment-861287820) Knuth with his "software is hard". It turns *me* down for sure.

Which is to say, I tried building ghdl on msys2 several times. It builds and exe runs, and displays help, but it does not produce anything. No complaints, no errors, but no output ether. No obj, no exe. Only work-obj93.cf. This is frustrating, and when I come hoping for a binary release, what do I see...

umarcor, 2021-06-16:
> Seriously, if building a release for Windows is as easy as you describe, why don't you do it once, spend 15 minutes in a year, produce a portable zip, save a lot of headache for your users. I don't [...]

@targeted, please, go ahead. I myself am not capable of doing it in 15 minutes. If you are confident about it being so easy and producing excellent results, please contribute! Should you have 15 spare minutes every few months, we might provide multiple installers a year. Moreover, if you want to take care of that specific part of CI, we will have nightly installers for everyone!

> Which is to say, I tried building ghdl on msys2 several times. It builds and exe runs, and displays help, but it does not produce anything. No complaints, no errors, but no output ether. No obj, no [...]

I believe this is unrelated to the build environment. It's likely because of the backend you are using (mcode), and how GHDL works. Please, read the *Hello World* example of the User Guide: https://ghdl.github.io/ghdl/quick_start/simulation/hello/index.html

> *Analysis generates a file, hello.o, which is the object file corresponding to your VHDL program. **This is not created with mcode**.*

> *-e can be bypassed with mcode, since -r actually elaborates the design and saves it on memory before running the simulation. But you can still use it to check for some elaboration problems.*

See also: https://ghdl.github.io/ghdl/development/building/index.html

> *The output of both GCC and LLVM is an executable file, but mcode does not generate any. Therefore, if using GCC/LLVM, the call with argument -r can be replaced with direct execution of the binary.*

Feel free to contribute enhancements to the docs, in case you find that not to be clear enough.

targeted, 2021-06-17:
@umarcor You are right. I will go try build what I need again, and if I can't, I still can use 0.37. Thank you.

targeted, 2021-06-18:
So, to amend some of my previous arrogance, here is what you need to do to install ghdl-1.0.0 on Windows and compile your project to a nice exe file.

1. Install msys2. 

Currently https://repo.msys2.org/distrib/x86_64/msys2-x86_64-20210604.exe

2. Update msys2 as per the official site.

From MSYS shell:
```
$ pacman -Syu
$ pacman -Su
$ pacman -S --needed base-devel mingw-w64-x86_64-toolchain
```

3. Install llvm version of ghdl (duh!)

From MSYS shell:
```
$ pacman -S mingw-w64-x86_64-ghdl-llvm
```

From MSYS-MinGW64 shell:
```
$ ghdl --version
GHDL 2.0.0-dev (1.0.0.r257.g82665d42) [Dunoon edition]
 Compiled with GNAT Version: 10.3.0
 llvm code generator
Written by Tristan Gingold.
```

4. Add environment vars:
```
GHDL=C:\msys64\mingw64\bin\ghdl.exe
GHDL_PREFIX=C:\msys64\mingw64\lib\ghdl
PATH=%PATH%;C:\msys64\mingw64\bin
```

5. Get at it:
```
c:> ghdl.exe -a hello.vhdl
c:> ghdl.exe -e hello_world
c:> ./hello_world.exe
Hello world!
```

umarcor, 2021-06-18:
That is really nice! Just a clarification:

This is required if you want to build GHDL from sources:

```sh
$ pacman -S --needed base-devel mingw-w64-x86_64-toolchain
```

Then, you would do:

```sh
# Get the sources
$ git clone https://github.com/ghdl/ghdl/

# Move to the directory with the recipe
$ cd scripts/msys2-llvm

# Build GHDL for generating a package/tarball
$ MINGW_ARCH=mingw64 makepkg-mingw --noconfirm --noprogressbar -sCLf

# Install the tarball which we just built
$ pacman -U *.zst
```

Conversely, when doing `pacman -S mingw-w64-x86_64-ghdl-llvm`, you are not building GHDL from sources, but installing the latest version available in MSYS2 repositories. Therefore, you don't need `base-devel` or `*--toolchain` packages for this purpose. You can skip the third step of your initial update/install of MSYS2.

As an intermediate solution, you can download https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst, and install it through `pacman -U *.zst`. That package is equivalent to the one you might build yourself, but it's ready to install/update.

Anyway, I myself update the MSYS2 packages, and I try to do it once a month, at least. Therefore, by using `pacman -S`, as you did, you should get a pretty up to date version. Remember to execute `pacman -Syu` once in a while. For instance, the version you got is from the 9th of May: https://packages.msys2.org/package/mingw-w64-x86_64-ghdl-llvm?repo=mingw64. But I pushed an update today, which is pending (search ghdl in https://packages.msys2.org/queue). In 1-2 days, when you run `pacman -Syu`, GHDL will be updated automatically for you.

---

> `PATH=%PATH%;C:\msys64\mingw64\bin`

I think it might work without adding that directory to the PATH. GHDL should be able to find everything it needs with just GHDL and GHDL_PREFIX. Please, let us know otherwise :wink:

---

Hint: there is an EDA package in MSYS2, where you can find other interesting tools apart from GHDL: https://packages.msys2.org/group/mingw-w64-x86_64-eda. `pacman -S mingw-w64-x86_64-eda` will let you choose whether to install all of them or just the ones you want. Find further details in https://hdl.github.io/MINGW-packages/.

Paebbels, 2021-06-18:
@targeted what is the size of such an installation? It sounds quite big as it installs a whole MSYS2 + complete LLVM.

umarcor, 2021-06-18:
I bet it's >2GB. It's 1-1.5GB if GHDL and LLLVM are installed only.

However that has been discussed with MSYS2 maintainers and there is room for packaging runtime LLVM libs only.

There is also #1547 for investigating an static build.

Paebbels, 2021-06-18:
Don't get me wrong this is a first nice start to get GHDL running as before as we had a ZIP deployment for Windows.

The question is, is a complete MSYS2 needed?
Do we need `cp`, `ls`, `mv`, ... ?

The approach I used in the linked issue is less then 100 MB.

umarcor, 2021-06-18:
> The question is, is a complete MSYS2 needed?

It is not. Ruby, Qt, Git for Windows and others provide a very stripped down version of it.

Paebbels, 2021-06-18:
Any idea how `Git for Windows` strips down MinGW to a suitable size?
Can we borrow scripts or CI from them?

---------------------
Ups, `Git for Windows` has 750 MB ...

umarcor, 2021-06-18:
Yeah. Maybe Git for Windows is not the best example, because they don't provide git only, but also some of those utilities you mentioned.

umarcor, 2021-06-18:
> Can we borrow scripts or CI from them?

The only semi-automatic tools I know are the ones mentioned in https://github.com/ghdl/ghdl/issues/1560#issuecomment-752750494. See https://www.msys2.org/wiki/Distributing/.

targeted, 2021-06-19:
@Paebbels when installed like that, it takes 2.7G from c:\ but for me it is not a concern. As @umarcor points out, it's not necessary to install build toolset to just get ghdl working. That reduces the size to 1.8G. It would have been possible to extract ghdl.exe along with the dll it depends on, but if I understand correctly, ghdl-llvm requires gcc. Also, my particular environment runs on makefiles, so I need make which is part of the toolset, and it's easier to just install everything at once.

umarcor, 2021-06-19:
> Also, my particular environment runs on makefiles, so I need make which is part of the toolset, and it's easier to just install everything at once.

@targeted, thanks for acknowledging this. While, I do understand why some users want an small (50-200 MB) standalone installer/zipfile, my personal preference is to just install everything at once, as you said. 

> but if I understand correctly, ghdl-llvm requires gcc.

Strictly, it's supposed to be "any compiler". In practice, there are some default parameters internally which default to GCC, so it's the most straightforward solution. That is also my main reason for using MSYS2 and not an isolated GHDL.

Anyway, we should be able to produce UCRT64 builds on MSYS2 soon, which should make it easier to e.g. use it with Visual Studio or other non-gnu toolchains.