ghdl/ghdl #1786: pyGHDL : is utils.constructs_iter working?

LeFl0w, 2021-06-17:
**Description**
I wanted to give a try to constructor approach for pyGHDL.
So far reading `constructs_iter` description, I assume this generator can handle entity declaration and signal declaration.
When I try the function it stops at the first level : Entity.
If I did something wrong, let me know.

**Expected behaviour**
I was expected the generator to release all the intermediates nodes like Interface_Signal_Declaration or Signal_Declaration. These nodes comes from the call of `chain_iter`  onto `Entity_Declaration` or `Architecture_Body` nodes. See the following extract from `utils.constructs_iter`
```
   elif k in (
        nodes.Iir_Kind.Entity_Declaration,
        nodes.Iir_Kind.Architecture_Body,
        nodes.Iir_Kind.Block_Statement,
        nodes.Iir_Kind.Generate_Statement_Body,
    ):
        for n1 in chain_iter(nodes.Get_Declaration_Chain(n)):
            yield n1
            for n2 in constructs_iter(n1):
                yield n2
        for n1 in chain_iter(nodes.Get_Concurrent_Statement_Chain(n)):
            yield n1
            for n2 in constructs_iter(n1):
                yield n2
```
**How to reproduce?**
The full project is hosted on [github ](https://github.com/LeFl0w/Test_pyGHDL/tree/test_generator) on branch test_generator. You can reproduce on any vhdl file using the following python code (change the vhdl file name at the bottom ): 

```python
import pyGHDL.libghdl     as libghdl
from pyGHDL.libghdl       import name_table, files_map, errorout_console
from pyGHDL.libghdl.vhdl  import nodes, sem_lib
import pyGHDL.libghdl.utils as pyutils
import pyGHDL.libghdl.vhdl.nodes_meta as nodes_meta
from typing import List, Any, Generator

def init():
    """Initialization: set options and then load libaries"""
    libghdl.initialize()

    # Print error messages on the console.
    errorout_console.Install_Handler()

    # Set options. This must be done before analyze_init()
    libghdl.set_option("--std=08")

    # Finish initialization. This will load the standard package.
    if libghdl.analyze_init_status() != 0:
        self.fail("libghdl initialization error")

#################################################
############ Global node functions 
#################################################
def getIdentifier(node):
    """Return the Python string from node :obj:`node` identifier"""
    return name_table.Get_Name_Ptr(nodes.Get_Identifier(node))

def getNodeLineInFile (node):
    """ Return the line in original file of the node :obj:`node` """
    loc = nodes.Get_Location(node)
    fil = files_map.Location_To_File(loc)
    pos = files_map.Location_File_To_Pos(loc, fil)
    line = files_map.Location_File_To_Line(loc, fil)
    return line

def getNodeColumInFile (node):
    """ Return the column in original file of the node :obj:`node` """
    loc = nodes.Get_Location(node)
    fil = files_map.Location_To_File(loc)
    pos = files_map.Location_File_To_Pos(loc, fil)
    line = files_map.Location_File_To_Line(loc, fil)
    col = files_map.Location_File_Line_To_Offset(loc, fil, line)
    return col

def GetNodeType (node) -> str:
    """Return the name of the node type"""
    if nodes.Get_Kind(node) == nodes.Iir_Kind.Design_Unit:
        return "Design name: "
    elif nodes.Get_Kind(node) == nodes.Iir_Kind.Entity_Declaration:
        return "Entity name: "
    elif  nodes.Get_Kind(node) == nodes.Iir_Kind.Interface_Signal_Declaration:
        return "Entity Interface port name: "
    elif nodes.Get_Kind(node) == nodes.Iir_Kind.Architecture_Body:
         return "Architecture name: "
    elif nodes.Get_Kind(node) == nodes.Iir_Kind.Constant_Declaration:
        return "Constant declaration name: "
    elif nodes.Get_Kind(node) == nodes.Iir_Kind.Signal_Declaration:
        return "Signal declaration name: "    
    else :
        return "Unknown type: "

def DisplayNodeInfo(node) -> str:
    """Return General information regarding the node"""
    return GetNodeType(node)+ str(getIdentifier(node)) + " |l-"+ str(getNodeLineInFile(node)) + " c-:"+ str(getNodeColumInFile(node))
#################################################
############End Global node functions 
#################################################

def list_units(filename):
    """ 
        Display informations about the VHDL file.
        At the end this function could be simlar has vhdl-disp_tree but for human eyes
     """
    # Load the file
    file_id = name_table.Get_Identifier(str(filename))
    sfe = files_map.Read_Source_File(name_table.Null_Identifier, file_id)
    if sfe == files_map.No_Source_File_Entry:
        print("cannot open file '%s'" % filename)
        return

    # Parse
    file = sem_lib.Load_File(sfe)

    # Display all design units
    designUnit = nodes.Get_First_Design_Unit(file)
    #print(DisplayNodeInfo(designUnit))

    #iterate over every VHDL constructs below designUnit
    #for libraryUnit in pyutils.constructs_iter(designUnit):
    for libraryUnit in libghdl.utils.constructs_iter(designUnit):
        print(DisplayNodeInfo(libraryUnit))

def main(self):
    init()
    list_units(self)

if __name__ == "__main__":
    main("adder_file.vhd")
```

The terminal output is `Entity name: adder |l-3 c-:7 ` and stop at entity level. There should be a line for each element returned by the generator.

**Context**

- OS: Linux ubuntu 20.04LTS
- Origin: 
       + GHDL 2.0.0-dev (1.0.0.r315.gf66524bb) [Dunoon edition]
       + Compiled with GNAT Version: 9.3.0
       + GCC back-end code generator
- Python : 
      + Python 3.6.13+ (heads/3.6:1b6f4e5, Jun  3 2021, 11:03:16) [GCC 7.5.0] on linux

umarcor, 2021-06-17:
@LeFl0w, I did not read your message properly. I apologise for that.

However, @Paebbels spent the past days working throroughly on pyVHDLModel and pyGHDL.dom. See the following references:

- https://gitter.im/ghdl1/Lobby?at=60ca2796c705e53c1c962a51
- #1590, #1788

Yesterday, I had a video meeting with him, for discussing/understanding what he is doing, how does it fit with the examples/use cases I want to cover, and which is the vision going forward. After the meeting, I did the following minimal example: https://github.com/umarcor/neorv32/commit/4861d542c18bc83f234141a2d62210674f9f9715

As you can see, that is a completely external repo ([NEORV32](https://github.com/stnolting/neorv32) is a soft CPU written in VHDL), because I wanted to try it in any arbitrary design. The installation procedure is retrieving branch `paebbels/libghdl-decorator` from Patrick's fork of GHDL, and the `dev` branch of pyVHDLModel. They depend on Python package `pydecor` only (apart from GHDL). I decided to use PYTHONPATH instead of installing the packages, but that is up to you. Moreover, I tried the same solution on Windows (MSYS2), apart from the GNU/Linux containers.

The usage, as you see, is as simple as:

```py
app = Application()
app.addFile(Path(path/to/source/file), "neorv32")
```

That will parse the source and use a default pretty print function. See the output in: https://github.com/umarcor/neorv32/runs/2844979363?check_suite_focus=true

```
Document 'rtl/core/neorv32_cpu_decompressor.vhd':
  Entities:
  - neorv32_cpu_decompressor
    Generics:
    Ports:
      - ci_instr16_i : in std_ulogic_vector(15 downto 0)
      - ci_illegal_o : out std_ulogic
      - ci_instr32_o : out std_ulogic_vector(31 downto 0)
    Declared:
  Architectures:
  - neorv32_cpu_decompressor_rtl
    Declared:
      - constant ci_opcode_lsb_c : natural := 0
      - constant ci_opcode_msb_c : natural := 1
      - constant ci_rd_3_lsb_c : natural := 2
      - constant ci_rd_3_msb_c : natural := 4
      - constant ci_rd_5_lsb_c : natural := 7
      - constant ci_rd_5_msb_c : natural := 11
      - constant ci_rs1_3_lsb_c : natural := 7
      - constant ci_rs1_3_msb_c : natural := 9
      - constant ci_rs1_5_lsb_c : natural := 7
      - constant ci_rs1_5_msb_c : natural := 11
      - constant ci_rs2_3_lsb_c : natural := 2
      - constant ci_rs2_3_msb_c : natural := 4
      - constant ci_rs2_5_lsb_c : natural := 2
      - constant ci_rs2_5_msb_c : natural := 6
      - constant ci_funct3_lsb_c : natural := 13
      - constant ci_funct3_msb_c : natural := 15
  Packages:
  PackageBodies:
  Configurations:
  Contexts:
```

There are several reasons for using pyGHDL.dom, instead of libghdl or pyGHDL.libghdl. The main motivation is that Patrick added some handy error handling. When you try to use some feature which is not implemented yet, you should get a nice warning/exception with some meaningul data, instead of a crash.

pyGHDL.dom does already handle getting a list of ports, signals and/or constants from either entities or architectures (I think that constants in packages might be supported too). You have been asking about how to get the location of those identifiers in the source. I don't know if you want to do that for retrieving the literal content, or as a workaround/mechanism for achieving some other target. Anyway, I believe that Patrick did already implement a nicer feature for the iterators that you are trying to use. It would make sense to add the "get the location in the source" on top of that.

I feel that you can use Patrick's branch and implement an alternative pretty print function, or just adapt it for your purposes. What do you think?

umarcor, 2021-06-17:
@Paebbels reorganised the content in the past hours. Now changes are merged in the master branch of pyVHDLModel and a new release was published in PyPI. So, no need to use a custom branch for that. With regard to GHDL, changes for testing are in branch Paebbels/pyGHDL now.

tgingold, 2021-06-17:
Your entity has no declarations (although it has ports).
And constructs_iter doesn't return any port or generic.
That's why you get an empty output.

I think constructs_iter should be modified to return ports and generics!

LeFl0w, 2021-06-23:
Hi thanks for the feedback.
 I'll definitely will be using pyGHDL.dom at the end. The problem is that it is not enough mature right now for my proof of concept.
As the python interface for GHDL is evolving a lot these days, I wanted to be as near as possible to libghdl avoid any change.

For now on , I can achieve listing of entity port name and signal  and perform the linting on them like this:

```
plasma.vhd
INFO:Analyze  plasma.vhd
INFO: Rule CNE_02500 : list all entity ports
mem_pause_in(54:8)
no_ddr_start(55:8)
no_ddr_stop(56:8)
INFO: Rule CNE_02600 : list all signals
address_next(63:10)
byte_we_next(64:10)
cpu_address(65:10)
cpu_byte_we(66:10)
cpu_data_w(67:10)
cpu_data_r(68:10)
cpu_pause(69:10)
data_read_uart(71:10)
write_enable(72:10)
eth_pause_in(73:10)
eth_pause(74:10)
enable_misc(77:10)
enable_uart(78:10)
enable_uart_read(79:10)
enable_uart_write(80:10)
enable_eth(81:10)
gpio0_reg(83:10)
uart_write_busy(84:10)
uart_data_avail(85:10)
irq_mask_reg(86:10)
irq_status(87:10)
irq_eth_rec(89:10)
irq_eth_send(90:10)
counter_reg(91:10)
ram_enable(93:10)
ram_byte_we(94:10)
ram_address(95:10)
ram_data_w(96:10)
ram_data_r(97:10)
cache_access(99:10)
cache_checking(100:10)
cache_miss(101:10)
cache_hit(102:10)
```
This is enough for now. The final question I have is the handling of external package while parsing the file. It seem to crash the analysis but I have to investigate it.

Paebbels, 2021-06-23:
@LeFl0w what is missing in DOM, maybe I can concentrate on that in the next days?

Today I finished a lot of different type definitions (enum, array, record, access).
DOM can also list all design units including entities.

At next I'll focus on functions and subprograms.

LeFl0w, 2021-06-23:
Excellent thank you very much. I will try it as soon as I can. Maybe you could integrate link back to the original file : name, line, column that will be of great use for me.
Florent

closed 2021-06-23 by LeFl0w

umarcor, 2021-06-23:
@LeFl0w, in fact, 2-3 days ago Patrick implemented linking back to the original file, as I was slightly annoying with him about that, so that the DOM would/could cover your particular use case. See the XFAIL tests in https://ghdl.github.io/extended-tests/, you will see that the captured stdout field contains the file, line and column where the problematic statement is located. Therefore, the functionality is there already. Moreover, we peeked at what you are doing for implementing it.

Would you kindly let us know what do you want to do with your sources? We will think about how it can be achieved with DOM, what we can provide at the moment, and what you can/need to implement on top of that.

One of the practical use cases we have in mind is the implementation of formatters (such as black, yapf, eslint, etc.). That is, implementing https://github.com/jeremiah-c-leary/vhdl-style-guide on top of the DOM. My personal priority is generating a "tree explorer" of "black-boxes"; that is, entities/components, their generics/ports, and the known architectures available. I believe that both of those use case do overlap with whatever application/tool you want to build.

LeFl0w, 2021-06-23:
Thanks for the feedback. In fact the project is called vhdltool (https://github.com/VHDLTool).

It started with a handbook of vhdl coding rules (https://github.com/VHDLTool/VHDL_Handbook_STD and https://github.com/VHDLTool/VHDL_Handbook_CNE).

We added of toolchain to convert the xml handbook from xml to pdf. Then we started to develop a rulechecker. Back then ghdl was sleeping, so we started with zamiacad ( https://github.com/VHDLTool/Zamiacad-Rulechecker) as a ghdl parser and AST generator.
We added a sonarqube plugin to monitor vhdl debt but also coverage (with modelsim and ghdl coverage) and fpga metrics (occupation, timing..). (https://github.com/VHDLTool/sonar-VHDLRC). Everything can be tested with docker.
We recently add rule checked with yosys (for state machine).

I wanted to add extra rule using ghdl with python. The main objective is to implement the remaining rules with pyghdl and then later on replace zamiacad with ghdl/pyghdl engine.

Hope this clarifies things.

umarcor, 2021-06-23:
Now I remember! I attended [Continuous VHDL inspection with opensource software - Florent Manni - ORConf 2019](https://www.youtube.com/watch?v=xjX4aTHUkDY). I don't recall talking to you directly, tho...

I think that pyGHDL.dom does absolutely fit as a replacement for the parsing and analysis engine in VHDLTool, in the mid-long term. See a minimal example in https://github.com/VHDL/pyVHDLModel/pull/18/files. We setup CI for having an up to date summary of how pyGHDL.dom is evolving: [ghdl.github.io/extended-tests](https://ghdl.github.io/extended-tests/).

I suggest you try the strategy we use in ghdl/extended-tests, in your own repos (either public or private): https://github.com/ghdl/extended-tests/blob/main/.github/workflows/Test.yml#L116-L130

```sh
# Install the version of pyGHDL from master or the one that matches your local GHDL:
$ pip3 install git+https://github.com/ghdl/ghdl.git@master
$ pip3 install git+https://github.com/ghdl/ghdl.git@$(ghdl version hash)

# Install pytest and, optionally, pytest-reporter and/or pytest-xdist
$ pip3 install pytest pytest-reporter pytest-reporter-html1 pytest-xdist

# Clone the repositories with VHDL sources which you want to have analyzed

# Run TestDOM with pytest
$ pytest TestDOM.py
# or, for executing tests in parallel
$ pytest -n $(nproc) TestDOM.py
# or, for generating the HTML report
$ pytest --template=html1/index.html --report=report.html TestDOM.py
# or, for executing tests in parallel and generating the HTML report
$ pytest -n $(nproc) --template=html1/index.html --report=report.html TestDOM.py
```

You can add you own variant of [DOM.py](https://github.com/ghdl/ghdl/blob/master/pyGHDL/cli/DOM.py) and call it in `TestDOM.py` instead of `ghdl-dom`. That is, replace the call to `prettyPrint` with your custom function where you evaluate your own rules. Even if your VHDL sources are not public, we can collaborate in that specific function/processor.

LeFl0w, 2021-06-24:
I installed 983236ac3dfd0c455a0ac910a9a468ea2c81e5d9 version.
I tried it with your example:

```
#ghdl DOM
from pathlib import Path
from pyGHDL.dom.NonStandard import Design, Document

def test_dom():   
    sourceFile = Path("adder_file.vhd")
    design = Design()
    #library = Design.GetLibrary("work")
    document = Document(sourceFile)
    #design.AddDocument(document, library)
    for entity in document.Entities:
        print("{}".format(entity.Name))
        print("  generics:")
        for generic in entity.Generics:
            print("  - {} : {!s} {}".format(generic.Name, generic.Mode, generic.SubTypeIndication))
    
        print("  ports:")
        for port in entity.Ports:
            print("  - {} : {!s} {}".format(port.Name, port.Mode, port.SubTypeIndication))

test_dom()
```

but I got this error
 ```
pyGHDL.dom.Common.DOMException: Unknown handling of subtype kind '75' of subtype indication '1318' while parsing port 't3'.
```
with the code: 

[adder_file.vhd.txt](https://github.com/ghdl/ghdl/files/6708202/adder_file.vhd.txt)

i think you do not support integer with ranges which is handle differently than std_logic_vector or integer maybe [this could help](https://github.com/LeFl0w/Test_pyGHDL/blob/3b6412fc44fe66afc6d2c5537539d6693f6225e4/lib_archi.py#L170-L182)

umarcor, 2021-08-15:
Hi @LeFlow!

pyVHDLModel and pyGHDL.dom were further improved during the last weeks. Among many other enhancements, they can now handle integer ranges, multiple identifiers in generics, ports, etc., which were some of the issues with you adder example.

This is the updated and simplified snippet:

```py
from pathlib import Path
from pyGHDL.dom.NonStandard import Document

document = Document(Path("adder_file.vhd"))
for entity in document.Entities:
    print("{}".format(entity.Identifier))
    print("  generics:")
    for generic in entity.GenericItems:
        print("  - {} : {!s} {}".format(generic.Identifiers, generic.Mode, generic.Subtype))
    print("  ports:")
    for port in entity.PortItems:
        print("  - {} : {!s} {}".format(port.Identifiers, port.Mode, port.Subtype))
```

And this is the output:

```sh
adder
  generics:
  ports:
  - ['i0', 'i1'] : in bit
  - ['ci'] : in bit
  - ['s'] : out bit
  - ['co'] : out bit
  - ['test'] : inout bit
  - ['t1'] : in std_logic
  - ['t2'] : out std_logic_vector
  - ['t3'] : in integer
  - ['t4'] : out integer
```

As commented in https://github.com/ghdl/ghdl/issues/1437#issuecomment-899115096, we created several demos on top of pyGHDL.dom. For instance, [OSVDE](https://umarcor.github.io/osvb/apis/project/OSVDE.html) is similar to the snippet above: https://github.com/umarcor/osvb/blob/0ae9d71e37b478c5e07f6a35379903db3c5c4356/mods/pyOSVDE/main.py#L254-L274. The Sphinx integration example produces a similar output too: https://umarcor.github.io/osvb/apis/project/DocGen.html#integration-with-sphinx.