ghdl/ghdl #1790: Using Microsemi Pre-Compiled Simulation Libraries with GHDL

justinjabate, 2021-06-17:
Microsemi provides pre-compiled simulation libraries
[here](https://www.microsemi.com/product-directory/fpga-design-tools/5535-compiled-simulation-libraries) for a handful of commercial simulators. 

Are any of these downloads able to be used with GHDL?

tmeissner, 2021-06-17:
I'm pretty sure that libraries pre-compiled with simulators other than GHDL aren't compatible with GHDL.

umarcor, 2021-06-17:
My understanding is that they might not be able to do it, even if they actually used GHDL for generating those pre-compiled libraries.

On the one hand, I think that GHDL does need access to the sources during elaboration. In fact, there is no such concept of pre-compiled libraries with mcode. We use that expression, but it means "create the cf table".

On the other hand, if they used LLVM/GCC, they might generate `*.o` objects. However, since both GHDL and GRT are licensed under GPL, Microchip might not be legally allowed to distribute those, unless they provide access to the HDL source. At that point, it is useless for them, because their strategy is based on opacity. Strictly, the `*.o` object might be not affected by GPL, that might apply to executables only. Yet, again, I think that the `*.o` files might be useless unless elaborated together with GRT (either through ghdl or using an external compiler).

If Microchip provided VHDL sources, as done by other vendors, those might be handled through https://github.com/ghdl/ghdl/tree/master/scripts/vendors, so we provide a more pleasant user experience to GHDL users. Unfortunately, there is little we can do if they want to keep their sources non open.

Paebbels, 2021-06-17:
The installation directory of Libero SoC 12.6 contains just a few VHDL sources like packages for every supported hardware architecture. Here they provide low-level components.

```vhdl
package COMPONENTS is

COMPONENT SLE
  port(
                                  D : in    STD_LOGIC;
                                CLK : in    STD_LOGIC;
                                 EN : in    STD_LOGIC;
                                ALn : in    STD_LOGIC;
                                ADn : in    STD_LOGIC;
                                SLn : in    STD_LOGIC;
                                 SD : in    STD_LOGIC;
                                LAT : in    STD_LOGIC;
                                  Q : out   STD_LOGIC
  );
END COMPONENT;

COMPONENT DFN1
  port(
                                  D : in    STD_LOGIC;
                                CLK : in    STD_LOGIC;
                                  Q : out   STD_LOGIC
  );
END COMPONENT;

...
```
Source: `C:\Microsemi\LiberoSoC\12.6\Designer\libfp\actel\vhdl`

This could help to have no instantiated components in a design, but there is no behavior defined unless someone writes a open source model for MicroSemi primitives ....

umarcor, 2021-06-17:
> This could help to have no instantiated components in a design, but there is no behavior defined unless someone writes a open source model for MicroSemi primitives ....

BTW, I think this is the case with other vendors too. Not all of them provide simulation VHDL for their components. Some provide Verilog only, some don't provide any.

tmeissner, 2021-06-17:
Microsemi only provides Verilog versions of the behavior models for their FPGA architectures. We compile them and use them with Questa at work.

justinjabate, 2021-06-18:
Thanks to everyone for the feedback! Appreciate it

closed 2021-06-18 by justinjabate