ghdl/ghdl #1792: Synthesis error "cannot extract same variable part for dynamic slice" in synthesizable files.

Nemmerle, 2021-06-17:
**Description**
When running synthesis in a file that was synthesized in the past with commercial tools, ghdl reports an error ```cannot extract same variable part for dynamic slice```, although it is previously run ghdl -a successfully.

**Expected behaviour**
No error shown and proper file synthesis.

**How to reproduce?**
I manage to clean the file and provide this Minimal Working Example (MWE) that reproduces the issue:

```vhd :file: foo.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity foo is
    port (
        clk   : in std_logic;
        reset : in std_logic;
        foo_input : in std_logic_vector (255 downto 0);
        foo_out   : out std_logic_vector (31 downto 0)       
    );
end entity;

architecture rtl of foo is

    type register_data is
        record
           msb_position : natural range 0 to 15;
           lsb_position : natural range 0 to 15;
        end record;

    type array_custom_t is array (0 to 1) of register_data; 

    constant register_A : register_data := 
    (   msb_position     => 15,
        lsb_position     => 0
    );

    constant register_B : register_data := 
    (   msb_position     => 15,
        lsb_position     => 0
    );

    constant array_registers : array_custom_t := (
        1   => register_A,
        others => register_B
    );
    

    signal long_signal : std_logic_vector(31 downto 0);    

begin

    process (clk)
        variable integer_variable: integer;
        variable reg_defs : register_data;
        variable msb : natural;
        variable lsb : natural;
    begin
        if rising_edge(clk) then
            if reset = '1' then
                long_signal <= (others=>'0');
            else
                for i in 0 to (array_registers'length - 1) loop   

                -- Retrieve subregister information
                    reg_defs := array_registers(i);
                    msb := reg_defs.msb_position;
                    lsb := reg_defs.lsb_position;

                    integer_variable  := to_integer(unsigned(foo_input(msb downto lsb)));

                    long_signal <= std_logic_vector(to_unsigned(integer_variable,long_signal'length));
                end loop;            
            end if;
        end if;
    end process;    

    foo_out <= long_signal;

end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a foo.vhd
ghdl --synth foo
```

**Context**

- OS: Linux AND Windows 10 (tested in both)
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `GHDL 2.0.0-dev (1.0.0.r246.g9fb3ee1a) [Dunoon edition]`

**Additional context**
I cleaned the sources after the build and I don't know what commit sha was at checkout. ghdl says
```sh
ghdl -version
GHDL 2.0.0-dev (1.0.0.r246.g9fb3ee1a) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
 GCC back-end code generator
```

The error was first noticed in a Windows 10 machine, with a ghdl-mcode version.

Thank you for your time.

umarcor, 2021-06-17:
@Nemmerle are you using ghdl-mcode on MSYS2's MINGW32?

Nemmerle, 2021-06-17:
I guess it is the migw version. I downloaded ghdl-mcode from a nightly build and the file name starts with mingw-.

In my linux station I am using a ghdl with gcc backend compiled from sources, and the error is also present there.

umarcor, 2021-06-17:
You might try https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst. That should be slightly newer. Maybe there is no difference at all, tho. However, if you built the Linux/GCC recently, then don't bother about trying. The version you reported above is ~100 commits behind master; not a huge number, but somehow outdated.

Nemmerle, 2021-06-17:
I have tried with last version you suggest (1.0.0.r345-g0c726ac3) and the error is also present.

tgingold, 2021-06-18:
This is due to:

>                     integer_variable  := to_integer(unsigned(foo_input(msb downto lsb)));

GHDL was not able to statically evaluate `msb` and `lsb`.  I am pretty sure this could be improved.  I have to have a look at it.

umarcor, 2022-05-02:
I can confirm that this error is still present in `GHDL 3.0.0-dev (v2.0.0-143-g01c330fa)`.

tgingold, 2023-06-16:
The right way to write the slice is to make the length of it explicit, like:

  foo_input(lsb+15 downto lsb)

In your case, the length is buried in the array.